stm32duino/Arduino_Core_STM32 #564: Use DMA

poky99, 2019-07-16:
Good day.
I tried to use DMA UART1, here is my code

UART_HandleTypeDef huart1;
DMA_HandleTypeDef hdma_usart1_tx;

void DMA2_Stream7_IRQHandler(void)
{
  HAL_DMA_IRQHandler(&hdma_usart1_tx);
}

void USART1_IRQHandler(void)
{
  HAL_UART_IRQHandler(&huart1);
}

static void MX_DMA_Init(void)
{
  /* DMA controller clock enable */
  GPIO_InitTypeDef GPIO_InitStruct = {0};
  __HAL_RCC_USART1_CLK_ENABLE();
  __HAL_RCC_GPIOA_CLK_ENABLE();
  GPIO_InitStruct.Pin = GPIO_PIN_9 | GPIO_PIN_10;
  GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
  GPIO_InitStruct.Pull = GPIO_PULLUP;
  GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
  GPIO_InitStruct.Alternate = GPIO_AF7_USART1;
  HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);
  __DMA2_CLK_ENABLE();
  //__HAL_RCC_DMA2_CLK_ENABLE();

  hdma_usart1_tx.Instance = DMA2_Stream7;
  hdma_usart1_tx.Init.Channel = DMA_CHANNEL_4;
  hdma_usart1_tx.Init.Direction = DMA_MEMORY_TO_PERIPH;
  hdma_usart1_tx.Init.PeriphInc = DMA_PINC_DISABLE;
  hdma_usart1_tx.Init.MemInc = DMA_MINC_ENABLE;
  hdma_usart1_tx.Init.PeriphDataAlignment = DMA_PDATAALIGN_BYTE;
  hdma_usart1_tx.Init.MemDataAlignment = DMA_MDATAALIGN_BYTE;
  hdma_usart1_tx.Init.Mode = DMA_NORMAL;
  hdma_usart1_tx.Init.Priority = DMA_PRIORITY_LOW;
  hdma_usart1_tx.Init.FIFOMode = DMA_FIFOMODE_DISABLE;
  if (HAL_DMA_Init(&hdma_usart1_tx) != HAL_OK)
  {
    Error_Handler();
  }
  __HAL_LINKDMA(&huart1, hdmatx, hdma_usart1_tx);

  /* DMA interrupt init */
  HAL_NVIC_SetPriority(DMA2_Stream7_IRQn, 0, 0);
  HAL_NVIC_EnableIRQ(DMA2_Stream7_IRQn);
}

static void MX_USART1_UART_Init(void)
{
  huart1.Instance = USART1;
  huart1.Init.BaudRate = 9600;
  huart1.Init.WordLength = UART_WORDLENGTH_8B;
  huart1.Init.StopBits = UART_STOPBITS_1;
  huart1.Init.Parity = UART_PARITY_NONE;
  huart1.Init.Mode = UART_MODE_TX_RX;
  huart1.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart1.Init.OverSampling = UART_OVERSAMPLING_8;
  if (HAL_UART_Init(&huart1) != HAL_OK)
  {
    Error_Handler();
  }
  /* Peripheral interrupt init*/
  HAL_NVIC_SetPriority(USART1_IRQn, 0, 0);
  HAL_NVIC_EnableIRQ(USART1_IRQn);
}

void setup() {
  HAL_Init();
  MX_DMA_Init();
  MX_USART1_UART_Init();
}

void loop()
{
  char buffer[7] = "abcdef";
  HAL_UART_Transmit_DMA(&huart1, (uint8_t*) buffer, sizeof(buffer));
  HAL_Delay(1000);
}
After the first sending, the board hangs. How to use DMA correctly?
thank!
OS Win 7 x64, Core 1.6.0, Arduino 1.8.9

fpistm, 2019-07-16:
Hi @poky99 

Please use the forum to get support as described in the README.

Thanks in advance.

closed 2019-07-16 by fpistm

poky99, 2019-07-16:
Well, please show me an example, before writing a post, I watched all the posts with DMA and the example was not found, I checked on the STM32CubeIDE DMA works. Part of the code I took from the file stm32f4xx_hal_msp.c

mrguen, 2019-12-02:
Also looking to use DMA and did not found DMA libraries for Arduino.