stm32duino/Arduino_Core_STM32 #569: error: arm_math.h: No such file or directory

odo2063, 2019-07-23:
**Please, before reporting any issue**
- Make sure you are using the latest Arduino_Core_STM32 version.
CHECK
  https://github.com/stm32duino/Arduino_Core_STM32/releases/latest
- Make sure the issue is not already reported/fixed on GitHub or discussed on the [stm32duino forum](http://stm32duino.com)
CHECK
- Submit a GitHub issue only for reporting a problem related to the [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).
CHECK
- **Avoid** to submit a GitHub issue for project troubleshooting.
MAYBE
 

When reporting any issue, please try to provide all relevant information:

**Describe the bug**
get "error: arm_math.h: No such file or directory
 #include <arm_math.h>"
when "#define ARM_MATH_CM4
#define __FPU_PRESENT 1
#include <arm_math.h>"

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Enter Code
2. Press Compile
3. See error

**Expected behavior**
clean compilation

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 1.8.9
 - STM32 core version: 1.6.1

**Board (please complete the following information):**
 - Name: Nucleo-144 L4R5ZI
 - Hardware Revision: ??

**Additional context**
Code comes from www.micromodeler.com

fpistm, 2019-07-23:
Hi  @odo2063 
I've no issue to build the  Nucleo-144 L4R5ZI.

Which sketch you trying to build ?

odo2063, 2019-07-23:
This should be the minimal example. imu_test.ino is based on http://www.micromodeler.com/articles/IntroductionToDSP/coefficients.jsp
filter1.h and filter1.cpp are straight copies from www.micromodeler.com.

```
//imu_test.ino
#include "filter1.h"

#define BNO055_SAMPLERATE_DELAY_MS (10)

Adafruit_BNO055 bno = Adafruit_BNO055();

filter1Type *filter1 = filter1_create(); //create Filter as in http://www.micromodeler.com/articles/IntroductionToDSP/coefficients.jsp
float wtf[1];
wtf[0] = 1.0;

void setup(void)
{

}

void loop(void)
{
  filter1Type *filter1 = filter1_create();
  while(1)
  {
  
  filter1_writeInput( &filter1, wtf[0] );       // Read a sample from the ADC and write it into the filter
  //wtf = filter1_outputToFloat( filter1_readOutput( filter1 )) ;

  delay(BNO055_SAMPLERATE_DELAY_MS);
  }
}
```
```
//filter1.h
/******************************* SOURCE LICENSE *********************************
  Copyright (c) 2019 MicroModeler.

  A non-exclusive, nontransferable, perpetual, royalty-free license is granted to the Licensee to
  use the following Information for academic, non-profit, or government-sponsored research purposes.
  Use of the following Information under this License is restricted to NON-COMMERCIAL PURPOSES ONLY.
  Commercial use of the following Information requires a separately executed written license agreement.

  This Information is distributed WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

******************************* END OF LICENSE *********************************/

// A commercial license for MicroModeler DSP can be obtained at http://www.micromodeler.com/launch.jsp

// Begin header file, filter1.h

#ifndef filter1_H_ // Include guards
#define filter1_H_

#define ARM_MATH_CM4  // Use ARM Cortex M4
#define __FPU_PRESENT 1   // Does this device have a floating point unit?
#include <arm_math.h> // Include CMSIS header

// Link with library: libarm_cortexM4_mathL.a (or equivalent)
// Add CMSIS/Lib/GCC to the library search path
// Add CMSIS/Include to the include search path
extern float32_t filter1_coefficients[20];
static const int filter1_numStages = 4;

typedef struct
{
  arm_biquad_casd_df1_inst_f32 instance;
  float32_t state[16];
  float32_t output;
} filter1Type;

filter1Type *filter1_create( void );
void filter1_destroy( filter1Type *pObject );
void filter1_init( filter1Type * pThis );
void filter1_reset( filter1Type * pThis );
#define filter1_writeInput( pThis, input )  \
  arm_biquad_cascade_df1_f32( &pThis->instance, &input, &pThis->output, 1 );

#define filter1_readOutput( pThis )  \
  pThis->output

int filter1_filterBlock( filter1Type * pThis, float * pInput, float * pOutput, unsigned int count );
#define filter1_outputToFloat( output )  \
  (output)

#define filter1_inputFromFloat( input )  \
  (input)

#endif // filter1_H_
```
```
//filter1.cpp
/******************************* SOURCE LICENSE *********************************
  Copyright (c) 2019 MicroModeler.

  A non-exclusive, nontransferable, perpetual, royalty-free license is granted to the Licensee to
  use the following Information for academic, non-profit, or government-sponsored research purposes.
  Use of the following Information under this License is restricted to NON-COMMERCIAL PURPOSES ONLY.
  Commercial use of the following Information requires a separately executed written license agreement.

  This Information is distributed WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

******************************* END OF LICENSE *********************************/

// A commercial license for MicroModeler DSP can be obtained at http://www.micromodeler.com/launch.jsp

#include "filter1.h"

#include <stdlib.h> // For malloc/free
#include <string.h> // For memset

float32_t filter1_coefficients[20] =
{
  // Scaled for floating point

  0.8477774141503206, -1.6955548283006412, 0.8477774141503206, 1.8975700676560592, -0.9006183651747467,// b0, b1, b2, a1, a2
  1, -2, 1, 1.9545851741285774, -0.9577162876370862,// b0, b1, b2, a1, a2
  1, 2, 1, -1.874282866033339, -0.878865954532929,// b0, b1, b2, a1, a2
  1, 2, 1, -1.943381371995732, -0.948120138561757// b0, b1, b2, a1, a2

};

filter1Type *filter1_create( void )
{
  filter1Type *result = (filter1Type *)malloc( sizeof( filter1Type ) ); // Allocate memory for the object
  filter1_init( result );                     // Initialize it
  return result;                                // Return the result
}

void filter1_destroy( filter1Type *pObject )
{
  free( pObject );
}

void filter1_init( filter1Type * pThis )
{
  arm_biquad_cascade_df1_init_f32(  &pThis->instance, filter1_numStages, filter1_coefficients, pThis->state );
  filter1_reset( pThis );

}

void filter1_reset( filter1Type * pThis )
{
  memset( &pThis->state, 0, sizeof( pThis->state ) ); // Reset state to 0
  pThis->output = 0;                  // Reset output

}

int filter1_filterBlock( filter1Type * pThis, float * pInput, float * pOutput, unsigned int count )
{
  arm_biquad_cascade_df1_f32( &pThis->instance, pInput, pOutput, count );
  return count;

}
```

odo2063, 2019-07-23:
This looks fine to me...but i might be wrong...

Detecting libraries used...
/home/user/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb @/tmp/arduino_build_586936/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -I/home/user/Arduino/imu_test_5 -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/avr -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32 -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/LL -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/usb -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/usb/hid -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/usb/cdc -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/STM32L4xx_HAL_Driver/Inc/ -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/STM32L4xx_HAL_Driver/Src/ -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/STM32L4xx/ -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -w -x c++ -E -CC -DSTM32L4xx -DARDUINO=10809 -DARDUINO_NUCLEO_L4R5ZI -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"NUCLEO_L4R5ZI\"" -DSTM32L4R5xx -DHAL_UART_MODULE_ENABLED **-I/home/user/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/Core/Include/** -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/ -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/ -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino -I/home/user/.arduino15/packages/STM32/hardware/stm32/1.6.1/variants/NUCLEO_L4R5ZI /tmp/arduino_build_586936/sketch/imu_test_5.ino.cpp -o /dev/null

odo2063, 2019-07-23:
OK...found something -I/home/user/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/**DSP**/Include/ is missing in plattform.txt(line 66)...but now i run into a new Problem...undefined reference to `arm_biquad_cascade_df1_init_f32'

fpistm, 2019-07-23:
Could you attach the zip file of your sketch (with all sources files).
Then I could help.

fpistm, 2019-07-23:
@odo2063 
you have partially solved your issue.
Adding the CMSIS DSP include path is not enough.
It requires to add the `FilteringFunctions.c` in the core path else Arduino will not build it.
I've made a branch on my fork and I'm able to build your example (I've made some fixes to your code).
https://github.com/fpistm/Arduino_Core_STM32/commit/e891921bd42e9f7819ef0d7004feed4f9ec53f96

odo2063, 2019-07-24:
nice compiles for me, Thank you very very much XD ...one last Question: the FPU only works for single precision(float) and will not improve double precision calculations...is that correct?

fpistm, 2019-07-24:
This mainly depends on your use case. working only on float, single precision is enough. For double using Double precision is better else I think you will lost precision.
https://www.st.com/content/ccc/resource/technical/document/application_note/10/6b/dc/ea/5b/6e/47/46/DM00047230.pdf/files/DM00047230.pdf/jcr:content/translations/en.DM00047230.pdf

odo2063, 2019-07-24:
We will stay on the L4R5ZI...so testing and see what happens...great, Thank for your Help, I do really appreciate it!!!

closed 2019-07-24 by odo2063