stm32duino/Arduino_Core_STM32 #572: black F407VG board

nopnop2002, 2019-07-23:
I use this black F407VG board.
[image]

I wrote a sketch using Nucleo.
[image]

The parameters for writing are as follows.
[image]

The writing ends normally and the black board starts to work.
However, if i turn the Nucleo off and on again, the black board will not work.
Do you have any advice? ?

fpistm, 2019-07-24:
Hi @nopnop2002 
I don't see any jumper on Boot pins.
I guess you restart in STM32 bootloader mode after power off the Nucleo then power on.

It works when you flash because the STM32CubeProgrammer start the program automatically.

nopnop2002, 2019-07-24:
OH! Thank you.
When i put a jumper like this page, it started to work correctly.

https://www.instructables.com/id/Upload-Arduino-Sketch-to-STM32F407-Board/

I will close this issues.

closed 2019-07-24 by nopnop2002

fpistm, 2019-07-24:
Welcome

BennehBoy, 2019-07-24:
Not related to the issue, but be VERY CAREFUL if using the USB connector on
the diymroe boards, they do not have a good mechanical connection to the
board, I would suggest some epoxy and perhaps more solder to better anchor
the connector - I learned this the hard way.

nopnop2002, 2019-07-24:
@BennehBoy 
Thank you for your advice.