stm32duino/Arduino_Core_STM32 #574: Support Forum Unavailable

BennehBoy, 2019-07-25:
Wasn't really sure of the best way to address this @fpistm so apologies if it should not be here!

The support forum is not accessible currently (12:45 BST), it's reporting "Server quota exceeded".

Not sure if the hosting is still Roger's or if the site has transferred.

fpistm, 2019-07-25:
Hi @BennehBoy,
the site has not been transferred.
@rogerclarkmelbourne met some issue with his web hosting company.

BennehBoy, 2019-07-25:
OK, but the important thing is he's aware so will close here.

closed 2019-07-25 by BennehBoy

rogerclarkmelbourne, 2019-07-25:
My hosting company said I'd gone beyond their acceptable use policy.

The account had used over 100Gb traffic this month and was using all the CPU resources on the shared hosting.

They told me I could pay to upgrade to a Virtual Private Server, but said that since I'd already exceeded 100% capacity of the server i was sharing, that a changing to a VPS with them would not be sufficient.

I had 5 domains on the shared server, and I the only one I can bring back online is my personal blog, because if I put anything else back online, I get emails from the hosting company, saying that I am over quota, and the entire hosting gets shut down again.

So its impossible to put stm32duino.com (including the wiki subdomain) back online, because the load from that domain would immediately take all the hosting back offline.
 
I am trying to transfer the domain to ST, but it will not be any of the content, because that's effectively prohibited by the GDPR in Europe and also my local data privacy laws.

BennehBoy, 2019-07-26:
Roger,

I know your free time is limited, but are there particular assets on the
site/wiki which are gobbling up that bandwidth?  Perhaps the images are
being used elsewhere etc, might be worth a quick logscrape/analysis to see
if the b/w can be limited through some pruning/rule implementation?

Just a thought, be an awful shame to lose the site content.

rogerclarkmelbourne, 2019-07-27:
There is a complete copy of the site on archive.org taken on 15th July

https://web.archive.org/web/20190715072346/http://www.stm32duino.com/

Fundamentally, HostGator seem to effectively reduced what I can host on my current plan, because of their acceptable use policy.

I've had to turn off 3 other domains as well, not just stm32duino.com

I'm need to bring 2 domains back online today, if I can. But bringing stm32duino.com back online will be impossible because of the CPU load it generates

TByte007, 2019-07-27:
Ever thought about changing the hosting ? What kind of traffic was the site generating ?

galileo-pkm, 2019-08-23:
If bandwidth is the issue you can get a $5 VPS at Hetzner with 20TB traffic limit. Enable cloudflare to reduce the load.