stm32duino/Arduino_Core_STM32 #575: SPI for STM32F407G-DISC1

nopnop2002, 2019-07-25:
I'm testing SPI support.
With necleo, SPI work fine.
With BluePill & BlackPill, SPI work fine.

|STM32||Device|
|:-:|:-:|:-:|
|MOSI|--|PA7|
|MISO|--|PA6|
|SCK|--|PA5|

With  DIY_MORE_F407VGT, SPI work fine.

|STM32||Device|
|:-:|:-:|:-:|
|MOSI|--|PB5|
|MISO|--|PB4|
|SCK|--|PB3|

But with F4 Discovery, SPI don't work.

I'm using this sketch for test.

Do you have any help.

	#include <SPI.h>
	#include <Wire.h>
	#include <Adafruit_GFX.h>
	#include <Adafruit_SSD1306.h>

	#if 1 // STM32
	// Uncomment this block to use hardware SPI
	//#define OLED_MOSI  11(FIX)
	//#define OLED_CLK   13(FIX)
	#define OLED_DC     PA10
	#define OLED_CS     PA8
	#define OLED_RESET  PA9 // vccでもOK
	Adafruit_SSD1306 display(OLED_DC, OLED_RESET, OLED_CS);
	#endif

	#define NUMFLAKES 10
	#define XPOS 0
	#define YPOS 1
	#define DELTAY 2

	#define LOGO16_GLCD_HEIGHT 16 
	#define LOGO16_GLCD_WIDTH  16 
	static const unsigned char PROGMEM logo16_glcd_bmp[] =
	{ B00000000, B11000000,
	  B00000001, B11000000,
	  B00000001, B11000000,
	  B00000011, B11100000,
	  B11110011, B11100000,
	  B11111110, B11111000,
	  B01111110, B11111111,
	  B00110011, B10011111,
	  B00011111, B11111100,
	  B00001101, B01110000,
	  B00011011, B10100000,
	  B00111111, B11100000,
	  B00111111, B11110000,
	  B01111100, B11110000,
	  B01110000, B01110000,
	  B00000000, B00110000 };

	#if (SSD1306_LCDHEIGHT != 64)
	#error("Height incorrect, please fix Adafruit_SSD1306.h!");
	#endif

	void setup()   {                
	  Serial.begin(9600);

	  // by default, we'll generate the high voltage from the 3.3v line internally! (neat!)
	  display.begin(SSD1306_SWITCHCAPVCC);
	  // init done
	  
	  // Show image buffer on the display hardware.
	  // Since the buffer is intialized with an Adafruit splashscreen
	  // internally, this will display the splashscreen.
	  display.display();
	  delay(2000);

	  // Clear the buffer.
	  display.clearDisplay();

	  // miniature bitmap display
	  display.drawBitmap(30, 16,  logo16_glcd_bmp, 16, 16, 1);
	  display.display();
	}

	void loop() {
	  
	}

fpistm, 2019-07-26:
I've tested SPI and it woks on F407 Disco.
Moreover as it works for the DYIMROE which is also based on F407 there is no reason to not work on DISCO.
Maybe an issue with wiring the wiring or the library you used which use a wrong pin hide in its code....

nopnop2002, 2019-07-26:
Hello.
Is MOSI = PB5?

fpistm, 2019-07-26:
MOSI is PA7

fpistm, 2019-07-26:
Ok I think I understand your issue.
On DiscoF407 the PB3/4/5 could not be used for SPI as they are commented in the PeripheralPins.c
https://github.com/stm32duino/Arduino_Core_STM32/blob/85fd492c15a87048086e7e82318c555fb6410a41/variants/DISCO_F407VG/PeripheralPins.c#L259

Default SPI instance use the PA5/6/7 pins with SPI1.

nopnop2002, 2019-07-26:
I confirmed that SPI is PA5, PA6, PA7.

I understood the cause of my sketch not working.

PA9, PA13, PA14 can not be used as GPIO
Please check with the sketch below.

	static int led[] = {PA9, PA13, PA14};

	// the setup function runs once when you press reset or power the board
	void setup() {
	  Serial.begin(115200);
	  while(!Serial);
	  Serial.println("Blink start");
	  for(int i=0;i<3;i++) pinMode(led[i], OUTPUT);
	}

	// the loop function runs over and over again forever
	void loop() {
	  static int pin = 0;
	  Serial.println(" LED=" + String(led[pin]));
	  digitalWrite(led[pin], HIGH);
	  delay(100);
	  digitalWrite(led[pin], LOW);
	  delay(100);
	  pin++;
	  if (pin == 3) pin = 0;
	}

fpistm, 2019-07-26:
@nopnop2002 
please read the UM and check also schematics of the board.
PA9 is USB VBUS
PA13/14 are the SWD signals.

I close this issue as original issue is answered: SPI worked.

closed 2019-07-26 by fpistm