stm32duino/Arduino_Core_STM32 #577: stm32duino forum "Server quota exceeded"

Andy2No, 2019-07-26:
The stm32duino forum, linked to from stm32duino/Arduino_Core_STM32, has been down for at least a few days.

Visiting the home page, http://stm32duino.com/, gives "Server quota exceeded".

fpistm, 2019-07-26:
Hi,
See #574

Andy2No, 2019-07-26:
I see.  I looked at Issues first, but of course that's only Open issues.

FWIW, closing an issue is a pretty sure fire way of avoiding people seeing it.

Also, having read that, I'm really none the wiser.  Is there a plan to get it running again, or not?

fpistm, 2019-07-26:
I didn't close it.
We try to find a solution but this required some time.

Andy2No, 2019-07-26:
Given the data protection law problems which Roger mentioned, it would seem the only viable solution, in the short term, would be to bail out Roger's account - and pay his hosting company for the traffic.

Before finding that forum, I tried very hard to make any sense out of the main STMicroelectronics website.  It seems to have been designed to bring web browsers to their knees, not for usability.  It's impenetrable, frankly.

I hope STM realises the value of having a forum like that.  I expect by now, most Nucleo and Discovery cards are bought either for educational purposes, or by hobbyists.  Without something like the stm32duino forum available, you're not going to sell nearly as many.

You could, of course, start a new one from scratch, but losing what's their already wouldn't just be a shame - it would be incredibly wasteful.

I expect you have sales and advertising staff who spend more than it would cost to run Roger's forum for the year, in the course of one afternoon.  And, I expect his forum has made a lot more sales for you than any of those people.

TByte007, 2019-07-27:
The ST's site is absolute ... well - I'm sorry - cr4p. And how hard is for a company like ST to just caught few dollars for the hosting ? stm32duino is the best advertising the are getting - ever. I've never heard of them before finding it. ATMEL is recognized all over but they are NOT and stm32duino was changing that. How can they be so slow to get it :(

fpistm, 2019-07-27:
@rogerclarkmelbourne already said he wants simply stop all stuff related to the forum so do not want this solution else this would be easy.
ST and Roger tried to find a proper way to handle GPDR without success.

TByte007, 2019-07-27:
> @rogerclarkmelbourne already said he wants simply stop all stuff related to the forum so do not want this solution else this would be easy.
> ST and Roger tried to find a proper way to handle GPDR without success.

Does that mean that the forum / Wiki would just die ? No ST and no roger taking care of it :) 
Can't they just pay for the hosting (or VPS) and administer it so they dont bother roger (without transferring ownership hence the GDPR issues) ? Sorry for the rambling but the forum / wiki was such an useful resource  it's a shame foe it to get lost.

fpistm, 2019-07-27:
> Sorry for the rambling but the forum / wiki was such an useful resource it's a shame for it to get lost.

I know we do our best to find a solution as mentioned several time.

TByte007, 2019-07-27:
> > Sorry for the rambling but the forum / wiki was such an useful resource it's a shame for it to get lost.
> 
> I know we do our best to find a solution as mentioned several time.

Thank you !

rogerclarkmelbourne, 2019-07-28:
Basically my hosting company pulled the plug, and told me it would not be possible to host this with them, even if I upgraded to their VPS package

The forum was taking 100% of the entire CPU capacity on one of their VPS's which is supposed to be shared by multiple users.

Also the traffic was approaching 100Gb this month, possibly with people making backups of the entire site. But even under normal conditions the traffic is often 50Gb per month.

I've been trying to transfer the site to ST since April, but GDPR and other privacy laws make that impossible.

The cost of hosting the site somewhere else would be thousands of dollars a year, which I'd need to pay for myself. 
Adding advertising to pay for hosting is not a practical option, because I doubt it would cover the hosting costs.

I suspect if ST funded or had any admin rights to the site that would also be in breach of GDPR.

Also, my paid work commitments mean I can no longer spend the 1000+ hours a year administering the site (unpaid)

So since GDPR and Australian privacy laws prevent the site site being transferred, they have effectively killed the site.

Andy2No, 2019-07-28:
What would be the legal situation if an archived copy of the site, without any user account details, was placed somewhere?

For example, it's already on the archive.org "Wayback" machine (at http://web.archive.org/web/20190715072346/http://www.stm32duino.com/), but that's notoriously unreliable to use - pages don't tend to link together properly, many are missing, and of course, the search doesn't work.

To be fair, that archive is working better than most I've found on archive.org, but I expect there's a limit to it - typically, a few clicks in to following a thread, it will say it can't find the corresponding page.

The fact that they are able to do that, suggests that there is no legal impediment to ST hosting its own archive, provided it's just read only pages, with no user account data.  I.e., the site as it would have been visible to someone who wasn't logged in.  That doesn't necessarily need any actual forum engine behind it, just some html pages with the content, and ideally, at least a Google site search box.

As far as I'm aware, there is no reason why any domain can't be transferred too - the domain itself (stm32duino.com) has no personal data associated with it.

FWIW, I'm still researching using STM Nucleos, and the STMF103 in particular.  Just about every search I do, leads me to a dead forum, called stm32duino.com.

For hobby, and education purposes, that forum was clearly THE resource for information about these things, and without it, they would not have had the success they have now.  

If all we're left with in the future, is the main ST Microelectronics site (which is truly terrible), and the Wayback engine, which most people don't know exists, they're probably just going to sink back into obscurity, and return to being something only professional electronic engineers know anything about.

blackrosezy, 2019-07-28:
stm32duino should start using Amazon EC2 to handle thousands of traffic request and about the billing cost, other than get a sponsorship from ST, we can also request a donation from users, something like Wikipedia is doing now.

Andy2No, 2019-07-28:
That's a good point, blackrosezy.  If all user account information, other than the forum display name / username, is overwritten or deleted, there should be no legal difference between the forum, and an archived copy of the forum.  At that point, it would seem that no data protection laws apply which don't already apply to the Wayback Machine.

Incidentally, I have already found several broken links, in the Wayback archived version of the forum - any "Last post" link after about mid March, fails to find a corresponding page.  Though browsing by category works better, it's not any more up to date.  They seem to have grabbed just the home page, a couple of weeks ago, but not a lot of the more recent pages it links to.

TByte007, 2019-07-28:
Well, my HOME internet is regularly going over 150GB per mouth if not more (yea torrents :D)  - so it's not an excuse for them (or any hosting). The server load is and if many people are trying to backup the site it could be high. BUT they should have had a request limiter per IP. I think they are living in 1999 or something or cant keep experienced admins with them.

minimum-necessary-change, 2019-07-31:
Mmm.. big shame.

Thinking on the back of a virtual fag packet, it seems like there is a perfect use case for a open source distributed public forum/wiki, where
* users agree to the public distribution (forever)
* data is encrypted and stored by multiple stations
* data is not available on any individual station in entirety (meaning passwords/emails etc are "striped" across multiple stations)
* server balances the data across the stations with redundancy, (ensuring multiple copies exist), so that the data could be rebuilt centrally. e.g. Server->Station100->"Gives some data to Station3000"
* the server itself would have to be distributed to avoid shutdown.
* etc..

Basically torrenty...

BlackBrix, 2019-08-06:
I can operate the forum and the wiki on my webspace if you want to.

(we can use the domain “stm32duino.com” for it -> but it will work also without using this domain name)

Before taking over the data we can “anonymize” the Forum User-Data (on database-level),
to not getting in conflict with GDPR-Law.

monthly traffic doesn’t matter but how much GB of HDD-space would be used ? (files and databases ?)

I run already some non commercial forums under European GDPR-Law – it’s not a big problem if you know and pay attention on all the necessary GDPR precautions carefully …

Now it’s up to you !

Loud160, 2019-08-07:
Seems I may be a bit late to the party here but rather then talking about what could be done I just went ahead and started working on something. I've been recreating the forum into a read only "Archive" format; at least for the time being. While it's possible what I've been working on could be integrated into another forum even with out having the original forums db I don't think it would be worth the effort. A new forum can be created and people can re-register, what matters is all the previous forum data(and attachments) are now preserved and will still be easily accessible. 

This is still very much a work in progress and is missing a fair amount of content/posts at this time, however I've already gotten a pretty good chunk of data restored and it'll become a lot more complete over the next couple of weeks. 

I'm pretty sure I've gotten most if not all the broken links fixed and the sites internal link structure working again. If you encounter a topic that I have not restored yet rather then giving you an error page it'll just redirect you back to the forum home page. All references/links to other stm32duino.com threads that were linked in topics have also had the URL updated and should load the topic using the new domain name. 

It's far from being perfect but I grew tired of searching for STM32 information and running into broken links on Google. 

http://www.stm32duinoforum.com/

Before anyone jumps down my throat there is ZERO personal data, just screen names. The only way any one could get personal information was if they had access to the original forums data base.

The search function is pretty broken still but if you find a topic on Google there's a good chance you'll be able to just change the URL from stm32duino.com to stm32duinoforum.com and pull up the topic still(assuming its a topic I've already restored). 

I figure it'll take me at least a couple more weeks before I finish restoring the rest of the posts so I'll apologize in advance for any information people are looking for that isn't avaible yet.

uzi18, 2019-08-07:
Maybe Roger is capable to export such static data from original forum to attach into your project?

Andy2No, 2019-08-07:
> I've been recreating the forum into a read only "Archive" format; at least for the time being. While it's possible what I've been working on could be integrated into another forum even with out [...]

That's very useful.  Thanks for all your effort.   Agreed - all that's needed is a prominent link from a new forum to the archived one.

> The search function is pretty broken still but if you find a topic on Google there's a good chance you'll be able to just change the URL from stm32duino.com to stm32duinoforum.com and pull up the topic still(assuming its a topic I've already restored).

Fair enough.  No doubt Google will catch up eventually.  They have been doing quite well at providing cached pages, but of course the links on those don't lead anywhere useful, and their cache entries will expire at some point.

I hope it can be brought up to date, to when it closed down, but I can see you've already got the bulk of it anyway.

I don't really see a problem with STM running their own forum... apart from the fact that they might not put enough effort into it, or be very good at it :)  An independent one would no doubt be better, but I guess these things aren't cheap to run, and it's STM that sees most of the financial benefit.

Loud160, 2019-08-07:
@Andy2No I should be able to bring all content back to with in a couple of weeks of the forum being taken down. 

Creating a new forum on the same URL that I've been recreating the old forum on wouldn't be a big deal and would have very little if any financial  impact on my web hosting even with a high volume of traffic. The site would simply slow down if it was under a heavy load. Creating a forum requires active users/moderators more then anything, spam bots have gotten a lot smarter over the last couple of years. 

IMO this comes down to more of a question what users feel is needed. The time/obstacles it would take to import all the threads from the old forum into a new forum even on the same software would be nightmare. A fresh forum is extremly simple to setup and there are a number of new forum formats allowing things like Facebook and Google logins so new users don't need at create a account, give out personal information and have yet another password to remember. It just comes down to what people want/need. I've found a number of spin offs of the original stm32duino and they all seem pretty dead/inactive. It may be new forums are hard to find or the community may have just split into a number of small groups. If that's the case, creating yet another forum would do more harm then good.

Andy2No, 2019-08-08:
> I've found a number of spin offs of the original stm32duino and they all seem pretty dead/inactive. It may be new forums are hard to find or the community may have just split into a number of small groups. If that's the case, creating yet another forum would do more harm then good.

Are you talking about specifically STM32 MCU related forums?  I haven't been able to find any that are specific to that topic.  I see some blogs and mini sites using the term stm32duino, but I haven't yet found a forum comparable to the former stm32duino.com forum.

There are related topics in arduino.cc, of course.  Personally, I find that a toxic environment, and largely avoid visiting it.  I've come across other people expressing the same view.  Also, the fact that its main focus is on AVR and, to a lesser extent, SAM based arduinos, means it's harder to search for things relating to STM32s.

Personally, I think a forum that's just about the STM32 Nucleos, Discoverys and related boards built with STM32 MCUs, like the Maple Mini or the "Blue Pill", would be useful to a lot of people.  

I don't think there's any doubt that it would help ST Microelectronics sell more boards, and raise the profile of their MCUs, leading to more people using them in new designs - and more people buying them for learning or hobby purposes.

I understand that, for Roger, the original purpose of the forum is over.  I'm finding it harder to understand why ST Microelectronics aren't falling over themselves, trying to get something similar working again.

SuperUserNameMan, 2019-08-08:
@Andy2No : what do you think of https://community.st.com/ ?
I've just noticed it when I had to install STM32CubeProgrammer  a few days ago.

Andy2No, 2019-08-08:
> what do you think of https://community.st.com/ ?

I'm not sure if I've seen that before.  Oddly, it doesn't seem to have come up when googling STM32 problems, or just trying to research products - or not that I've noticed.

The first thing that strikes me is that the user interface is woefully inadequate - it's just one page that can grow a little each time you press "View more".  There are only two sort options, with only a subtle distinction between them - recent activity or recent posts.

No one is going to look very far through that.  It seems like an efficient way of burying information, rather than a way of making it available.

It needs organising into sections, for a start.  At least it has a search box, but as far as browsing is concerned, it's a non-starter.

Andy2No, 2019-08-08:
... and for specifically arduino based STM32 purposes, we have bigger problem:

Posts tagged as arduino : 48;

https://community.st.com/s/topic/0TO0X000000BWWSWA4/arduino

Results when searching for the word arduino... 3+?

The main focus appears to be Cube MX.

blackrosezy, 2019-08-08:
I really hope the new forum will be using the modern https://github.com/discourse/discourse as a platform (open source). In my opinion, we should keep both forum separately 

**community.st.com**
Official forum for ST products. A bit formal, usually for people in electronic industries looking for support or help

**stm32duino.com**
This is the place for hobbyist, electronic enthusiast and fun place for students

noweare1, 2019-08-13:
I aplaud ST for putting out the resources for their arduino core and tasking a talented engineer fpistm with support.  ST does some awesome stuff but are not  that great at doing websites/forums.  I would rather ST not put  STM32duino forum with the rest of their  forums. It is rather a clunky format and it is mostly commericial based.

noweare1, 2019-08-13:
I thought I saw/read something about ST taking over the forum later this year independent of this happening.  Maybe the GDPR issues were a roadblock.  Kind of funny, GDPR causing the forum to not exist and facebook gets away with giving all types of personal information away with no problem.

geosmall, 2019-08-14:
+1 for letting the old forum go and starting new non-commercial forum around this core.   STM32duino had an array of incompatible cores, causing much confusion for newcomers and wasted motion for moderators.

Andy2No, 2019-08-14:
> +1 for letting the old forum go and starting new non-commercial forum around this core. Separate STM32duino had an array of incompatible cores, causing much confusion for newcomers 

That is a good point, though it wasn't the thing that confused me most, reading and searching that forum.  I was more confused by the threads about the forum closing, then finding some references to it being taken over by ST, which I'd hoped was the end of the problem.

I quickly worked out that some of it applied to the the Leaflabs Maple / Maple Mini STM32F103 core, but not to the official cores.  At worst, I could have devised simple tests to tell if something still applied - like the differences in pinmode(), for example.

It's a fair point though.   Perhaps the banner at the top of the Arduino for STM32 (archived forum) could point that out - that older posts relate to the Leaflabs cores, and might not apply.

I agree with the other points too - ST have clearly put a lot of work into making the arduino cores for so many STM32 variants, and fpiSTM's involvement in that forum was clearly very useful.  I've seen a lot of replies by him.  He answered the one support question I remember asking, before the forum closed down, and was very professional - clearly knowing a lot about the subject.  I hope he can be involved in a new forum.

And, yes,  I really don't think the main ST support forum is well thought out.  They would do well to start again, along the lines of a mainstream forum - I can suggest a few examples of forums that work well, if needed.  A forum engine similar to phpBB is perfectly good enough.  

Personally, I think a simple layout, that would work in a ten year old browser, is far better than something that tries to use lots of "modern" tricks - simply because it will work consistently on more or less anything, and not make people lose patience, waiting for pages to finish loading.

An independent forum would be better, in some ways, but at some point, I'd expect to see it struggling financially (held hostage by a hosting company, who keep jacking up prices, for something that can't be moved elsewhere), or maybe taken over by a small clique of egomaniacs, who make it a miserable place to visit... again, I could give examples.

Either way, a dedicated stm32 for arduino forum of any kind is better than none.

Loud160, 2019-08-19:
About 2 weeks ago I posted a link to an archived version of the STM32duino forum I was working on. Since then I've gotten most of the content fixed up and working in a "read only" version. I'd guess I've got about 90-95% of the forum posts restored, I'm still working on getting the wiki working again, while the current wiki on github is very useful it doesn't have the simplicity that the old one had IMO. 

I've added a discussion forum using an open source platform called Vanilla forums that's a bit more modern and far less rigid then phpBB. But before I go and invest a bunch of time setting up things like Google and Facebook login's(so users don't have to register) I'd like to make sure a new forum would actually get used. 

I already run a number of other websites and my hosting is currently pre-paid though 2022 so the only cost of any kind is the domain renewal.  I've been with my web host for years, have root access and everything was installed with out any type of auto script. I can pack up everything and re-host it with just about any company and not loose a thing.....so there is no need to worry about anything being held hostage by a web host. 

I'm doing this because stm32duino community helped me more then any other resource on the web and I feel it's important to keep it alive. 

The "New" forum can be found at https://stm32duinoforum.com
The "Archives" can be found on the "New" forum or directly at https://stm32duinoforum.com/forum/index_php.html

I'm open to design and layout suggestions, if you don't like what I've setup... lets talk about what you would like to see changed.

Andy2No, 2019-08-19:
That's great.  Thanks, Loud160.

One thing that strikes me is that there probably should be a separate section for Nucleo boards, rather than lumping them in with "Any other STM32 based boards including Nucleo boards ".  There are quite a few different ones, and all very reasonably priced - plus they seem to all be supported by the official core...  Maybe there are exceptions, but maybe ST plan to address them if there are.

It seems to me, those are a better choice than cheap generic boards based on the same CPUs.  They all have an ST-Link built in, for one thing (which is a second STM32 MCU), so they work with the Arduino IDE, and the stlink utility... and maybe the Texane one, for Linux.  That's a big advantage.  

They're very cheap, for what they are, so it doesn't seem right to treat them the same as a cheap Chinese board with a single STM32 MCU on it, and very little else, implying that they're more or less equivalent.

Also, I don't see a category that the Maple Mini, or their clones, fit into, other than "All other", though those are supported by the official core too, cheap, and popular.  

I'm not clear whether the "Blue Pill", based on a version of the STMF103 with less Flash memory, is supported by the standard core or not - because I haven't tried those.  Even so, they seem to have been a popular choice, so maybe they deserve their own category or sub category too?

raghur, 2019-08-21:
@Loud160 - Amazing work and thank you. I'm sure that a number of community members feel the same  about keeping stm32duino alive.

BTW, given that the forum copies have been around for 2weeks+ is there a reason why google isn't finding it? I even tried with `site:stm32duinoforum.com` and nothing shows up. Is there something blocking the site from being indexed (robots.txt etc) and/or could you submit a sitemap xml at GOogle webmaster console and give big G a nudge?

Loud160, 2019-08-21:
@raghur 
that was due to my initial upload that was rather incomplete and did not have https enabled at the time. If you slightly alter what you have it loads 
https://stm32duinoforum.com/forum/viewforum_f_28.html

once I get the rest of the files imported and finish fixing a few more broken links I will submit an xlm sitemap for indexing. 

Some help would be seeing if the forum works for you and if you have any issues. One person has said its almost unusable for them but I've had several friends try the forum and none of them had an issue.  I'm not stuck on keeping the current forum format, its just a very simple solution that also has open source code avaible and is easy to pack up and move if necessary at some point down the road.  I'd like to try and stay away from phpBB just due to whats required on the back end for maintenance and updates.

raghur, 2019-08-21:
> Some help would be seeing if the forum works for you and if you have any issues. One person has said its almost unusable for them but I've had several friends try the forum and none of them had an [...]

I signed up and took the poll. I also did not have any issues typing messages on the forum as the other poster indicated. maybe a one off.

pinchies, 2019-08-21:
Let me know if you need any help with vanilla forum config/troubleshooting - I use it for jgauroraforum.com too! Joining now :D

pinchies, 2019-08-21:
You can use the " Add Menu Item " addon to allow you to add a link to the archive to the main menu. You might want to add a "custom google search" option to allow the old archives to be searched too.

me21, 2019-08-21:
There's already a new forum at https://mcu.selfip.com. Maybe it's better to place the link to that one in addition to/instead of creating another one?

BlackBrix, 2019-08-21:
just for information:
does the search function in the archived forum work ?
( I can not find my threads & postings there )

If not: how can one find content ?

is it intended to use the google search syntax with:
`site: stm32duinoforum.com mySearchword`
?

BlackBrix, 2019-08-21:
> 
> 
> There's already a new forum at https://mcu.selfip.com. Maybe it's better to place the link to that one in addition to/instead of creating another one?

I second that.
The "experienced" users are already there like stevestrong, mrburnette, etc...

Andy2No, 2019-08-21:
> There's already a new forum at https://mcu.selfip.com. Maybe it's better to place the link to that one in addition to/instead of creating another one?

I missed seeing that.  It looks well established, and it's based on a forum engine I'm used to.

I appreciate what you've done, @Loud160, but the Vanilla forum was barely useable for me, due to the typing lag.  Although I'd like to think I'm unique, the chances are, I'm not.  There must be other people it would affect too.  

It could be browser specific (I'm using Waterfox, a little bit out of date, for various reasons), and it might even be plug-in specific - but if so, there will be plenty of other people using the offending plugin.

I tried googling about a typing lag, with the Vanilla forum, but didn't get very far - mostly because the name is so vanilla, I expect :)  If they'd called it something like p45kLambda, I'd no doubt have found more relevant results.

The archive of the old site is great, but I agree with @BlackBrix about the other forum - it looks established, and I can use it, so I'm going with that one.  A prominently placed sticky there, to point to your archived site, would be good - maybe at the top of the "Arduino for STM32" section.

Andy2No, 2019-08-21:
> is it intended to use the google search syntax with:
> `site: stm32duinoforum.com mySearchword`

I find the space after site: makes a difference.

site: stm32duinoforum.com nucleo
- 6 results

site:stm32duinoforum.com nucleo
- 118 results

I'm not sure about the broken links though.  Maybe google is still using out of date cached data?

BlackBrix, 2019-08-21:
> The archive of the old site is great, but I agree with @BlackBrix about the other forum - it looks established, and I can use it, so I'm going with that one. A prominently placed sticky there, to point to your archived site, would be good - maybe at the top of the "Arduino for STM32" section.

I took a quick first look at the new forum (https://mcu.selfip.com),
but there is are forum sections anymore about "Rogers core" and "Rogers Bootloader",
or am I wrong ?
( "Rogers core" =  https://github.com/rogerclarkmelbourne/Arduino_STM32 )
( "Rogers Bootloader" = https://github.com/rogerclarkmelbourne/STM32duino-bootloader )

It would be great to have these too,
even when planning to migrate to the new official ST-core ( https://github.com/stm32duino/Arduino_Core_STM32 ),
just for referece / learning / old  projects / ...

Andy2No, 2019-08-21:
> I took a quick first look at the new forum (https://mcu.selfip.com),
> but there are no forum-sections anymore about "Rogers core" and "Rogers Bootloader",
> or am I wrong ?
> ( "Rogers core" = https://github.com/rogerclarkmelbourne/Arduino_STM32 )
> ( "Rogers Bootloader" = https://github.com/rogerclarkmelbourne/STM32duino-bootloader )
> 
> It would be great to have these too,
> just for reference / learning / old projects / ...

Agreed.  We should petition for a sticky about that too.

Andy2No, 2019-08-21:
> 
> 
> > The archive of the old site is great, but I agree with @BlackBrix about the other forum - it looks established, and I can use it, so I'm going with that one. A prominently placed sticky there, to point to your archived site, would be good - maybe at the top of the "Arduino for STM32" section.
> 
> I took a quick first look at the new forum (https://mcu.selfip.com),
> but there are no forum-sections anymore about "Rogers core" and "Rogers Bootloader",
> or am I wrong ?
> ( "Rogers core" = https://github.com/rogerclarkmelbourne/Arduino_STM32 )
> ( "Rogers Bootloader" = https://github.com/rogerclarkmelbourne/STM32duino-bootloader )
> 
> It would be great to have these too,
> even when planning to migrate to the new official ST-core ( https://github.com/stm32duino/Arduino_Core_STM32 ),
> just for reference / learning / old projects / ...

I've registered and posted in that forum, asking for stickies to the archive, and Roger's stuff:

https://mcu.selfip.com/viewtopic.php?f=11&t=129

Feel free to chime in :)

Andy2No, 2019-08-21:
> The "Archives" can be found on the "New" forum or directly at https://stm32duinoforum.com/forum/index_php.html

I see the trick now, in following a broken google link to the old forum, is to change a link like

https://www.stm32duino.com/viewtopic.php?t=3493&start=230

to

https://stm32duinoforum.com/forum/viewtopic.php?t=3493&start=230

I preferred the old way, but I guess I'll get used to that.  Hopefully google will stop sending us to the dead site, eventually.

Andy2No, 2019-08-25:
> There's already a new forum at https://mcu.selfip.com. Maybe it's better to place the link to that one in addition to/instead of creating another one?

There's also a more easily remembered url for that forum, which is

https://www.stm32world.com

DaLiV, 2019-08-25:
seems restored data is outdated ...
as sample : https://www.stm32duino.com/viewtopic.php?t=501&start=30 - no such topic can be viewed in any of "restored" points..
are not possible to get back data from last point when error "quota exceed" started ?

Andy2No, 2019-08-25:
> https://stm32duinoforum.com/forum/viewtopic_f_18_t_501.html - give result as forum list ... not the topic ...

The archive is missing many of the more recent pages.  It's set to redirect to that main page, when you click on a link to one that's not there yet.

> are not possible to get back data from last point when error "quota exceed" started ? - or simple back up database + files via ftp "as is" and restore on new location ?

Legally, the entire database could not be transferred because it contains users personal details, which are protected by GDPR.  In theory, it could be done if all those details were blanked out.  The target / new host site would have to be set up with a version of phpBB that matched well enough to cope with the database.

It could probably be done, but I expect it would involve a lot of work for Roger, and he might have to go to some lengths to cover himself legally.

> new link stm32world.com = Total posts 1157 • Total topics 135 - totally unusable

The new forum works fine.  There's not much there yet, but there are some knowledgeable people, a lot of them from the old forum, and it's an active, growing, forum.  It's what there is.  I suggest just accepting it and joining in.

There are a lot of empty topics because it started with the same structure of the old one - same main sections and sub sections.  It won't necessarily stay like that.

> even instead of stm32duinoforum.com = Total posts archived 54426 • Total topics archived 4544 /with topics+posts inaccessible - see begin of message/

I've found if I see a link to a thread that's dated at least a few months ago, that link works, and so do ones I click from there - e.g. turning pages, referencing other threads and so on.

Unfortunately, the top level section pages / indexes tend to be newer than the pages they link to.  I'm not sure when the cut off date is, but anything before spring seems likely to work.  Anything in the last couple of months, doesn't.

It's a shame, but we may just have to settle for what we've been given.  PeteS (Loud160) has put a lot of effort and resources into giving us as much ass he could from the old forum, and I'm grateful.  I thought we'd just lose it all for ever.

The newer forum was set up before the old site closed, so it stayed very quiet for months - so quiet, I didn't find it when I was googling for one.  It's now waking up though, and that's a good place to go to ask questions, or post about your projects.

Loud160, 2019-08-25:
@DaLiV 

> All restored data is outdated ...
> as sample : https://www.stm32duino.com/viewtopic.php?t=501&start=30 - no such topic can be viewed in any of "restored" points..
> https://stm32duinoforum.com/forum/viewtopic_f_18_t_501.html - give result as forum list ... not the topic ...
> are not possible to get back data from last point when error "quota exceed" started ? - or simple back up database + files via ftp "as is" and restore on new location ?

Many topics from archived sites are incomplete, when a crawler collects data it does not crawl every single page on the site, it will look at top level pages and then select pages at random to collect for archiving. Places like the Wayback machine attempt to stitch various pages back together to recreate a complete looking site, however not every page will come from the same date the site was crawled and this creates breaks in pages and links. For example lets say the topic you listed was crawled on 2-23-17 and in the forum index it was 7 pages deep in the category it was posted in and was the third topic from the top of the threads listed on that page. The forum was crawled again on 6-11-18 and the same page was found but this time the thread had been continued and more posts had been added and it was now on the front page in the category it was listed in. 

The "newer" version of the page is obviously the one you would want but by using this page and trying to included it with other pages that were crawled on the original date 2-23-17 you now have an inconsistency on where this new version of a topic should be placed. Do you put it back 7 pages deep to keep the file structure compete for the 2-23-17 crawl and just add in the new posts? Well by doing so you also bring in all the links that would be associated with the newer version of the page including the site index  that had it on page 1. 

How this plays out is when you find the link to the original topic on page 7 and click on it, it loads the newer version of the page. The only way to get back to the place you found the topic on page 7 is by using your browser back button. If you clicked on the page index for the page your viewing you would be taken to page 1 and linked to the most recent version of page one.  Where this really creates issues is that when it takes you to page 1, its not going to be page 1 that was crawled at the same time the topic your viewing was taken, it will be some version of the page that's newer then what you were just viewing where once again the topic you just looked at is no longer on the first page. 

This of this like the movie back to the future 2 where an alternative time line was created by something that happened in the past and affected the current reality. Every time a page is crawled it creates a slightly altered version of the entire site and the only way to get past this is by going further back into the past. Unfortunately in this case going back to older dates also means loss of content. 

So the simple answer is no, there is no way of getting back  "Everything" as it was with out the original forums data base. However a mostly complete time line can be recreated by manually rebuilding the forum page by page updating every link....on every page...to everything.... so things work as you would expect. The average page will have something like 15-20 active links that need to be changed in order for the page to function correctly with other pages when the dates they were archived is not the same. It is a very slow process that's mind numbing boring and because of this I've only focused on important links that break site navigation.  Every page will still have a number of "Dead links" and what  I've done to address those is just redirect back to the "Home Page" for the time being. 

Also keep in mind while the archives on stm32duinoforum.com may "Look" like its still a forum, it isn't. Every page you see is simply HTML and CSS that makes it look like the forum did. I've done some experimenting with a bit of Java script trying to create a way to search the pages but it did not turn out very well. 

Links your finding on google that you would expect to take you directly to a topic and instead take you to a page are the same issues I ran into while trying to recreate the search function. The way the HTML pages have been built and indexed is not playing nicely with any type of search engine right now either internally or externally. 

Any page that would take you to a "Log In" screen is broken or will take you to a dead end page. This is because when the site was crawled the bot was not logged in and was unable to get past a log in screen and the data on the other side of the log in screen is data that would have been apart of the forums DB that can not be accessed or recovered outside of the original forum software/web hosts SQL DB's. 

@Andy2No 

> I see the trick now, in following a broken google link to the old forum, is to change a link like
>
> https://www.stm32duino.com/viewtopic.php?t=3493&start=230
>
> to [...]

I needed to move the archives off the base URL while I'm working on restoring things. Every time I add something there is a chance it will break the entire archive.....or at least break certain parts of the archives and I may not be aware of thee broken pages when it happens or may not have time to "Undo" what ever I did that broke the archives. To make sure that there is still a mostly working version of the archives up at all times I create [... cut ...] tacted me asking to setup something. I'd have been happy to set something up and let the community run the forum them self. All of this is just being run with unused resources so I really didn't care if a group of people wanted to run a forum for free on the back of my server.

 I have server resources provisioned in such a way that anything STM32duino related won't impact the things I use my server for.  This is one of the advantages of long term pre-paid web hosting with a re-seller account :)

Andy2No, 2019-08-25:
I understand that the problem I had using Vanilla could probably have been fixed - or more likely, I'd have had to get used to using a different browser, just for that.    It just made more sense to me to join in with a forum that's already established, that some people have been using for a few months.

I don't know anything much about phpBB, other than having used forums based on it (as a user, not a provider), but I do have some experience with PHP.  Manipulating databases is most of what it does, so it would be quite simple to write a script to blank a column of a table, for example, or overwrite it with random data - to someone who's used to writing PHP, that is.  For a site admin who simply uses it as an off the shelf package, it would be a lot more work to find out how to do that.

Typically what you have on a PHP based website is a mySQL database.  You also normally have phpMyAdmin, or similar, which lets an admin or developer do things to the database without writing any PHP - e.g. search it, browse it (more useful for smaller databases), export it or import to it (e.g. back the whole thing up as a text file, if it's not too big) or execute SQL commands / queries - which are very powerful, so you have to be careful, but that would let you get the same results just with a few lines of SQL.

In Roger's position, my worry would be being sure I'd done it properly.   If there was a danger of being sued or prosecuted for a mistake, I'd want to be very sure, before I handed it over.  It would be a lot safer to just delete the whole thing, and tell the hosting company to delete any backups - at which point, they would be the only people who could be held liable.

But, yes, that forum may die too one day, and it might not be all that far in the future.  It's certainly worth having more than one.

raghur, 2019-08-26:
What if the original is brought up on a bare IP by Roger and then crawled using wget etc? No gdpr problems and you'd get a complete history. Host it on http so that it's called aggressively and doesn't cost much? I realize that this will require Roger's support, but seems less work+ more complete than trying to reverse engineer an archive

rtek1000, 2019-10-05:
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/700

adammunich, 2019-11-18:
Howdy folks,

I built this forum and wiki website some time ago to document my friend's projects, and host it myself on a very good server with gigabit symmetrical connection. We could make a group on there for stm32duino?

https://civilpedia.org/s/#w_1

The wiki editor would be very well suited for documentation, I think.

fpistm, 2019-12-18:
Hi all,

https://www.stm32duino.com/viewtopic.php?f=61&t=8

😉

closed 2019-12-18 by fpistm