stm32duino/Arduino_Core_STM32 #580: Finally Blue Pill R10 has been corrected

SRQJohnson, 2019-07-27:
Just a quick Heads up, as STMduino seems to no longer exist.   The Chinese have finally corrected the dreaded R10 resistor on the Blue Pill.  These boards are the variety which has a rectangular white plastic reset button rather than the round metallic one.  I have tried to put in a photo, but have just started here.

A package from ebay China arrived today with 1.5 K resistors.
[image]

TByte007, 2019-07-29:
You are right - I just checked. It's not '103' anymore but something like '01C'  (not 125 like yours).

fpistm, 2019-07-30:
Hi @SRQJohnson 
thanks for the info, anyway this is not the place for this kind of info.
I know the forum is currently down and this is the main issue.
You can add a Wiki page on the BluePill:
https://github.com/stm32duino/wiki/wiki

closed 2019-07-30 by fpistm