stm32duino/Arduino_Core_STM32 #583: STM32 I2C driver "repeated start" issue

jmchiappa, 2019-07-29:
Hello,

I'm trying to connect to the MMA8451 accelerometer. This chip requires an I2C driver with repeated-start feature enabled to discuss with.
I've  a NUCLEO-F411RE, lib 1.6.1 and no repeated start is generated due to some low level limitations.
Let me explain: 
XferOptions variable is always set to `I2C_NO_OPTION_FRAME` in the `HAL_I2C_Transmit_IT`. So, once the Xfer is completed, it fires `I2C_MasterTransmit_TXE`. As the following condition is always false (stm32f4xx_hal_i2c.c):
`if ((CurrentXferOptions != I2C_FIRST_AND_LAST_FRAME) && (CurrentXferOptions != I2C_LAST_FRAME) && (CurrentXferOptions != I2C_NO_OPTION_FRAME))`
a STOP will be generated (line 5025):
```
      /* Disable EVT, BUF and ERR interrupt */
      __HAL_I2C_DISABLE_IT(hi2c, I2C_IT_EVT | I2C_IT_BUF | I2C_IT_ERR);

      /* Generate Stop */
      SET_BIT(hi2c->Instance->CR1, I2C_CR1_STOP);

```
Does someone can confirm ?
How could we override XferOptions by setting to `I2C_LAST_FRAME_NO_STOP` ?
Ok, I understand that it's not directly related to the framework by itself, but I observed that I2C driver is really different regarding the variant.
In other word, by using `TwoWire::endTransmission(false)`, it should not send a STOP condition (instead of UNUSED(endStop) ).

Thank you,
jean-marc

fpistm, 2019-07-30:
Hi Jean-Marc,
thanks for the report. Will try to investigate when got a time slot. (will be on vacation soon)
Maybe @LMESTM will have a look when he came back from vacation.

jmchiappa, 2019-07-31:
Hi Frédéric, 
Thank you for your reply.
After deeper investigation, it seems that `HAL_I2C_Master_Seq_Transmit_IT` is more relevant to generate a 'repeated-start' as XferOptions is passed in argument list.
Curiously, this function is called `HAL_I2C_Master_Sequential_Transmit_IT` in F0,F1,F2,F3 series but it's correctly bind in legacy file. 
I'll have a try as soon as I can (in holidays soon too).

Enjoy your holidays !

fpistm, 2019-07-31:
Thanks, enjoy too.

The function has been renamed in newer STM32CubeFW Release for or MISRA-C 2012 compliancy:
See the [Release_Notes.html](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32F4xx_HAL_Driver/Release_Notes.html)

I guess at term all series will be renamed.  The next cube update add this too for some of this series, see #566.

fpistm, 2019-09-13:
Fixed by #590

fpistm, 2019-09-23:
During non reg tests for 1.7.0 release found issue with STM32 F2 and F4 related to #590. 
So I've reverted it to perform further investigation.

PR #748 (2019-11-05): I2C Fix
* Added the repeated-start feature in I2C driver (prviously done thanks #590).    
   Fixes #583
* Fix error management in Read/Write
   When NACK is received by master, instead of immediate re-transmission
   by twi driver, error is reported. Up to application to manage re-transmission.
* Review errors management to be aligned with Arduino [Wire.endTransmission()](https://www.arduino.cc/en/Reference/WireEndTransmission) API Reference:
  - 0:success
  - 1:data too long to fit in transmit buffer
  - 2:received NACK on transmit of address
  - 3:received NACK on transmit of data
  - 4:oth [... cut]
commit f56b7a6b (2019-11-05): Added the repeated-start feature in I2C driver (#590)

Fixes #583
closed 2019-11-05 by fpistm