stm32duino/Arduino_Core_STM32 #584: How to use SerialEvent ?

arminsalcin, 2019-07-29:
Hello to all , 

So my question is how to use it , i want to use it with Serial1 but it doesnt work if i regulary define it in my code. So how i need to do it properly , and can i use Serial for debugging while using Serial1 for communicating with sensor and using SerialEvent1 .

If anyone can write some small simple example code that will be nice 
- I'm using stm32f103C in board "Generic STM32f103c"

Thanks !

fpistm, 2019-07-29:
Hi @arminsalcin 
By default, if USB CDC is not enabled is not defined. `Serial1` is the same than `Serial`.
I guess you use USB CDC so Serial is `SerialUSB`.
To enable the `SerialEvent1()` for `Serial1`, add a `build_opt.h` file alongside your main .ino file with: `-DENABLE_HWSERIAL1`. This will define the `Serial1` instance using the first `USART1` instance found in the P`eripheralPins.c` of your variant.

This is in the wiki:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

arminsalcin, 2019-07-29:
Okay , just to clearify im using bluepill board and programm it via ftdi connected to (PA9 , PA10) , and my sensor is connected to (PA2 , PA3) , in my code i use Serial1 to communicate with sensor and it work. 

So when i add `build_opt.h` like this :

``
#include <build_opt.h>
``
I get error no build_opt found..

So what can be problem ?

arminsalcin, 2019-07-29:
Ok i think i realize what is problem with in include if i choose from STM32 board selected from submenu it compiles okay if i select from stm32duino.com it doesnt want to compile 

[image]

So i dont understand this thing  `-DENABLE_HWSERIAL1` how to write this in arduino ide ?

fpistm, 2019-07-29:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

arminsalcin, 2019-07-29:
Ok i manage to do it , so after this embedded i just use Serial1 as i would regulary , example Serial1.begin(9600)

fpistm, 2019-07-29:
> Ok i manage to do it , so after this embedded i just use Serial1 as i would regulary , example Serial1.begin(9600)

Right

arminsalcin, 2019-07-29:
Thank you after i make all to work i will leave example here and close issue

fpistm, 2019-08-02:
Hi @arminsalcin 
before leave I made some sweep.
I think your question is answered. Feel free to update Wiki with your example if you want.

closed 2019-08-02 by fpistm

jonghaboy, 2020-03-11:
i have followed this example but i get  "expected unqualified-id before '-' token

can anyone help?
the build_opt.h file only has the line "-denable_hwserial1" right?

jonghaboy, 2020-03-11:
by the way i am using stm32f103c series

fpistm, 2020-03-11:
@jonghaboy 
Please use the forum for support.
Anyway your syntax is not correct... it is case sensitive.

jonghaboy, 2020-03-13:
> @jonghaboy
> Please use the forum for support.
> Anyway your syntax is not correct... it is case sensitive.

-DENABLE_HWSERIAL3
-PIN_SERIAL3_RX=PB11
-PIN_SERIAL3_TX=PB10

this is my code in build_opt.h
 it does not work

use forum to write a new case you mean?