stm32duino/Arduino_Core_STM32 #587: the enum constants is not possible to use in preprocessor conditions

linvinus, 2019-07-31:
Because of most constants (like pins Px[0-16], AFIO_*,  STM_PIN_* )  defined as enum, it is not possible to use them in preprocessor conditions.

Because of that Marlin port to STM32  have unpredictable behavior 
https://github.com/MarlinFirmware/Marlin/issues/14777
https://github.com/MarlinFirmware/Marlin/issues/14711
https://github.com/MarlinFirmware/Marlin/issues/14732
 
In contrast the arduino port to LPC176x, where  pins and other constants is defined as macro definitions https://github.com/p3p/pio-framework-arduino-lpc176x/blob/b3635e53cbd2b6481b207eda668e47dbe712f1e7/cores/arduino/pinmapping.h

Marlin developers has make workaround to resolve this problem https://github.com/MarlinFirmware/Marlin/pull/14785/files

But i believe that problem must be resolved on stm32duino side, i think implementation similar to LPC176x port will be nice.

Similar problem already was discussed earlier, for example here https://github.com/stm32duino/Arduino_Core_STM32/issues/140#issuecomment-339595248

fpistm, 2019-07-31:
Hi @linvinus 
I already think about that and have some private request about that.
What is the status of the STM32 porting in Marlin?
If I'm not wrong this core is integrated in the 2.x version?

linvinus, 2019-07-31:
>What is the status of the STM32 porting in Marlin?

In this year several new the electronic board,based on SMT32 , for 3D printer  was released.

For example:
SKR E3 DIP V1.0 (STM32F103)
SKR E3 mini (STM32F103)
FYSETC Cheetah Board (STM32F103)

So marlin port for STM32 will be more and more popular.

>If I'm not wrong this core is integrated in the 2.x version?
Yes marlin 2.0 (bugfix-2.0.x branch) is already supporting STM32F1 and STM32F4 based electronic boards, but currently it in alpha state, mostly working but have some problems, not all of them may reveal at compiling time.

Official status is here https://github.com/MarlinFirmware/Marlin/issues/14345

fpistm, 2019-07-31:
OK, thanks for the feedback.

hasenbanck, 2019-07-31:
@linvinus The issues you have linked seem to affect the STM32F1 HAL (HAL_STM32F1), which uses another [Arduino Core](https://github.com/rogerclarkmelbourne/Arduino_STM32). Are you sure that it also affects the STM32 HAL? (HAL_STM32) Marlin right now has THREE STM32 HALs. I and some other developers added a generalized STM32 HAL, but not all boards made the switch yet.

linvinus, 2019-07-31:
@hasenbanck Oopse, you are right!

marlin use platformio which have ststm32 module, 
According to platformio documentation https://docs.platformio.org/en/latest/platforms/ststm32.html
by default  stm32duino is used, so i decide to report here.

I'm sorry, my mistake!

But any way the same problem is present here.

hasenbanck, 2019-07-31:
Do you simply assume that the problem is there (when using this core) or
did you test it? The issue you linked also seem to fix the problem inside
Marlin (referenced PR).

linvinus, 2019-08-01:
>Do you simply assume that the problem is there (when using this core) or
did you test it?

As I said above, I did not know that Marlin uses three different HALs, so no, I have not tested stm32duino as a core yet.

fpistm, 2019-08-01:
Well, until a clear the status. 
I will wait.

@hasenbanck IIRW you told me Marlin waited for a Hardwer Timer library, FYI the PR was submitted.

xC0000005, 2019-08-01:
I think this has already been fixed. Most boards add a set of #defines similar to this: https://github.com/stm32duino/Arduino_Core_STM32/blob/85fd492c15a87048086e7e82318c555fb6410a41/variants/MALYANM200_F070CB/variant.h#L43  

And that hardware timer library should make it possible to support most ST boards from one HAL. Good stuff.

fpistm, 2019-08-02:
Ok, looking deeply in issues referenced by @linvinus.
Pins PXn are already a define not enum. 
This was covered by #309 thanks @ghent360 for Marlin and merged thanks  #356.
This new issue is probably linked to @rogerclarkmelbourne core: [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32)
ex:
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/a3a56866505eeca4b49fc435700dcd4bd5540c8c/STM32F1/variants/generic_stm32f103c/board/board.h#L79-L83

So I close this one as already solved.
Thanks all for your feedback.

closed 2019-08-02 by fpistm