stm32duino/Arduino_Core_STM32 #591: Serial.print using BLUE_F407VE_Mini

nopnop2002, 2019-08-04:
I tried this sketch:

When i used DIYMROE_F407VGT/BLACK_F407VE/BLACK_F407VG/FK407M1, Serial.print work fine.

[image]
[image]
[image]

But when i used BLUE_F407VE_Mini, Serial.print don't work.

[image]

What is wrong??
[image]

	#if defined(ARDUINO_DIYMROE_F407VGT) || defined(ARDUINO_BLACK_F407VE) || defined(ARDUINO_BLACK_F407VG) \
	 || defined(ARDUINO_BLUE_F407VE_Mini) || defined(ARDUINO_FK407M1)
	#if defined(ARDUINO_DIYMROE_F407VGT)
	#define MODEL "DIYMROE_F407VGT"
	#endif
	#if defined(ARDUINO_BLACK_F407VE)
	#define MODEL "BLACK_F407VE"
	#endif
	#if defined(ARDUINO_BLACK_F407VG)
	#define MODEL "BLACK_F407VG"
	#endif
	#if defined(ARDUINO_BLUE_F407VE_Mini)
	#define MODEL "BLACK_BLUE_F407VE_Mini"
	#endif
	#if defined(ARDUINO_FK407M1)
	#define MODEL "FK407M1"
	#endif
	#define S1 "PA9"
	#define S2 "PA2"
	#define S3 "PB10"
	HardwareSerial Serial2(PA3, PA2);
	HardwareSerial Serial3(PB11, PB10);
	#endif

	void setup() {

	  Serial.begin(115200);
	  if (S1 != "") Serial1.begin(115200);
	  if (S2 != "") Serial2.begin(115200);
	  if (S3 != "") Serial3.begin(115200);
	  Serial.println(String(MODEL) + ":Serial start");
	  if (S1 != "") Serial1.println(String(MODEL) + ":Serial1 start to " + String(S1));
	  if (S2 != "") Serial2.println(String(MODEL) + ":Serial2 start to " + String(S2));
	  if (S3 != "") Serial3.println(String(MODEL) + ":Serial3 start to " + String(S3));
	}

	void loop() {
	  Serial.println(String(MODEL) + ":Serial print");
	  if (S1 != "") Serial1.println(String(MODEL) + ":Serial1 TX to " + String(S1));
	  if (S2 != "") Serial2.println(String(MODEL) + ":Serial2 TX to " + String(S2));
	  if (S3 != "") Serial3.println(String(MODEL) + ":Serial3 TX to " + String(S3));
	  delay(5000);
	}

edogaldo, 2019-08-08:
Looks like a crystal issue, you sure you are using [this board](https://www.ebay.com/itm/STM32F407VET6-STM32-ARM-Cortex-M4-Development-Board-System-Board-/173407441568?nav=SEARCH)? It should have a 25Mhz crystal while the others a 8Mhz one.

nopnop2002, 2019-08-08:
I used a DIYMROE board.
I didn't know that there was such a difference.
I will close this issue.

closed 2019-08-08 by nopnop2002