stm32duino/Arduino_Core_STM32 #592: How to implement printf from "newlib nano + Float Printf"

pkourany, 2019-08-06:
It became clear that compiling with `newlib nano + Float Printf` did not automatically extend the Stream or Print class with `printf` methods.  It is not clear what needs to be done to enable `printf` though I suspect the form `SerialX.printf()` will never work.  I'm not sure where `printf` "prints" to but how can it be redirected to a Serial device?

uzi18, 2019-08-06:
It is possible, habe use here small patch, can paste it late evening.

pkourany, 2019-08-06:
It would also be nice for STM to (also) provide an "official" solution here.

uzi18, 2019-08-06:
we can try to ask @fpistm to integrate it into repo

pkourany, 2019-08-06:
It would be good since there is an expectation that selecting the menu options will enable this functionality.

uzi18, 2019-08-06:
@pkourany this is also kind of idea: https://gist.github.com/ridencww/4e5d10097fee0b0f7f6b

pkourany, 2019-08-06:
I have working "lite" printf implementation that was developed for the STM32 Maple Core but I would rather not deviate from the STM32 Core.  Worst case, we do a PR for print.x in the Core code.

uzi18, 2019-08-06:
can't find this "lite" version

pkourany, 2019-08-06:
@uzi18, it is not in Roger's code.  I added it based on a contribution from a member on STMDuino:
... added to Print.cpp
```

#ifdef LEAN_PRINTF
//------------------------------------------------
#ifdef toupper
#undef toupper
#endif
#ifdef tolower
#undef tolower
#endif
#ifdef islower
#undef islower
#endif
#ifdef isdigit
#undef isdigit
#endif

#define toupper(c) ((c)&=0xDF)
#define tolower(c) ((c)|=0x20)
#define islower(c) ((unsigned char)c >= (unsigned char)'a' && (unsigned char)c <= (unsigned char)'z')
#define isdigit(c) ((unsigned char)c >= (unsigned char)'0' && (unsigned char)c <= (unsigned char)'9')

typedef union {
    unsigned char  byte[5];
    long           l;
    unsigned long  ul;
    float          f;
    const char     *ptr;
} value_t;

size_t Print::printDigit(unsigned char n, bool lower_case)
{
    register unsigned char c = n + (unsigned char)'0';

    if (c > (unsigned char)'9') {
        c += (unsigned char)('A' - '0' - 10);
        if (lower_case)
            c += (unsigned char)('a' - 'A');
    }
    return write(c);
}

static void calculateDigit (value_t* value, unsigned char radix)
{
    unsigned long ul = value->ul;
    unsigned char* pb4 = &value->byte[4];
    unsigned char i = 32;

    do {
        *pb4 = (*pb4 << 1) | ((ul >> 31) & 0x01);
        ul <<= 1;

        if (radix <= *pb4 ) {
            *pb4 -= radix;
            ul |= 1;
        }
    } while (--i);
    value->ul = ul;
}

size_t Print::printf(const char *format, ...)
{
    va_list ap;
    bool   left_justify;
    bool   zero_padding;
    bool   prefix_sign;
    bool   prefix_space;
    bool   signed_argument;
[... 221 lines cut ...]
                    else
                        width = 0;
                }
                 /* output the digits */
                while( length-- ) {
                    lsd = !lsd;
                    if (!lsd) {
                        pstore++;
                        value.byte[4] = *pstore >> 4;
                    } else {
                        value.byte[4] = *pstore & 0x0F;
                    }
                    charsOutputted+=printDigit(value.byte[4], lower_case);
                }
            }
        } else {
            charsOutputted+=write(c);
        }
    }
    va_end(ap);
    return (size_t)charsOutputted;
}

#endif
```

fpistm, 2019-08-19:
Hi @pkourany 
your issue is to extend class with ::printf() method ? Right?

About standard C printf, it is supported thanks the newlib nano implementation.
By default there is no float support in this printf, to enable it you have to enable it explicitely using the option: `newlib nano + Float Printf`

See [newlib-nano_readme](https://www.st.com/content/ccc/resource/technical/document/user_manual/group1/cd/29/43/c5/36/c0/40/bb/Newlib_nano_readme/files/newlib-nano_readme.pdf/jcr:content/translations/en.newlib-nano_readme.pdf)

> 
> Newlib-nano is different from newlib in addition to the libraries' name.
> Formatted input/output of floating-point number are implemented as weak symbol.
> If you want to use %f, you have to pull in the symbol by explicitly specifying"-u" command option.
>   -u _scanf_float  
>   -u _printf_float
> 
> e.g. to output a float, the command line is like: 
> $ arm-none-eabi-gcc --specs=nano.specs -u _printf_float $(OTHER_LINK_OPTIONS)

By default C `printf` will output on the `DEBUG_UART`:
https://github.com/stm32duino/Arduino_Core_STM32/blob/a8350866daf5abfd97b7cd76ec514a1f22cace39/cores/arduino/stm32/uart.c#L587
this is done thanks the `_write()` in syscall which can be redefined as weak:
https://github.com/stm32duino/Arduino_Core_STM32/blob/a8350866daf5abfd97b7cd76ec514a1f22cace39/cores/arduino/syscalls.c#L84-L85

It is also used by the core_debug feature to print some debug message (disabled per default).

pkourany, 2019-08-19:
@fpistm, I defined DEBUG_UART in my variant.h file and used printf() instead of Serial.printf() and all compiled.  I will be testing and reporting back.  Thanks!

uzi18, 2019-08-19:
my patch:
```
diff --git a/cores/arduino/Print.h b/cores/arduino/Print.h
index 05b585ea..68058b1f 100644
--- a/cores/arduino/Print.h
+++ b/cores/arduino/Print.h
@@ -103,6 +103,44 @@ class Print {
     void println(uint64_t, uint8_t = DEC);
     void print(uint64_t, uint8_t = DEC);
 #endif
+
+#include <stdarg.h>
+#define PRINTF_BUF 80 // define the tmp buffer size (change if desired)
+   void printf(const char *format, ...)
+   {
+   char buf[PRINTF_BUF];
+   va_list ap;
+        va_start(ap, format);
+        vsnprintf(buf, sizeof(buf), format, ap);
+        for(char *p = &buf[0]; *p; p++) // emulate cooked mode for newlines
+        {
+                if(*p == '\n')
+                        write('\r');
+                write(*p);
+        }
+        va_end(ap);
+   }
+#ifdef F // check to see if F() macro is available
+   void printf(const __FlashStringHelper *format, ...)
+   {
+   char buf[PRINTF_BUF];
+   va_list ap;
+        va_start(ap, format);
+#ifdef __AVR__
+        vsnprintf_P(buf, sizeof(buf), (const char *)format, ap); // progmem for AVR
+#else
+        vsnprintf(buf, sizeof(buf), (const char *)format, ap); // for the rest of the world
+#endif
+        for(char *p = &buf[0]; *p; p++) // emulate cooked mode for newlines
+        {
+                if(*p == '\n')
+                        write('\r');
+                write(*p);
+        }
+        va_end(ap);
+   }
+#endif
+
 };
 
 #endif

```

pkourany, 2019-08-21:
@fpistm, I got everything working with and without float support.  Thanks!

closed 2019-08-21 by pkourany