stm32duino/Arduino_Core_STM32 #594: [New Variant]STM32F401

mrguen, 2019-08-10:
Helllo, Thanks for the huge quantity of work!

I would like to know if you plan to include in the release 1.7.0 these generic variants for F401

STM32F401CC in UFQFPN48 and LQFP64 packages

STM32F401RCT6  in  LQFP64 package

Generally speaking how do you handle the different packages versions?

pkourany, 2019-08-12:
@mrguen, I have the RCT6 variant working on my system with v1.61 but haven't create a PR yet.  I you want the variant file package and changes boards.txt file, I can post it.

mrguen, 2019-08-12:
@pkourany Oh, Thanks a lot. That would be nice of you.

pkourany, 2019-08-14:
Here is the working RCT6 variant. I have attached a zip of the variant folder. Do not rename this folder. The following needs to be added to the boards.txt file under the Generic F1 section of the file:
```
# Generic F103RCT6 board
GenF1.menu.pnum.GENERIC_F103RC=Generic F103RCT6
GenF1.menu.pnum.GENERIC_F103CB.upload.maximum_size=262144
GenF1.menu.pnum.GENERIC_F103CB.upload.maximum_data_size=49152
GenF1.menu.pnum.GENERIC_F103CB.build.board=NUCLEO_F103RB
GenF1.menu.pnum.GENERIC_F103CB.build.product_line=STM32F103xE
GenF1.menu.pnum.GENERIC_F103CB.build.variant=GENERIC_F103RCT6
```
[GENERIC_F103RCT6.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3499425/GENERIC_F103RCT6.zip)

You should review the pin used for an onboard LED the variant.h file.  The included variant.h file specifies PA15 for the LED.

mrguen, 2019-08-14:
Ok, Thanks Paul. The issue was open for STM32F401... I don't know if these files might help for the RCT6 type...

pkourany, 2019-08-14:
Oh boy.  I must have dyslexia cause I read F103RCT6 somehow!  I'll work on the 401RCT6 for you and post when done and see what I can do for the 401CC. I'm going to hide that post since it isn't relevant.

mrguen, 2019-08-14:
Ok. I tried to do it by following your method and https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

A) **************
There was a compilation problem because the board is not considered in Arduino15\packages\STM32\hardware\stm32\1.6.1\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include\stm32f4xx.h

So I added

#if defined(STM32F401xC)
  #include "stm32f401xc.h"

and uncommented #define STM32F401xC

B) **************
I modified boards.txt as :

################################################################################
# Generic F4

GenF4.name=Generic STM32F4 series

GenF4.build.vid=0x0483
GenF4.build.core=arduino
GenF4.build.board=GenF4
GenF4.build.extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial} {build.bootloader_flags}
GenF4.build.mcu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard
GenF4.build.series=STM32F4xx
GenF4.build.cmsis_lib_gcc=arm_cortexM4l_math

# Secret Garden F401RC
GenF4.menu.pnum.STM32F401RCT6=Secret Garden F401RCT6
GenF4.menu.pnum.STM32F401RCT6.upload.maximum_size=262144
GenF4.menu.pnum.STM32F401RCT6.upload.maximum_data_size=65536
//GenF4.menu.pnum.STM32F401RCT6.build.board=STM32F401RCT6
GenF4.menu.pnum.STM32F401RCT6.build.product_line=STM32F401xx
GenF4.menu.pnum.STM32F401RCT6.build.variant=STM32F401RCT6 

C) **************
But now there is a compilation error:

WARNING: Category 'Language' in library ArduinoStreaming is not valid. Setting to 'Uncategorized'
Un sous-r�pertoire ou un fichier sketch existe d�j�.

In file included from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_def.h:30,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_rcc.h:29,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/STM32F4xx/stm32f4xx_hal_conf_default.h:291,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/STM32F4xx/stm32f4xx_hal_conf.h:13,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal.h:30,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:252,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/stm32_def.h:35,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/clock.h:43,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/wiring_time.h:23,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/wiring.h:38,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/Arduino.h:32,
                 from sketch\Blink.ino.cpp:1:
 #ifndef __STM32F4xx_H
 
In file included from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/stm32_def.h:35,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/clock.h:43,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/wiring_time.h:23,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/wiring.h:38,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/Arduino.h:32,
                 from sketch\Blink.ino.cpp:1:
 #ifndef __STM32F4xx_H
 
In file included from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_rtc.h:29,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/LL/stm32yyxx_ll_rtc.h:21,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/backup.h:26,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/board.h:9,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/wiring.h:41,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/Arduino.h:32,
                 from sketch\Blink.ino.cpp:1:
 #ifndef __STM32F4xx_H
 
In file included from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_gpio.h:29,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/LL/stm32yyxx_ll_gpio.h:21,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/stm32/digital_io.h:43,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/board.h:12,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/wiring.h:41,
                 from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino/Arduino.h:32,
                 from sketch\Blink.ino.cpp:1:
[... cut ...]
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at cc.arduino.Compiler.message(Compiler.java:525)
	at cc.arduino.i18n.I18NAwareMessageConsumer.message(I18NAwareMessageConsumer.java:80)
	at cc.arduino.MessageConsumerOutputStream.flush(MessageConsumerOutputStream.java:71)
	at cc.arduino.MessageConsumerOutputStream.write(MessageConsumerOutputStream.java:54)
	at java.io.OutputStream.write(OutputStream.java:75)
	at cc.arduino.Compiler.lambda$callArduinoBuilder$3(Compiler.java:307)
	... 2 more

pkourany, 2019-08-14:
@mrguen can you zip your variant directory and share here?

mrguen, 2019-08-14:
Here it is: [STM32F401RCT6.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3502764/STM32F401RCT6.zip)

[boards.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/3502768/boards.txt)

mrguen, 2019-08-14:
Boards.txt did not like the upload...
[boards.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3502772/boards.zip)

mrguen, 2019-08-14:
Here is what I have done :

1)	Copy C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\variants\board_template 
to
C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\variants\STM32F401RCT6

2)	In STM32CubeMX
Used the default settings for STM32F401RC but changed 
* the clock for HSE / Crystal/Ceramic, set 8 MHz crystal M=4 PLL= 168 P=4 Q=7 
SYSCLK = 84 MHZ AHBPrescaler = 1 APB1Prescaler = 2

·	Activated USART1 Asynchronous
·	Activated SPI1 / Full Duplex master
·	Activated USB / OTG/Dual role device
Generated the code for SW4STM32

3)	Copied STM32F401RCTx_FLASH.ld to ldscript.ld

4)	Copied SystemClock_Config(void) from the generated main.c to variant.cpp

5)	Copied the files PeripheralPins.c and PinNamesVar.h from https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap/Arduino/STM32F401R(B-C)Tx to the STM32F401RCT6 folder

6)	Copied digitalPin array in variant.cpp and pin defines in variant.h from NUCLEO_F401RE 

In fact I am using this board https://fr.aliexpress.com/item/32894670982.html?spm=a2g0s.9042311.0.0.27426c37RrpWiJ  and really don’t know there pin mapping. I have asked them a documentation… I suppose I can manage with the Nuclea numbering. I only changed the BOOT1 that was not considered on the NUCLEO board.

7)	Modified boards.txt by adding in the “# Generic F4” section
 
# Secret Garden F401RC
GenF4.menu.pnum.STM32F401RCT6=Secret Garden F401RCT6
GenF4.menu.pnum.STM32F401RCT6.upload.maximum_size=262144
GenF4.menu.pnum.STM32F401RCT6.upload.maximum_data_size=65536
//GenF4.menu.pnum.STM32F401RCT6.build.board=STM32F401RCT6
GenF4.menu.pnum.STM32F401RCT6.build.product_line=STM32F401xx
GenF4.menu.pnum.STM32F401RCT6.build.variant=STM32F401RCT6

8) There was a compilation error. I had to add at line 144 in  C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include\stm32f4xx.h

#if defined(STM32F401xC)
#include "stm32f401xc.h"
 
And uncomment #define STM32F401xC

mrguen, 2019-08-15:
The seller sent me the link to the documentation : https://wetransfer.com/downloads/493b7605d31718be87a56133e7cc70d920190814152406/f44f44?spm=a2g0s.imconversation.0.0.7d7a3e5fzulqrL 

There is a lot of material. Including an STM32CubeMX project that I used to replace ldscript.ld and ·	SystemClock_Config()

The compile error is still the same.

fpistm, 2019-08-15:
Do not modify the core file.
Your issue comes from your board definition:

`GenF4.menu.pnum.STM32F401RCT6.build.product_line=STM32F401xC`

mrguen, 2019-08-15:
I modified the boards.txt as explained but the error is still raised.

fpistm, 2019-08-15:
It seems you have an preprocessor directive issue, missing #endif or something like that.
I could not help this week I'm on vacation. I will test next week,

mrguen, 2019-08-15:
Ok, Thanks, I let a #if instead of changing to #elif in stm32f4xx.h

Now, despite the #define STM32F401xC  in stm32f4xx.h

 it seems to be unknwown by stm32_def_build.h

In file included from C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino\stm32\startup_stm32yyxx.S:1:
C:\Users\TG\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.6.1\cores\arduino\stm32\stm32_def_build.h:336:2: error: #error UNKNOWN CHIP
 #error UNKNOWN CHIP

[stm32f4xx.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3505621/stm32f4xx.zip)

fpistm, 2019-08-19:
Hi @mrguen 

> I would like to know if you plan to include in the release 1.7.0 these generic variants for F401
> 
> STM32F401CC in UFQFPN48 and LQFP64 packages
> 
> STM32F401RCT6 in LQFP64 package

There is no plan for this. This is a community project and all contributions are welcome.

> Generally speaking how do you handle the different packages versions?

In fact, variant is mainly how you order the pin (pin mapping) and linked to a board. Then the board determines the package used.
Anyway a fully generic variant dedicated only to a MCU could be done and then a dedicated entry in the boards.txt could be used to determine the package.

mrguen, 2019-08-19:
@fpistm Ok, so if you can help me debug I will provide these...

fpistm, 2019-08-19:
About your build issue, as I mentioned you don't have to modify the core files.
The include is already present and the define have to be done thanks the boards.txt.
https://github.com/stm32duino/Arduino_Core_STM32/blob/6d9575ec6a4452d52a2b0d73656c3a060dd0e2c6/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h#L160

Using your variant and updating properly the boards.txt there is no build issue.
I will provide a branch on my fork with some update.

mrguen, 2019-08-19:
It's written in the file line 72: "/* Uncomment the line below according to the target STM32 device used in your
   application
  */"

fpistm, 2019-08-19:
Well, it's also written this:
https://github.com/stm32duino/Arduino_Core_STM32/blob/6d9575ec6a4452d52a2b0d73656c3a060dd0e2c6/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h#L112-L114

Please, come back to original core files and only update your variant and boards.txt as described in the wiki.

fpistm, 2019-08-19:
@mrguen ,
here a proposed variant for your board:
https://github.com/fpistm/Arduino_Core_STM32/commit/a83cf0ce25fa4b66d86e3baa307af6850ba9fbe8

mrguen, 2019-08-20:
@fpistm,

Thanks. I could compile blink.

fpistm, 2019-08-20:
@mrguen 
Note that, I will not be able to test as I don't have this board.

mrguen, 2019-08-20:
@fpistm: do you have a test protocol?

fpistm, 2019-08-20:
Well, I have to deploy a test bench but don't have time yet to do it with a tests lists. 😭 
Usually, I use standard Arduino sketch to test each feature and some of our sketches examples of our libraries.

mrguen, 2019-08-21:
@fpistm I tried to configure the device to upload using USB but it seems to be work in progress and I could not do it:

Installed bootloader https://github.com/Serasidis/STM32_HID_Bootloader/releases/tag/2.2.1 via serial

But the PR https://github.com/stm32duino/Arduino_Core_STM32/pull/525 talks about a version 3 !

I suppose PR 525 was patched in 1.6.0 ...

No COM port is enumerated.

I tried to upload with upload method HID 2.2 but got this error

+-----------------------------------------------------------------------+
|         HID-Flash v2.2.1 - STM32 HID Bootloader Flash Tool            |
|     (c)      2018 - Bruno Freitas       http://www.brunofreitas.com   |
|     (c) 2018-2019 - Vassilis Serasidis  https://www.serasidis.gr      |
|   Customized for STM32duino ecosystem   https://www.stm32duino.com    |
+-----------------------------------------------------------------------+

> Trying to open the [COM2]...
> Unable to open the [COM2]
> Searching for [1209:BEBA] device...
##########
Error - [1209:BEBA] device is not found :(> Searching for [COM2] ...
> Finish
the selected serial port > Finish
 does not exist or your board is not connected

fpistm, 2019-08-21:
@mrguen you could not use this as the bootloader  is for F407 not F401.

mrguen, 2019-08-21:
@fpistm ok, so I guess
1) Uploading using only serial
2) asking for a STM32F401 bootloader / I don't know anything about this or can it be easily adapted from STM32F407?

fpistm, 2019-08-21:
F4 series include DFU in its builtin bootloader so you can use DFU.
See https://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf

In your case, you should have to set Boot0(pin) = 1 and Boot1(pin) = 0 then plug USB you should be in DFU mode.

For bootloader, I do not manage this. You can try to build your own based on @Serasidis repo or open an issue on his repo.

mrguen, 2019-08-21:
Unfortunately I just wrote the HID 2.2.1 bootloader...
Also I tried upolad method STM32CubeProgrammer(DFU) but got "STM32_Programmer_CLI.exe not found."

fpistm, 2019-08-21:
You have to install the STM32CubeProgrammer.
See https://github.com/stm32duino/wiki/wiki/Upload-methods#stm32cubeprogrammer

mrguen, 2019-08-21:
Error: Target device not found
Did I erase the factory bootloader by downloading bootloader HID 2.2.1?

mrguen, 2019-08-21:
[image]

fpistm, 2019-08-21:
Well this should work.
Try to open the STM32CubeProgrammer and select `USB `then `connect` to see if it can attach properly.

mrguen, 2019-08-21:
11:37:11 : Error: Establishing connection with device failed

I tried with and without the jumper that connects the D+ pull up resistor

fpistm, 2019-08-21:
Well, I think you have a driver issue. "STM device in DFU mode" should be "STM32  BOOTLOADER"
so probably you installed another driver for DFU.

fpistm, 2019-08-21:
You can try the driver provided with the tool:
C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeProgrammer\Drivers\DFU_Driver

mrguen, 2019-08-21:
Yes, it is working now. 
It does switch from DFU to USB/CDC after downloading.
It is very fast. I could set Serial @ 2 MHz (really?)

fpistm, 2019-08-21:
Fine. 
After upload the STM32CubeProgrammer start your binary.
For USB, the speed is not relevant.

mrguen, 2019-08-21:
My binary? Do you mean the HID 2.2.1 bootloader?

mrguen, 2019-08-22:
Hello @fpistm !

I am reviewing PeripheralPins.c and I can see that

Some timers functions are defined on two pins but other that could use 2 pins are defined on one pin only. Some timer signals are missing a pin (BKIN, ETR) etc...

Should the timers definition section list all pins/functions possible as referenced in "STM32F401xB STM32F401xC" datasheet?

fpistm, 2019-08-22:
The file is generated automatically and provides only used signal. BKIN and ETR are ignored when extracted. If they are required later they could be added.
Here all the generated file and script to generate them using CubeMx xml files.
https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap

mrguen, 2019-08-22:
1) There are alos CHxN pins defined...

2) What about arbitrating between pins/functions?

For example TIM2_CH1  is defined for PA0 and PA5 while PA0 could be used for TIM5_CH1 which is not mapped.

Can I change it / is there any other impact?

3) Some USART alt pins are not defined (PD5, PD6, PA11, PA12) while USART1 alt pins are.

4) Generally speaking does a pin has to be defined for only one function or for all functions possible 
Example 
PA_7 is mapped to
ADC7, TIM1_CH1N, SPI_MOSI

mrguen, 2019-08-22:
Also I just realize that the files you provided are not those from https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap/Arduino/STM32F401R(B-C)Tx

mrguen, 2019-08-22:
Here is the proposed PeripheralPins.c
[PeripheralPins.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3529515/PeripheralPins.zip)

mrguen, 2019-08-22:
I see two differences in the ldcript provided as compared to Altium's:

1) possibly missing
No _Min_Heap_Size = 0x400;      /* required amount of heap  */ that was in the original Altium project

2) Added properties . =   ALIGN(4); in many places

mrguen, 2019-08-22:
The proposed PeripheralPins.c discards TIM9, TIM10 and TIM11 but I see in variant.cpp 
#define TIMER_TONE              TIM10
#define TIMER_SERVO             TIM11

So it might be necessary to reactivate 
TIM10_CH1 instead of TIM4_CH3
TIM11_CH1 instead of TIM4_CH3
?

mrguen, 2019-08-22:
Here is the updated PeripheralPins.c  + excel sheet detailing the code review
[PeripheralPins.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3529986/PeripheralPins.zip)

mrguen, 2019-08-22:
About BOOT configuration:
To upload using STM32Prog, BOOT0= 1 but in this case pushing reset does not reset the program and BOOT0 must be set to 0 to manually reset.

mrguen, 2019-08-22:
So far I have tested:

Passed:
Blink: ok
AnalogReadSerial: A0 and A15 : ok
DigitalReadSerial : all 46 pins Ok but PA11 and PA12 that are DM and DP thus canno't have serial 

Not ok:
IWDG_button: WDT does not seem to reset
eeprom_wite (value = 100) then eeprom_read: all values are '255'

fpistm, 2019-08-22:
> 
> 
> I see two differences in the ldcript provided as compared to Altium's:
> 
>     1. possibly missing
>        No _Min_Heap_Size = 0x400;      /* required amount of heap  */ that was in the original Altium project
> 
>     2. Added properties . =   ALIGN(4); in many places

This is correct.
https://github.com/stm32duino/Arduino_Core_STM32/pull/604/commits/37f6252e8b4f1948e839f9f6c78fe50ad267f9cf#diff-fddda161300d628fd4ca3bfe7a9fa835R58

> 
> 
> The proposed PeripheralPins.c discards TIM9, TIM10 and TIM11 but I see in variant.cpp
> #define TIMER_TONE TIM10
> #define TIMER_SERVO TIM11
> 
> So it might be necessary to reactivate
> TIM10_CH1 instead of TIM4_CH3
> TIM11_CH1 instead of TIM4_CH3
> ?

No, there is no link between Tone or Servo  and the Pinmap PWM array which is used for `analogWrite`... Tone or Servo use the hardware timer resources and are not link to a pin....

> To upload using STM32Prog, BOOT0= 1 but in this case pushing reset does not reset the program and BOOT0 must be set to 0 to manually reset.

Right this is normal... to start the program you have to come back to default Boot pattern else you start in Bootloader mode...

> Passed:
> Blink: ok
> AnalogReadSerial: A0 and A15 : ok
> DigitalReadSerial : all 46 pins Ok but PA11 and PA12 that are DM and DP thus canno't have serial
> 
> Not ok:
> IWDG_button: WDT does not seem to reset
> eeprom_wite (value = 100) then eeprom_read: all values are '255'

For IWDG and eeprom I've tested a F401RE without any issue.
About IWDG ensure to have boot0 set to 0. For eeprom, I can not help.

fpistm, 2019-08-22:
Your update of PeripheralPins.c is not correct and I don't understand why you do this.
Anyway I will kept the variant as it is.  Basic features worked including USB.
Thanks for the test.

mrguen, 2019-08-22:
PeripheralPins.c is the file from https://github.com/stm32duino/Arduino_Tools/tree/master/src/genpinmap/Arduino/STM32F401R(B-C)Tx 

It seems the other lacked some pin/timer mapping (maybe I am wrong of course I don't know this as you do)

With pin arbitration between Timers defined as in the excel file in the .zip.

mrguen, 2019-08-22:
I will do some other things tomorrow and come back to this afterward if you'd like.

fpistm, 2019-08-22:
This is the same files except the commented lines...

mrguen, 2019-08-22:
Yes possibly but I find it nice to have all possible mappings written. I have checked all are correct / datasheet and chose to promote TIM2 and TIM5 instead of TIM9/10/11 because they are 32 bits and that is what I am looking for with this IC !

fpistm, 2019-08-23:
Well as said the PWM array is for `analogWrite()` to generate the PWM so we don't bother if 16 or 32 bits.
Moreover with the Hardware Timer library, we use all timer as 16 bits for genericity purpose:
https://github.com/stm32duino/wiki/wiki/HardwareTimer-library#1-introduction
But nothing prevents you from using a timer in 32 bits using LL or HAL directly!

mrguen, 2019-08-23:
Thanks for the info... I am very much surprised. I will have a look at the library. Usually it is also possible to access the registers so I don't see why it wouldn't be 32 bits.

fpistm, 2019-08-23:
For the provided API 16 bits is enough. For further use case HAL/LL could be used.

closed 2019-08-28 by fpistm

mrguen, 2019-11-08:
I also noticed other things:

I tried to clean AppData and reinstall boards:

- installing STM32 Cores takes forever
- it is HUGE aroung 835 Mb!

commit 5bb50be9 (2019-08-28): Core Board F401RC

Fixes #594
PR #604 (2019-08-28): Core Board F401RC
Fixes #594

Based on @mrguen first try for this board:
 https://fr.aliexpress.com/item/32894670982.html?spm=a2g0s.9042311.0.0.27426c37RrpWiJ

Shematics: [STM32F401RCT6_CORE_BOARD_原理图.pdf](https://github.com/stm32duino/Arduino_Core_STM32/files/3519013/STM32F401RCT6_CORE_BOARD_.pdf)
Pinouts [STM32F401RCT6_CORE_BOARD尺寸图.pdf](https://github.com/stm32duino/Arduino_Core_STM32/files/3519014/STM32F401RCT6_CORE_BOARD.pdf)