stm32duino/Arduino_Core_STM32 #595: Implement one-second interrupts from RTC

pkourany, 2019-08-12:
The STM32 Core contains HAL code in stm32f1xx_hal_rtc_ex.c to implement the RTC's built-in one second interrupt. However, this functionality is not currently included in the STM32RTC library.  Can this be added?

Otherwise, given that many of the functions called in STM32RTC are defined in rtc.c (the RTC driver) is this the right file to add the extra functionality to?

fpistm, 2019-08-20:
Hi @pkourany 
well all features can be added 🗡 
Do not hesitate to provide a PR for this here and in the STM32RTC library.
And yes rtc.* is the right place.

fpistm, 2020-10-07:
This issue will be solved thanks https://github.com/stm32duino/STM32RTC/pull/42
I close it here as the RTC drivers has been moved to STM32RTC library.

closed 2020-10-07 by fpistm