stm32duino/Arduino_Core_STM32 #596: Black F407VET6 pins messed up since 1.5.0

trylika, 2019-08-12:
On version 1.5.0 there was added support for Black F407VG/ZE/ZG boards, which messed up pin numbering on Black F407VET6 ( https://wiki.stm32duino.com/index.php?title=STM32F407 https://stm32-base.org/boards/STM32F407VET6-STM32-F4VE-V2.0 ) board.

Good example would be LED pins PA6(LED_BUILTIN) and PA7, which correspond to 58 and 59 respectively, but should be 6 and 7. There are more discrepancies among pin assignments.

trylika, 2019-08-12:
or maybe it was already not correct in 1.4.0... feeling a bit confused, maybe variant Black F407VET6 is not for mentioned board?

uzi18, 2019-08-12:
there was no Black F407VE as you can see

trylika, 2019-08-12:
@uzi18 https://github.com/stm32duino/Arduino_Core_STM32/tree/1.4.0/variants/BLACK_F407VE

uzi18, 2019-08-13:
it seem to be 2 BLACK F407VE

trylika, 2019-08-13:
it might as well be, i just assumed that it's the one like in here https://stm32-base.org/boards/STM32F407VET6-STM32-F4VE-V2.0 
because previously it was the only black f4ve board board mentioned on wiki.stm32duino.com (can't confirm because site is down).
if it's not, then this should be not a bug but rather variant request type of issue

fpistm, 2019-08-20:
Hi @trylika 

> Good example would be LED pins PA6(LED_BUILTIN) and PA7, which correspond to 58 and 59 respectively, but should be 6 and 7. There are more discrepancies among pin assignments.

I don't understand your point about pin assignments?
why specifically 6 and 7?
Simply use pin naming PXn in your sketch when a pin numbering is required.

trylika, 2019-08-21:
but that's the point - i can't use pin naming, because when using `BLACK_F407VE` variant on this board ( https://stm32-base.org/boards/STM32F407VET6-STM32-F4VE-V2.0 for which i assumed it was built ), because pin naming does not correspond to actually pins

in other words, if i try to switch on built in led using constants LED_BUILTIN or PA6 or PA7, nothing happens

fpistm, 2019-08-21:
I guess this is not the same.
I have this board and it works well.
Led are well defined:
https://github.com/stm32duino/Arduino_Core_STM32/blob/8134a78a09de5b4f6f041c967e2cfaa2b373c210/variants/BLACK_F407XX/variant.h#L138-L139
If you well select Black F407VE this should work. 
So, I guess your board is not exactly the same or there is another mistake (boot0/1 pin config, flash issue,...)

trylika, 2019-08-21:
PA6 and PA7 translate later into these on this board

https://github.com/stm32duino/Arduino_Core_STM32/blob/8134a78a09de5b4f6f041c967e2cfaa2b373c210/variants/BLACK_F407XX/variant.h#L102-L103

but from what i found, in this example, to control LED'is, they have to be specified the same as it's done in here:

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/a3a56866505eeca4b49fc435700dcd4bd5540c8c/STM32F4/variants/generic_f407v/generic_f407v.h#L181

that is to 6 and 7 respectively

maybe i'm missing something in my setup, i'll try it out later

fpistm, 2019-08-21:
This is not the same core, pins number are not the same.
If you use @rogerclarkmelbourne core then it is 6 and 7 but not from this core.
This is only pin mapping. Using PXn notation will avoid any issue.
So which core you used?

trylika, 2019-08-22:
spent some time digging, it seems i misinterpreted issue, to give more details:
i have simplest possible project setup using Platformio:
```
#include <Arduino.h>

void setup()
{
    pinMode(PA6, OUTPUT);
    pinMode(PA7, OUTPUT);
}

void loop()
{
    digitalWrite(PA6, HIGH);
    digitalWrite(PA7, LOW);
    delay(1000);
    digitalWrite(PA6, LOW);
    digitalWrite(PA7, HIGH);
    delay(1000);
}
```
and `platformio.ini`
```
[env:stm32blink]
platform = ststm32
board = black_f407ve
framework = arduino

upload_protocol = stlink
```

with such setup and current versions of libraries it seems that compiled binary is not executed at all after upload.
if i change board to `genericSTM32F407VET6` (which changes core to `maple` internally) compiled binary is executed without any problems, thus got me an impression that pin numbering is messed up.

Tried exact same project from Arduino IDE and `Arduino_Core_STM32` setup - it works flawlessly, so it seems there is completely different issue in here and related to how platformio links final binary. Unfortunately, i don't have enough knowledge in this field currently to find exact issue, if somebody is interested, i could provide more details/logs. But for comparison, some logs from Platformio

for `black_f407ve`:
```
> Executing task in folder STM32Blink: platformio.exe run --target upload <

Processing stm32blink (platform: ststm32; board: black_f407ve; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/black_f407ve.html
PLATFORM: ST STM32 5.5.0 > Black STM32F407VE
HARDWARE: STM32F407VET6 168MHz, 192KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, tool-dfuutil 1.9.190708, framework-arduinoststm32 3.10601.190716 (1.6.1), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-openocd 2.1000.190422 (10.0)       
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 24 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio\build\stm32blink\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.4% (used 828 bytes from 196608 bytes)
PROGRAM: [          ]   2.4% (used 12828 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\stm32blink\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
hla_swd
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080001cc msp: 0x20020000
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
wrote 16384 bytes from file .pio\build\stm32blink\firmware.elf in 0.598720s (26.724 KiB/s)
** Programming Finished **
** Verify Started **
verified 13248 bytes in 0.097959s (132.071 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1800 kHz
shutdown command invoked
```

for `genericSTM32F407VET6`:
```
> Executing task in folder STM32Blink: platformio.exe run --target upload <

Processing stm32blink (platform: ststm32; board: genericSTM32F407VET6; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VET6.html
PLATFORM: ST STM32 5.5.0 > STM32F407VE (192k RAM. 512k Flash)
HARDWARE: STM32F407VET6 168MHz, 128KB RAM, 502.23KB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, tool-dfuutil 1.9.190708, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-openocd 2.1000.190422 (10.0) 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 21 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pio\build\stm32blink\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   5.6% (used 7376 bytes from 131072 bytes)
PROGRAM: [          ]   3.8% (used 19620 bytes from 514288 bytes)
Configuring upload protocol...
AVAILABLE: dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\stm32blink\firmware.elf
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
hla_swd
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
none separate
adapter speed: 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08002f28 msp: 0x20030000
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
wrote 32768 bytes from file .pio\build\stm32blink\firmware.elf in 1.076965s (29.713 KiB/s)
** Programming Finished **
** Verify Started **
verified 19620 bytes in 0.157681s (121.512 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 1800 kHz
shutdown command invoked
```

fpistm, 2019-08-23:
OK this is more clear.
Unfortunately, I don't use PIO. As I can see you ask them support and this is the right think to do.
https://github.com/platformio/platform-ststm32/issues/271#issuecomment-524095851

thanks

closed 2019-08-23 by fpistm