stm32duino/Arduino_Core_STM32 #597: Nucleo-L4R5ZI Can't I2C

Yonits12, 2019-08-13:
Hi, 
I have a Nucleo-L4R5ZI and I just do not know which of the I2C pins I should use with your package. I've tried all of the options (I2C1, I2C2, I2C3) and none of them worked...

Do you have any suggestions for that issue? 
Is it possible that the I2C feature is not working on the current version yet?

Thanks.

GianniDPC, 2019-08-15:
Have you tried using the I2C scanner from Examples -> Wire? You can also remap the SCL and SDA pins using :

```
  Wire.setSDA(PB9);
  Wire.setSCL(PB8);
```

fpistm, 2019-08-19:
Hi @Yonits12,

By default Wire is instantiate with D14 (SDA) and D15 (SCL) of the Arduino UNO R3 connector.
Those pins are  PB9 (SDA) and PB8 (SCL) and linked to I2C1.
https://github.com/stm32duino/Arduino_Core_STM32/blob/cfe6dd8ea25b9f64bf99f72ff59681eb026750f1/variants/NUCLEO_L4R5ZI/variant.h#L58-L59

https://github.com/stm32duino/Arduino_Core_STM32/blob/cfe6dd8ea25b9f64bf99f72ff59681eb026750f1/variants/NUCLEO_L4R5ZI/PeripheralPins.c#L82
https://github.com/stm32duino/Arduino_Core_STM32/blob/cfe6dd8ea25b9f64bf99f72ff59681eb026750f1/variants/NUCLEO_L4R5ZI/PeripheralPins.c#L104

Do not forget to use a pull-up resistors on both lines.

fpistm, 2019-08-27:
Considered answered.
I2C works.

closed 2019-08-27 by fpistm