stm32duino/Arduino_Core_STM32 #599: STM32F103C8T6 (BluePill) I2C1 not working

GianniDPC, 2019-08-13:
I2C1 does not seem to work for me, I2C2 does seem to work as expected.

I used the I2C scanner and a TC74 temperature sensor to test.

Using I2C2 the address of the sensor is detected, when using I2C1 it tells me that no devices are found.

Can anyone confirm this? It could also be that my board is defective I guess.

I am using a STM32F103C8T6 board with the latest STM32duino release 1.6.1.

nopnop2002, 2019-08-17:
I2C1 of STM32F103C8T6 works fine.
SDA=PB7
SCL=PB6

Maybe you need a PullUp resistor

[image]

GianniDPC, 2019-08-18:
@nopnop2002  already tried, I guess I have a bad solder somewhere on the chip connecting to the pcb.

closed 2019-08-18 by GianniDPC