stm32duino/Arduino_Core_STM32 #600: Won't Compile

zpowellman, 2019-08-16:
I'm using Arduino Version 1.8.9.  I'm using a STM32F407VET6 Board.
I'm using the "Generic STMF4 series" with the STM32F40VET card selected.
I get this message on compile.
 
arm-none-eabi-g++": file does not exist

I have the Adafruit  SAM Boards Cortex-M3 1.6.12 installed
I have the latest STM32 core loaded from this site.
Any ideas

uzi18, 2019-08-16:
try to remove stm32 core from board manager and install one more time

zpowellman, 2019-08-16:
Hi Bartłomiej ,

Thanks very much for your reply and hint.  It now compiles, but I’m having difficulty loading it.  I’m trying to use the ST-Link to upload, but I’m not sure how to set the tool options, in particular, the “Upload Method”  I have the programmer set to ST-Link but I get errors in the upload. They are different errors based on my selection in “Upload Method”

Any more hints?

 

Thanks again 

Dan Powell

nopnop2002, 2019-08-17:
Try this:

[image]

COM22 is STLink Virtual COM Port:

[image]

zpowellman, 2019-08-18:
Thanks for all the responses.

zpowellman, 2019-08-18:
I downloaded the STM32Cube programmer and I can now program the device with arduino and my ST-Link.  I've also been successful using the Serial Comm and I2C interfaces.  Now trying to get SPI and then SD card operational.
Thanks All
Dan

closed 2019-08-18 by zpowellman