stm32duino/Arduino_Core_STM32 #603: Enabling multiple serial devices on variant

pkourany, 2019-08-19:
@fpistm,

I have built a variant for an STM32F103RCT6 based on the STM32F103xE.  The RCT6 has several USART/UART peripherals of which I use four of them in my application.  When I compile my code with the variant, I get errors indicating Serial2, Serial3, Serial4 and Serial5 are not defined.  The `PeripheralPins.c` has entries for these serial peripherals.

The ONLY place I see the Serial ports defined is in `Hardware.c` in the constructor `HardwareSerial::HardwareSerial(uint32_t _rx, uint32_t _tx)`.  In the constructor code, the only way a Serial device is built is if `PIN_SERIALx_RX` and `PIN_SERIAL1_TX` are defined.

Am I correct in assuming that these definitions must be included in the `varian.h` file so the corresponding SerialX objects can be built?

fpistm, 2019-08-19:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

pkourany, 2019-08-19:
@fpistm, I completely missed that in the wiki!  I built the objects in my code as suggested.  The maple core did this by default.  That solved the problem. Thanks!

closed 2019-08-19 by pkourany