stm32duino/Arduino_Core_STM32 #606: Support precompiled libraries

massimiliano-mantione, 2019-08-21:
I am trying to use a precompiled library but the build system does not recognize it properly.

The reason: I am mixing Rust code into an Arduino sketch.
I compile the Rust code into a static library and then link this library into the Arduino sketch.

To link it, I put the Rust library in the Arduino library folder with a proper `library.properties` file and directory structure.

This works perfectly for Teensy 3.5 and 4 boards.

I tried to do it for an STM32 board (specifically, a Nucleo L432KC), with the following `library.properties` file:

```
name=rust-main-lib
version=1.0.0
author=Massimiliano Mantione <massimiliano.mantione@gmail.com>
maintainer=Massimiliano Mantione <massimiliano.mantione@gmail.com>
sentence=A library to test Arduino-rust integration
paragraph=Just a test
category=Communication
url=http://github.com/massimiliano-mantione
architectures=*
includes=rust-main-lib.h
precompiled=true
ldflags=-lrust-main-lib
```

where `find rust-main-lib` from the Arduino libraries directory gives the following:
```
rust-main-lib
rust-main-lib/src
rust-main-lib/src/cortex-m4
rust-main-lib/src/cortex-m4/librust-main-lib.a
rust-main-lib/src/rust-main-lib.h
rust-main-lib/library.properties
```

but the build process terminates with the following error message:
```
Compiling libraries...
Compiling library "rust-main-lib"
Using library rust-main-lib at version 1.0.0 in folder: /home/massi/Arduino/libraries/rust-main-lib 
stat /home/massi/Arduino/libraries/rust-main-lib/src/cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard: no such file or directory
Error compiling for board Nucleo-32.
```

The message is confusing: the `cortex-m4` directory is there, and it contains the precompiled library.

When doing the same for the Teensy boards if I did not create the CPU-specific directory the error message was the proper one (it could not find the library file).

How can this be fixed?

Or, if this is not the proper repository where to file the issue, where should I file it?

fpistm, 2019-08-21:
Hi @massimiliano-mantione 
I never try to do this and do not see why you get this error.
It seems more an Arduino issue.
Anyway could you provide me your library and the sketch you try to build to test on my side.
Thanks

fpistm, 2019-08-26:
@massimiliano-mantione any update on this ? Thanks.

fpistm, 2019-08-30:
No feedback since and enough information to reproduce.
I close it.
@massimiliano-mantione  feel free to comment and add requested information then I will reopen if needed.

massimiliano-mantione, 2019-08-30:
Sorry for the late reply.
I opened the issue just before leaving for vacation, which with hindsight is not nice :-/

There is this repository on github that shows what I am trying to do: [rust-arduino-st-nucleo-l432kc](https://github.com/ma
ssimiliano-mantione/rust-arduino-st-nucleo-l432kc).

It contains a small rust library, an arduino sketch, and a script that puts everything in place so that an arduino build of that sketch uses the rust library as an arduino precompiled library.

You should look at it at [this](https://github.com/massimiliano-mantione/rust-arduino-st-nucleo-l432kc/tree/c4303d215046acd55153f8cf5e78e54f1fdc007b) commit, particularly at the "[build.sh](https://github.com/massimiliano-mantione/rust-arduino-st-nucleo-l432kc/blob/c4303d215046acd55153f8cf5e78e54f1fdc007b/build.sh)" script.

The script does the following:

* it builds the rust library (don't worry about it for now)
* it creates the directories for the library and the sketch
* it copies the library, its header file and its `library.properties` file in place
* it also copies the sketch in a proper place (this is actually useless)

After this, in principle, building the sketch should succeed.

I have been able to make this procedure work for a Teensy 3.5 board.

If I try this with stm32duino I get the error described at the beginning of this issue.

I am attaching a zip file of the full library directory, I hope this helps.

Thanks!

massimiliano-mantione, 2019-08-30:
[rust-main-lib.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3560915/rust-main-lib.zip)
This is the zip file of the library dir.

fpistm, 2019-08-30:
Ok. Thanks for update.

fpistm, 2019-09-03:
Hi @massimiliano-mantione 
I was able to build with your lib.
Referring to https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

> * **precompiled - (available from Arduino IDE 1.8.6/arduino-builder 1.4.0)** (optional) set to true to allow the use of .a (archive) and .so (shared object) files. The .a/.so file must be located at `src/{build.mcu}` where `{build.mcu}` is the architecture name of the target the file was compiled for. Ex: `cortex-m3` for the Arduino DUE. The static library should be linked as an ldflag.

In this core the `build.mcu` contains also `fpu` and `float-abi` options, that's why you get this error, instead of searching `cortex-m4`  it searches `cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard`.

So, to get it working, `{build.mcu}` should only contain `cortex-mx` value and no floating options.

fpistm, 2019-09-03:
@massimiliano-mantione 
I've submit a PR which fix your issue.

```
Hello CONFIG
Config done
Hello PRE
Hello from Rust!
Hello from Rust!
Hello from Rust!
Hello from Rust!
```

PR #629 (2019-09-04): Fix `{build.mcu}` definition
Library can provide a pre-compiled library.
See `precompiled` option here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

The .a(archive)/.so(shared object) file must be located at: `src/{build.mcu}`

So the `build.mcu` could not contain `fpu` and `float-abi` options
and should only contain `cortex-mx` value.

Else Arduino IDE, instead of searching `cortex-mx`, searches
`cortex-mx -mfpu=fpv4-sp-d16 -mfloat-abi=hard` if floating-point
options are set.

`build.flags.fp` has been added to contain floating- [... cut]
commit 70b8bce1 (2019-09-04): Fix `{build.mcu}` definition

Library can provide a precompiled library.
See precompiled option here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

The .a(archive)/.so(shared object) file must be located at: `src/{build.mcu}`

So the `build.mcu` could not contain `fpu` and `float-abi` options
and should only contain `cortex-mx` value.

Else Arduino IDE, instead of searching `cortex-mx`, searches
`cortex-mx -mfpu=fpv4-sp-d16 -mfloat-abi=hard` if floating-point
options are set.

`build.flags.fp` has been added to contain floating-po [... cut]
PR #719 (2019-10-23): Add new official library.properties field for static linking flags in platform.txt compiler recipe.
**Summary**

This PR implements the following **features**

* [x] Add compiler.libraries.ldflags variable to platform.txt

This adds the ability to link against statically compiled library files in Arduino Libraries using the precompiled and ldflags keys described [here](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format).

**Closing issues**

Fixes #606 which I do not believe was fully fixed by https://github.com/stm32duino/Arduino_Core_ [... cut]
closed 2019-09-04 by fpistm