stm32duino/Arduino_Core_STM32 #609: SharkyMKR from Midatronics

uallievi, 2019-08-23:
**Board description (please complete the following information):**
 - Name: Sharky MKR
 - Link to all relevant information:
   * Documentation attached below (User and Programmer's Guide DRAFT)
   * The hardware will be soon available at Arrow Electronics
   * Based on STM32WB55CE

Hello, I'm Umberto Allievi, senior firmware engineer at Midatronics. We are producing some STM32 based boards that will be sold by Arrow Electronics. The first one is named Sharky, and is based on STM32WB55CE. We'd like to have our board included in the stm32duino project.
You will find attached a zip file with all is needed for stm32duino integration: the SHARKY_MKR directory to be copied in the /variants directory and the updated boards.txt to be merged with current one.

Please let me know if it is possible to have our board added to your project and let me know if you need more infos or corrections to provided files. I've tested this against ver. 1.6.1

Best Regards, Umberto Allievi

[SharkyMKR_Arduino.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3534976/SharkyMKR_Arduino.zip)
[PG_MDX-MKR-STWBx_DRAFT_01.pdf](https://github.com/stm32duino/Arduino_Core_STM32/files/3534981/PG_MDX-MKR-STWBx_DRAFT_01.pdf)
[UG_MDX-MKR-STWBx_DRAFT_01.pdf](https://github.com/stm32duino/Arduino_Core_STM32/files/3535001/UG_MDX-MKR-STWBx_DRAFT_01.pdf)

fpistm, 2019-08-23:
Hi @uallievi and welcome to GitHub ;)
Thanks for sharing and of course this could be added.
I would prefer you do a PR (Pull request) instead of a zip.
Anyway I will do one and review it.

uallievi, 2019-08-23:
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "
> "name": "View Issue"
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",

Hi Frederic,
thank you very much for your help.
I'm not a github expert, so if you can do this for me the first time
I'll be very pleased.
Next time I'll try to do it by myself
Have a good weekend
Regards, Umberto

uallievi, 2019-08-27:
> Hi @uallievi
>
> I'm currently reviewing your variant and documentations.
>
> Is there HSE and LSE mounted on SHARKY MKR boards? I didn't find info
> in the doc.

Hi Frederic
HSE and LSE are mounted in the sharky module.
Attached the DRAFT documentation and the flyer of the sharky modules
Best regards, Umberto

uallievi, 2019-08-27:
> Ok, thanks for quick answer, I assume HSE is a 8MHz.

NO !!! HSE is 32 MHz
This project has been developed in close contact with ST guys and
revised by them .
Nowadays Sharky is the only module based in STM32WB55
Regards, Umberto

fpistm, 2019-08-27:
Hi @uallievi 
I'm currently reviewing your variant and documentations.
Is there HSE and LSE mounted on SHARKY MKR boards? I didn't find info in the doc.

fpistm, 2019-08-27:
Ok, thanks for quick answer, I assume HSE is a 8MHz.

uallievi, 2019-08-27:
> @uallievi
>
> I've made the PR.
>
> Let me know if it is ok.
>
> I've made some update and fix:
>
>
> D6/D7 was reversed (based on the schematics in your doc)
> SYS_WKUPx updated with correct pin names (see
> https://www.st.com/resource/en/datasheet/stm32wb55ce.pdf)
> Remove stm32wbxx_hal_conf.h
> Apply Astyle
> Define default Wire and SPI pins to the right one. Then you do not
> need to do this in your example:
>
> SPI.setMISO(PB_4); // SHARKYSPI.setMOSI(PB_5); //
> SHARKYSPI.setSCLK(PB_3); // SHARKY
> or
>
> Wire.setSCL(PB_8); // SHARKYWire.setSDA(PB_9); // SHARKY
> About SS pin I've set PB2 (D6)  as it is the HW SSEL but you can set
> any pins.
>
> Finally add a reference in the README.md.
>
> Which url should be set for the Sharky MKR?

Thank you very much for your great work !
I'll look at your updates/fixes to learn and make a better PR next time
!!
the url for sharky MKR is 
https://midatronics.com/Development_Boards/MKR_Sharky_I 
D6/D7 error is a leftover of a previous proto board that had the signal
inverted ..... the correct image is at page 7 and 8 of programmer's
guide (PG_MDX-MKR-STWBx).
The image in the user's guide (UG_MDX-MKR-STWBx) is wrong (pag. 16, 17)
(old version not revisioned, pardon !!)
Best regards, Umberto

uallievi, 2019-08-27:
I've attached the corrected  User Guide with D6/D7 in the right place.
I fear I have confused you with this error, I think that the variants
files with pin mapping where correct.
Can you please chech before pushing the files .....

Thanks, Umberto

uallievi, 2019-08-27:
> No worry.
> I've updated url and set properly D6/D7.

And SSEL is D7 .....

uallievi, 2019-08-27:
> As I use the STM32 notation PB2 and not D6 then it is OK 😉

Great !!

fpistm, 2019-08-27:
@uallievi 
I've made the PR.
Let me know if it is ok.
I've made some update and fix:
* `D6`/`D7` was reversed (based on the schematics in your doc)
* `SYS_WKUPx` updated with correct pin names (see https://www.st.com/resource/en/datasheet/stm32wb55ce.pdf)
* Remove `stm32wbxx_hal_conf.h`
* Apply Astyle
* Define default Wire and SPI pins to the right one. Then you do not need to do this in your example:
```
SPI.setMISO(PB_4); // SHARKY
SPI.setMOSI(PB_5); // SHARKY
SPI.setSCLK(PB_3); // SHARKY
```
or 
```
Wire.setSCL(PB_8); // SHARKY
Wire.setSDA(PB_9); // SHARKY
```
About SS pin I've set PB2 (D6)  as it is the HW SSEL but you can set any pins.

Finally add a reference in the README.md.
Which url should be set for the Sharky MKR?

fpistm, 2019-08-27:
No worry.
I've updated url and set properly D6/D7.

fpistm, 2019-08-27:
As I use the STM32 notation `PB2` and not `D6` then it is OK 😉

commit b84dca7e (2019-08-28): Add SharkyMKR from Midatronics

Fixes #609
PR #614 (2019-08-28): Add SharkyMKR from Midatronics
Fixes #609
closed 2019-08-28 by fpistm