stm32duino/Arduino_Core_STM32 #611: Timer example

theo574, 2019-08-24:
Hi! I'm new in this area and I'm sorry for the question!
Can anyone post an example in which use timer with one callback? 
I know how do that with STM32SW but not with Arduino IDE.

Thanks to every one.

fpistm, 2019-08-24:
Hi, in the next release HardwareTimer library will be available
https://github.com/stm32duino/wiki/wiki/HardwareTimer-library#Examples
Or you can use the repo to get it now

theo574, 2019-08-25:
Hi! thanks a lot! 
I've installed board manager yesterday from Arduino IDE and when i try to compile it tell me that there isn't HardwareSerial.h. 
I repeat you that I'm new with Arduino and i tried to add library by zip file downloaded from this repository in GitHub and it tell me that isn't containing valid library and if i put the first directory in zip file it  tell me that there aren't .h file.
I'm really sorry but in which I'm wrong?

fpistm, 2019-08-25:
HardwareTimer is part of the core. This repo is not a library.
See https://github.com/stm32duino/wiki/wiki/Using-git-repository to replace the instaaled core by the git repo.

theo574, 2019-08-25:
Hi! I'm sorry for confusion.

I followed the guide: deleted old version(that was 1.6.1) and cloned the repo with name of folder 1.6.1, that's ok?
Re-opened Arduino IDE and i see all Nucleo's bords, but when i try to compile i get:

Board Nucleo_144 (platform stm32, package STM32) is unknown
Error compiling for board Nucleo-144.

fpistm, 2019-08-25:
Seems correct. 
This error is returned by IDE. When 2 cores are availables in the same directory. You delete it not simply renamed?
Or the cache files of the IDE are not properly updated.

theo574, 2019-08-25:
Yes, I've deleted the old and cloned repo in directory with name 1.6.1
Maybe before i have to uninstall STM32 from board manager?
How can update correcly?

theo574, 2019-08-25:
I tried now to uninstall from board manager and clone the repo but nothing.

Thank a lot for all!!

fpistm, 2019-08-25:
Well if you follow the wiki this should work.
I use this method since a while without any issue. Could you describe your environment?

ABOSTM, 2019-08-26:
Hi @theo574 , once you have HardwareTimer correctly installed, you can find examples (including one for timer callback) : https://github.com/stm32duino/STM32Examples/tree/master/examples/Peripherals/HardwareTimer

theo574, 2019-08-27:
Hi! I was so busy.
Now I tried  another time but nothing. Always same procedure, delete the version installed with board manager and cloned with name 1.6.1.
My environment is a macbook with partition on ssd with linux.
Tomorrow i will try in mac OS.

theo574, 2019-08-27:
Yeaaaah!
In last time writed above i forgot to set Nucleo in board manager!
Works fine! thank you so much to everyone!

fpistm, 2019-08-27:
Welcome.

closed 2019-08-27 by fpistm

SuperBoss9, 2019-09-16:
Hello everyone,

As the forum doesn't work I have to ask here.

I'm working with PlatfromIO inside VSCode and with Nucleo 103RB. The idea is to make the internal LED blink with a power of a timer and interuption.

In the latest version of Core_STM32 we can see that there is HardwireTimer library that can be used for this work. But in the version that is available inside PlatformIO we have only 1.6.1 and only timer.h library. I searched for the Wiki and other places but couldn't find any reference manual for this lib.

So the question - is there an example how to use timer.h library from 1.6.1 version of Core_STM32? The task is very simple: I'd like to receive interruption from a timer in order to change the internal LED state. And I don't want to use FreeRTOS or just loop routine for that.

Any help is welcomed :)

uzi18, 2019-09-16:
@SuperBoss9 https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/Timebase_callback/Timebase_callback.ino

SuperBoss9, 2019-09-16:
> @SuperBoss9 https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/Timebase_callback/Timebase_callback.ino

This example works with HardwareTimer that is different from ordinary Timer. 1.6.1 doesn't have HardwareTimer only Timer.

uzi18, 2019-09-16:
there is tutorial on wiki/readme how to use development version of this core but will be 1.7.0 available in near future :)

SuperBoss9, 2019-09-16:
> there is tutorial on wiki/readme how to use development version of this core but will be 1.7.0 available in near future :)

But what about the Timer? Is there any description of this lib?

fpistm, 2019-09-16:
https://github.com/stm32duino/wiki/wiki/HardwareTimer-library

fpistm, 2019-09-16:
1.7.0 should ne released this week. Then probably pio will update quickly