stm32duino/Arduino_Core_STM32 #617: HardwareTimer stuck when using refresh()

stas2z, 2019-08-28:
Using latest git snapshot to test hardware timer (trying to port SoftSerialIntAP to this core)
When i try to use resfresh() before resume() timers never start.

It's easy to reproduce it using timebase_callback example provided, just add refresh before resume in it
https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/Timebase_callback/Timebase_callback.ino

 - OS: linux 4.x
 - Arduino IDE version: 1.8.9
 - STM32 core version: 1.7.0 git
 - Upload method: STLink
 - Name: Nucleo 64 F103RE (RB chip burned, replaced with RE and added new variant for it)

fpistm, 2019-08-29:
Hi @stas2z ,
Thanks for reporting.
we will investigate this. Mainly  @ABOSTM our expert 💪 😉

ABOSTM, 2019-08-29:
Hi @stas2z ,
In fact it is not possible to call refresh() before the fisrt call to resume() because timer (the hardware registers) will not be yet initialized.
And there is no reason to do so: refresh() is usefull only while timer is running, in case you modify some parameters like capture/compare to make sure value is taken into account immediately, otherwise value is taken into account at next update event (timer rollover).
So as a general rule call refresh() only after resume(). I will add a comment about that in source code.

PR #618 (2019-08-29): Add comment on refresh() function usage
**Summary**

Add comment on `refresh() `function usage.
By the way fixe typo in `setPrescaleFactor()`.

**Closing issues**

Fix #617
commit 21d3d5c5 (2019-08-29): Add comment on refresh() function usage

By the way fixo typo in setPrescaleFactor()

Fix #617
closed 2019-08-29 by fpistm