stm32duino/Arduino_Core_STM32 #621: STM32RTC

poky99, 2019-08-31:
Good day.
I tried to use the STM32RTC standard RTCClockSelection example on the BluePill stm32f103 board, but each time after the reboot only hours and minutes are saved, the date is set to 01/01/00
Arduino 1.8.9
Core 1.6.1
Win 7 x64.

fpistm, 2019-08-31:
Hi @poky99 
in fact there is a restriction for F1 due to HAL:
https://github.com/stm32duino/Arduino_Core_STM32/blob/151d737c9ec6b21f5903245ddbfc068ae91e5a09/system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_rtc.c#L61-L64

You could try to save it manually in a backup register.

poky99, 2019-09-16:
I was thinking about that.
But what if the device is without power for more than a day?

fpistm, 2019-09-16:
Well I guess the best is to wake up one time per day to update.

fpistm, 2019-10-02:
@poky99 
any update on this?

rtek1000, 2019-10-05:
> unfortunantetly there is no solution for the lost of date because it is a bug in the STM32F1 series.
> There is little comment in one RTC examples of the cube-library:
> @note On STM32F1 families, as there are restrictions on the RTC version V1, date
>       will be lost in all the cases.

Source: https://community.st.com/s/question/0D50X00009XkekSSAR/stm32f103-rtc-vbat-date-resets-after-power-off

Note: A great alternative is to use a DS3231 or DS3232, which already has thermal compensation. See the uRTCLib library, if it interests you: https://github.com/Naguissa/uRTCLib

fpistm, 2019-10-28:
No feedback since a while

closed 2019-10-28 by fpistm