stm32duino/Arduino_Core_STM32 #628: Encoder Mode on Stm32f103c8

pasindu-sandima, 2019-09-03:
Is there feature where I can use the timers in the encoder mode in STM32f103c8. @fpistm 
[image]
[image]

fpistm, 2019-09-03:
@pasindu-sandima 
There is no dedicated `Arduino` API to use it.
You can use the encoder feature using HAL/LL or direct register access:
https://github.com/stm32duino/Arduino_Core_STM32/blob/e638225486c529bf787735d6a14ddd326156f015/system/Drivers/STM32F1xx_HAL_Driver/Inc/stm32f1xx_hal_tim.h#L1692-L1705

ABOSTM, 2019-09-03:
Hi @pasindu-sandima,
The HardwareTimer API https://github.com/stm32duino/Arduino_Core_STM32/pull/576
doesn't support encoder mode. But any contribution is welcome.
Nevertheless it is still possible to use STM32Cube API (HAL or LL) to access encoder mode. 
Note: HAL/LL API can be used directly in your arduino sketch.

pasindu-sandima, 2019-09-03:
@ABOSTM I tried to compile the code generated from CubeMX in arduino. But it did not work.  Is there any specific way that I can use HAL library in arduino sketches??

ABOSTM, 2019-09-03:
@pasindu-sandima, there is no specific way.
I am not sure to understand what you mean by 'compile the code generated from CubeMX in arduino':
CubeMX doesn't manage specific arduino functions like setup() and loop(), so you need to copy piece of code from CubeMX within setup() or loop() in your arduino sketch.
Also what kind of issue do you have ? compiler error, runtime error ?

pasindu-sandima, 2019-09-03:
What is the toolchain that has to be used in CubeMX. To be able to use with Arduino core

pasindu-sandima, 2019-09-03:
I had a compiler error. But now I tried doing direct register editing and now its working. Thanks a lot for your help.

closed 2019-09-03 by pasindu-sandima