stm32duino/Arduino_Core_STM32 #630: Bootloader

zpowellman, 2019-09-03:
Just a question?  Does the Roger Clark Bootloader work with this core?  If not, is there one that does work?
Many Thanks,
Dan Powell

fpistm, 2019-09-03:
Hi @zpowellman ,
yes it works.
See https://github.com/stm32duino/wiki/wiki/Upload-methods#maple-dfu-bootloader

BennehBoy, 2019-09-03:
Yes it does, but only with the board variants supported by that bootloader.

fpistm, 2019-09-04:
is it ok @zpowellman ?

zpowellman, 2019-09-04:
Thanks for the information guys.  I asked because I tried to burn the bootloader into a blue pill.  It worked fine (Maple Com30) until I uploaded the blink app.  It appears to overwrite the bootloader.  So...It looks like I did something incorrectly.  I'll read the documentation you provided and give it a try again.  I was just making sure I wasn't just spinning my wheels. 
Thanks again
Dan

fpistm, 2019-09-04:
You have to enable USB in CDC.

zpowellman, 2019-09-04:
Thanks for the response Fredric.  Not sure what CDC is though.

zpowellman, 2019-09-05:
Thanks, figured it out

closed 2019-09-05 by zpowellman