stm32duino/Arduino_Core_STM32 #633: Procedure for adding new stm32 controllers

mayurharge, 2019-09-04:
Hello,
We were searching for best suitable STM32 controller for our upcoming project and figured out that STM32L433CB series works best for us.
We were trying to use  STM32cubeIDE for development but it's time consuming for starting development on new environment.
I am wondering what is the procedure to add new controller support in existing CORE as we would like to contribute in adding controller support.
Thanks.

fpistm, 2019-09-04:
Hi @mayurharge 
in fact this core support several STM32 series, including STM32L433CB.
Currently there is no variant (board) added with the STM32L433CB.
You can add your board following this Wiki:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

Or add a generic boards in GENF3 menu as it was made for STM32F103Rx:
https://github.com/stm32duino/Arduino_Core_STM32/pull/615

mayurharge, 2019-09-04:
Thanks @fpistm.

fpistm, 2019-09-06:
Welcome @mayurharge

closed 2019-09-06 by fpistm