stm32duino/Arduino_Core_STM32 #635: [New variant] STM32F407ZGT6 Mini & STM32F103ZET6 Mini

dds90, 2019-09-04:
Hi everyone, thanks for your great work!
I would like to ask if you have the possibility to integrate these two variants in the two versions:

[STM32F407ZGT6 Mini](https://stm32-base.org/boards/STM32F407ZGT6-VCC-GND-Large)
&
[STM32F103ZET6 Mini](https://item.taobao.com/item.htm?&id=41867070778)

fpistm, 2019-10-23:
As there is more and more support request for new variant, all are grouped in a single issue to ease tracking: #722.

closed 2019-10-23 by fpistm

dds90, 2019-12-29:
Hi @fpistm,
I'm trying to add a new variant by myself but I cannot compile with it. Please can you help me to tell me where I'm wronging?
[VCC-GND-F103ZET6.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4008414/VCC-GND-F103ZET6.zip)

Thank you so much

fpistm, 2019-12-29:
Hi @dds90 
Could you open a PR, this will be easiest to review. Thanks 
And add a link to the board ( schematics,..)

dds90, 2019-12-29:
I've done, but maybe is incorrect

fpistm, 2019-12-29:
There is no PR about this, only this issue.
An archive is not a Pull Request. Anyway I will check that when I'm back

fpistm, 2020-01-30:
Added thanks #851