stm32duino/Arduino_Core_STM32 #636: Porting marlin timers to new HardwareTimers code

ghent360, 2019-09-05:
Hello Frederic,

I was working on porting the existing Timer code in Marlin to the new HardwareTimer code in 1.6. I encountered the following code in Marlin:

void HAL_timer_set_compare(const uint8_t timer_num, const uint32_t compare) {
  __HAL_TIM_SET_AUTORELOAD(&TimerHandle[timer_num].handle, compare);
  if (HAL_timer_get_count(timer_num) >= compare)
    TimerHandle[timer_num].handle.Instance->EGR |= TIM_EGR_UG; // Generate an immediate update interrupt
}

I can replace the __HAL_TIM_SET_AUTORELOAD call with HardwareTimers[num]->setOverflow(compare, TICK_FORMAT).

I was wondering what to do with the next two lines of code? It seems the goal they were trying to achieve is if one sets the overflow to a value greater than the current counter, generate an update interrupt.
(does this even work? I can see several race conditions)

Should we add something similar to the setOverflow method in the HardwareTimers class?

fpistm, 2019-09-05:
Hi @ghent360,
that's nice you start this porting.
@ABOSTM is our Timer expert. Alexandre could you give your feedback about this? Thanks in advance.

ABOSTM, 2019-09-05:
Hi @ghent360,
First, yes it works: hardware doesn't prevent someone to set Autoreload lower than current value. Counter will continue to increment (in case of upcounting) until it reach its maximum value 0xFFFF (for 16bit timer) and then restart from 0. but generally it is not the expected behavior, because it is more or less like if you have an intermediate period set to max.
Now you can replace ' TimerHandle[timer_num].handle.Instance->EGR |= TIM_EGR_UG;' by a call to refresh() method of HardwareTimer. And replace the HAL_timer_get_count() condition with getCount() method.

You may also consider removing the condition and call systematically refresh() . It will reset counter (restart from 0) with directly the good frequency.

ghent360, 2019-09-07:
Thank you for the advice. I'll test that.

ghent360, 2019-09-08:
Hi Frederic & Alexandre,

I found a small bug in the HT implementation. I sent PR #641 with a
proposed fix.

~V

ghent360, 2019-09-14:
Issues have been resolved.

closed 2019-09-14 by ghent360