stm32duino/Arduino_Core_STM32 #637: Reset takes forever on STM32F103CB / Maple DFU bootloader

mrguen, 2019-09-05:
Hello,

On this configuration:

STM32F103C8BT6 mounted on a barebone board (external resonator 8 MHz)

Arduino IDE:
* Board STM32F103C8(128k)
* USART support : Enabled generic Serial
* USB support: CDC supersed USART
* USB Low/Full speed
* Upload method: Maple DFU bootloader 2.0

1) After reseting it takes a long time before effectively calling setup() since the board USB COM port is de-enumerated.
2) Of course since it is not enumerated there is no more Serial monitor.

So the solution might be to upload using STM32CubeProgrammer but I don't think it is supported by this chip / is this possible to change the ROM bootloader?

fpistm, 2019-09-05:
Hi @mrguen 
I don't understand your pb/question.
If you use an STLink properly connected to SWD pins you can use STM32CubeProgrammer.

mrguen, 2019-09-05:
Yes, but I want the USB support. The basic problem is when using USB serial to monitor datas: whenever the board is reseted it is de-enumerated so 1) it takes forever to reset 2) the Serial Monitor is lost.

fpistm, 2019-09-05:
Unfortunately, I will not be able to help.
This is your custom board.
You probably have to review system clock config to ensure USB clok is 48MHz and USB wiring (PULL up resistor, is there a discovery pin,...)
Finally I do not manage the Maple DFU bootloader .

mrguen, 2019-09-05:
The USB is working fine. The thing is that resetting does toggle between DFU and COM

fpistm, 2019-09-05:
So this probably BL issue and this is not supported here.
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

closed 2019-09-05 by fpistm

mrguen, 2019-09-05:
Ok, I tried without USB / dowloading with STlink (works).

The MCU is not resetting when pulling down NRST.

mrguen, 2019-09-05:
The problem is the same when using the bootloader "hid_generic_pb0" and uploading with STLink.

The board is de-enumerated on reset : it takes forever and the Serial port is lost.

mrguen, 2019-09-05:
Ok, I have got the problem: I used a 10 KOhm pullup. RM008 shows 0 Ohm - it works.

fpistm, 2019-09-05:
Fine.