stm32duino/Arduino_Core_STM32 #644: Changing PWM (analogWrite) resolution

mcer12, 2019-09-09:
Is there any way to change the PWM resolution / range for analogWrite? The STM32s are capable of 16bit PWM but I didn't find any information of how to change that in Arduino IDE. 
If there is an example of this that I missed, please post a link, I will close the issue asap.

fpistm, 2019-09-09:
Hi @mcer12 
See https://www.arduino.cc/reference/en/language/functions/zero-due-mkr-family/analogwriteresolution/
by default it is 8 bits.
So the resolution used as argument of analogWrite is 0-1023 but it is converted to 12 bits internally  thanks:
https://github.com/stm32duino/Arduino_Core_STM32/blob/f2ef8f27b40ee2706db8f9ce27b48a6991047760/cores/arduino/wiring_analog.c#L118

Where:
https://github.com/stm32duino/Arduino_Core_STM32/blob/f2ef8f27b40ee2706db8f9ce27b48a6991047760/cores/arduino/pins_arduino.h#L307-L309

The internal  `PWM_RESOLUTION` can also be redefined using the `build_opt.h`:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

You can also change the PWM freq (note that it will be more accurate in the 1.7.0 as it will use HardwareTimer library):
https://github.com/stm32duino/wiki/wiki/API#frequency

mcer12, 2019-09-09:
Great thank you for this! It didn't even cross my mind to try the Due page.

mcer12, 2019-09-09:
So I found out that the PWM_RESOLUTION is in fact 8bit and not 12bit by default. I used analogWriteResolution(12);
which allowed me to use 12bit values but those were converted back to 8bit. I got around that by setting 
PWM_RESOLUTION 12
PWM_MAX_DUTY_CYCLE 4095
in pins_arduino.h

fpistm, 2019-09-09:
In the next release this will be 12 per defaut. In tbe 1.6.1 it is 8 bits.

fpistm, 2019-09-10:
avoid to edit core files, and as said:

> The internal  `PWM_RESOLUTION` can also be redefined using the `build_opt.h`:
> https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h

This allow to change core  behavior at sketch level.

closed 2019-09-12 by fpistm