stm32duino/Arduino_Core_STM32 #652: Ethernet and SPI

palmerr23, 2019-09-14:
Frederic,

I'm trying to get a WIZ5100 module working on a Black 407 VET6 (Core V1.6.1). I'm quite familiar with the hardware and SPI/Ethernet.

W5100.cpp in the Arduino core Ethernet libraries calls SPI.Begin() - with no parameters. Perfectly fine, except that the default MISO/MOSI pins are 20/21, which map to (in variant.cpp)
#define PA12 20 // USB_DP
#define PA11 21 // USB_DM

To find which pins are actually being used, I added the following code to your SPI.cpp after setSSEL(...) (equivalent to Paul Stoffregen's extra code in the Teensy version).
	//rp added for debug
	uint32_t mosiPin(void)
	{
		return _spi.pin_mosi;
	};
	uint32_t misoPin(void)
	{
		return _spi.pin_miso;
	};
	uint32_t sclkPin(void)
	{
		return _spi.pin_sclk;
	};
	uint32_t sselPin(void)
	{
		return _spi.pin_ssel;
	};

at runtime
mosiPin() returns 21
misoPin() returns 20

What I'm having trouble tracing is how the defaults for these pins are set in the STM32 core. The traditional defaults at runtime are:
PIN_SPI_MOSI and MOSI  = 6
PIN_SPI_MISO and MISO  = 74 
... so these aren't the values being used

Can you provide me with a clue of where the SPI defaults are set for the STM32 variants, so that they can be changed to the most sensible values for this board - which would be the standard pins on the NRF24L01 header:
SPI1-MISO PB4
SPI1-MOSI PB5
SPI1-SCK = PB3

Richard

fpistm, 2019-09-14:
Hi @palmerr23,
The default SPI pins are 
https://github.com/stm32duino/Arduino_Core_STM32/blob/22804538a21692e244911a184babd9a4d41ed4be/variants/BLACK_F407XX/variant.h#L287-L291

palmerr23, 2019-09-14:
Thanks Frederic,

Those tally with what I'm reading at runtime.

I'll keep digging into the code to see why these aren't translating into the right defaults in SPI.h 

Richard

fpistm, 2019-09-14:
As far as I know it was ok. Maybe a misalignment appears when introduced other black F4 variants.

palmerr23, 2019-09-15:
Frederic,

Basic SPI is working fine on its own, without Ethernet. I can't get the SPI library to control the SSEL line directly, but I'll explore whether that's my misconfiguration or the code, later.

The board mappings look fine - at least the SPI definitions haven't slipped as other boards were added. 

After not playing round in the core of Arduino-land for a while, I got confused by the two ways of referring to pins, and forgot about the digitalPinToPinName() mapping. This led me astray while looking for problems.

I'll delve into the Ethernet libraries next to see how they are calling the SPI.

Richard

fpistm, 2019-09-15:
Hi Richard,
I've also reviewed pins mapping and it is OK.
I've also tested the SPI using the NRF connector with a TFT and SD on SPI and it works fine.
So, I guess issue is in the lib.

palmerr23, 2019-09-16:
Great thanks, Frederic.

I've now got it basically working and the max UDP rate I can get is 1000 pps (60 byte packets), which is a bit poor.  No where near even the low end of Paul S's benchmarks (80 kb/Sec on a Uno WIZ5500). I get dropped packets if I don't have a substantial wait on the send side between packets (500+ uS). 

I'm going to play with the SPI speeds next to see if this is part of the issue. 

I've been using a WIZ5100 board so far, and might try a pair of 5500s to see if they are any better.

Unless you care to track me optimizing Ethernet on STM32, we could close this issue now.

Richard

fpistm, 2019-09-16:
Fine.
Probably related to #257.
I close this one as I will not invest on this library and I guess willbe covered by #257.

closed 2019-09-16 by fpistm

uzi18, 2019-09-16:
You can set CS pin to use with Paul's library.
Please remember Wiznet chips has got some limits, like socket and max spi frequency, Wiznet 5500 has got some boost - sequential reading - dma capabilities.

palmerr23, 2019-09-16:
Thanks uzi18.