stm32duino/Arduino_Core_STM32 #656: Maple Mini DFU upload stops working after 1st time

dolfandringa, 2019-09-18:
I'm sorry for posting an issue here, but since stm32duino.com is gone including its forum, and therefore many links on other sites are dead too, I really can't find a solution to this. I'm not 100% sure it is a bug, or if I am doing something wrong. But without docs/forum, it's hard to say.

I have a maple-mini (clone? it does say maple-mini on the board, but got it as a gift from a friend). I want to use it through the Arduino IDE, so I uploaded the latest [STM32duino-bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader) using `./flash/stm32loader.py -p /dev/ttyUSB0 -evw binaries/maple_mini_boot20.bin` using a USB-Serial converter (CP2102 based). The flashing works fine, and it ends up with `Verification OK`. If I then unplug it from the USB-serial converter, remove the BOOT1->GND wire, and plug in the USB port, `/dev/ttyACM0` pops up on my computer and shows me the congratulations sketch. So-far-so-good.

In the arduino IDE I installed the STM32 Cores v1.6.1. I can pick the "Generic STM32F1 series"->"Maple Min F103CB" board. For upload method I use the "Maple DFU bootloader 2.0". I upload a blink sketch, and everything is fine still! Nice! (it shows the Download progress and the blink starts correctly)

But from then on, it stops working. It refuses to upload a second sketch and always goes back to the originally uploaded blink sketch (for testing, all I am doing is changing the blink frequency). The /dev/ttyACM0 port never comes back. It kind of feels like the bootloader isn't working anymore. From that point onwards, any sketch upload results in 
```
/home/me/.arduino15/packages/STM32/tools/STM32Tools/1.3.0/tools/linux/maple_upload ttyACM0 2 1EAF:0003 /tmp/arduino_build_146338/Blink.ino.bin 
Failed to open serial device.
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: Cannot set alternate interface

dfu-util 0.8
<copyright stuff left out for brevity>

Opening DFU capable USB device...
ID 1eaf:0003
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Waiting for /dev/ttyACM0 serial...Done
```

`dfu-util -l` shows:
```
Found DFU: [1eaf:0003] ver=0201, devnum=86, cfg=1, intf=0, path="1-1.1", alt=2, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [1eaf:0003] ver=0201, devnum=86, cfg=1, intf=0, path="1-1.1", alt=1, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [1eaf:0003] ver=0201, devnum=86, cfg=1, intf=0, path="1-1.1", alt=0, name="UNKNOWN", serial="UNKNOWN"
```

For background info, I am doing all of it on Fedora 30 with the following packages installed
```
arm-none-eabi-newlib-3.1.0-3.fc30.noarch
arm-none-eabi-binutils-cs-2.32-1.fc30.x86_64
arm-none-eabi-gcc-cs-c++-9.2.0-1.fc30.x86_64
dfu-util-0.9-7.fc30.x86_64
dfu-programmer-0.7.2-4.fc30.x86_64
glibc-arm-linux-gnu-2.27-4.fc29.x86_64
arm-none-eabi-gcc-cs-9.2.0-1.fc30.x86_64
```
and arduino 1.8.6 tarball directly downloaded from the site (no package manager)

dolfandringa, 2019-09-18:
As a note, I can still flash the bootloader just fine using a USB-Serial converter, which then allows me to upload another sketch (once). Just never more than once.

fpistm, 2019-09-18:
Hi @dolfandringa 
You have to enable `CDC` in `USB support` menu to be able to reset the board in DFU BL mode.

dolfandringa, 2019-09-18:
OK, yeah, that did help. Even though I tried it before. What I find now is that I do sometimes have to manually press the reset button before the programming works. Sometimes I do have to do it, sometimes I don't. But at least turning on CDC with generic Serial does allow me to reprogram without flashing the bootloader again. Thanks @fpistm

closed 2019-09-18 by dolfandringa