stm32duino/Arduino_Core_STM32 #660: Adafruit VL53L0X example does not compile

odo2063, 2019-09-19:
When reporting any issue, please try to provide all relevant information:

**Describe the bug**
Adafruit VL53L0X does not compile for STM (tested for L4R5ZI and G071RB). 
Compiles for other Vendors.

/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb @/tmp/arduino_build_563224/sketch/build_opt.h -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -I/home/teste/Arduino/libraries/Adafruit_VL53L0X/examples/vl53l0x -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/avr -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32 -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/LL -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/usb -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/usb/hid -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/stm32/usb/cdc -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/STM32G0xx_HAL_Driver/Inc/ -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/STM32G0xx_HAL_Driver/Src/ -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/STM32G0xx/ -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -DSTM32G0xx -DARDUINO=10809 -DARDUINO_NUCLEO_G071RB -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"NUCLEO_G071RB\"" -DSTM32G071xx -DHAL_UART_MODULE_ENABLED -D__CORTEX_SC=0 -I/home/teste/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/Core/Include/ -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/CMSIS/Device/ST/STM32G0xx/Include/ -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/system/Drivers/CMSIS/Device/ST/STM32G0xx/Source/Templates/gcc/ -I/home/teste/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/DSP/Include/ -I/home/teste/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/DSP/Source/FilteringFunctions/ -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/variants/NUCLEO_G071RB -I/home/teste/Arduino/libraries/Adafruit_VL53L0X/src -I/home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/libraries/Wire/src /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/vl53l0x_api_calibration.cpp -o /tmp/arduino_build_563224/libraries/Adafruit_VL53L0X/core/src/vl53l0x_api_calibration.cpp.o
/home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/vl53l0x_api_core.cpp: In function 'VL53L0X_Error VL53L0X_calc_sigma_estimate(VL53L0X_DEV, VL53L0X_RangingMeasurementData_t*, FixPoint1616_t*, uint32_t*)':
/home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/vl53l0x_api_core.cpp:1910:47: error: call of overloaded 'abs(FixPoint1616_t)' is ambiguous
   xTalkCorrection  = abs(diff1_mcps/diff2_mcps);
                                               ^
In file included from /home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/cstdlib:75,
                 from /home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/stdlib.h:36,
                 from /home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/wiring.h:24,
                 from /home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/Arduino.h:32,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_i2c_platform.h:1,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_platform.h:35,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_api_strings.h:33,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_api.h:32,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/vl53l0x_api_core.cpp:29:
/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/stdlib.h:70:5: note: candidate: 'int abs(int)'
 int abs (int);
     ^~~
In file included from /home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/cstdlib:77,
                 from /home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/stdlib.h:36,
                 from /home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/wiring.h:24,
                 from /home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/cores/arduino/Arduino.h:32,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_i2c_platform.h:1,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_platform.h:35,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_api_strings.h:33,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/../../vl53l0x_api.h:32,
                 from /home/teste/Arduino/libraries/Adafruit_VL53L0X/src/core/src/vl53l0x_api_core.cpp:29:
/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/bits/std_abs.h:78:3: note: candidate: 'constexpr long double std::abs(long double)'
   abs(long double __x)
   ^~~
/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/bits/std_abs.h:74:3: note: candidate: 'constexpr float std::abs(float)'
   abs(float __x)
   ^~~
/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/bits/std_abs.h:70:3: note: candidate: 'constexpr double std::abs(double)'
   abs(double __x)
   ^~~
/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/bits/std_abs.h:61:3: note: candidate: 'long long int std::abs(long long int)'
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/home/teste/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/arm-none-eabi/include/c++/8.2.1/bits/std_abs.h:56:3: note: candidate: 'long int std::abs(long int)'
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
Bibliothek Adafruit_VL53L0X in Version 1.0.5 im Ordner: /home/teste/Arduino/libraries/Adafruit_VL53L0X  wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: /home/teste/.arduino15/packages/STM32/hardware/stm32/1.6.1/libraries/Wire  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Nucleo-64.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

Steps to reproduce the behavior:
1. Install Adafruit VL53L0X Liberies with Lib Manager
2. Load Example
3. Hit compile button

**Expected behavior**
Compile without errors

**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 1.8.9
 - STM32 core version: 1.6.1

fpistm, 2019-09-19:
Hi @odo2063 
Issue does not come from the core.
This core use the STL C++ instead of macro.
That's why you get this error as no match found for `abs(FixPoint1616_t)` and it propose you some candidate.
So you can fix it on your side or provide a PR to the library, for example.
https://github.com/adafruit/Adafruit_VL53L0X/blob/cc4ca55e938891a2903a9fedd5e10c811da20579/src/core/src/vl53l0x_api_core.cpp#L1910

```patch
-		xTalkCorrection	 = abs(diff1_mcps/diff2_mcps);
+		xTalkCorrection	 = (FixPoint1616_t)abs((long long)(diff1_mcps/diff2_mcps));`
```

odo2063, 2019-09-20:
That I do call a quick and helpful Response! I do really apreciate it.
Since it solved the Problem I close that issue.

closed 2019-09-20 by odo2063