stm32duino/Arduino_Core_STM32 #661: I2C write a buffer bug ?

amirc2013, 2019-09-19:
Im trying to send a buffer over a I2C communication.
this is the code i used :

` Wire.beginTransmission(SCREEN_WRITE_addr);
  Wire.write(addr);
  unsigned char* ptr = data;
  for (int i = 0 ; i < len ; i++) {
    Wire.write(*ptr);
     if (DEBUG) {
      Serial.print("WE WROTE :");
      Serial.println(*ptr,HEX);
    }
    ptr++;
  }
  Wire.endTransmission(true);`

some how only the last byte has been sent ... why ?

fpistm, 2019-09-20:
Hi @amirc2013 
could you be more precise about your issue as described in the GitHub issue template:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible (no extra code, extra hardware,...)

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.4.0]
 - Tools menu settings if not the default: [e.g. Newlib Standard, No Serial]
 - Upload method: [e.g. STLink]

**Board (please complete the following information):**
 - Name: [e.g. Nucleo F429ZI]
 - Hardware Revision: [e.g. Rev B]
 - Extra hardware used if any: [e.g. X-Nucleo IKS01A1]

**Additional context**
Add any other context about the problem here.

Full sketch. Which I2C device you used. how it is wired. Do you use PULLUP resistors on the line...

fpistm, 2019-09-24:
@amirc2013 
without any feedback I will close this issue.

amirc2013, 2019-09-24:
Bug :
Can't write a big buffer to the i2c line. Seems like there is a limitation which I can't find where or why.

fpistm, 2019-09-24:
@amirc2013 
OK but you do not provide enough information.
Which board you used, which I2C hardware, how it is wired.
As said I could not help without more info. 
I do not ask to close the issue, only to have more info to be able to help...

fpistm, 2019-09-24:
No worries ;)

Pararera, 2019-09-24:
I2C buffer is 32 bytes. Maybe that is the problem?

fpistm, 2019-09-25:
Right by default `I2C_BUFFER` is 32 bytes but it is a dynamic allocated buffer which can growth if needed.
I ask about more info because if @amirc2013  used the GitHub repository then he could met an issue I've seen after repeated start feature introduction. For the 1.7.0, I've revert it.
So, while I do not get all info I could not help.

amirc2013, 2019-09-25:
Writing and uploading an answer right now. sorry for the delay

‫בתאריך יום ג׳, 24 בספט׳ 2019 ב-11:42 מאת ‪Frederic Pillon‬‏ <‪
notifications@github.com‬‏>:‬

amirc2013, 2019-09-25:
This is the simple formation of the code Im using for writing to the I2C.

My Real Program is really big and it is not understandalbe
so for this issue I will just use this instead.

`static inline void i2c_write(
  unsigned short addr,
  unsigned char *data,
  unsigned short len)
{
  Wire.beginTransmission(SLAVE_addr);
  Wire.write(addr);
  int ret = Wire.write((const uint8_t *)data,(size_t)len);
  Wire.endTransmission(true);
}`

**The bug :**
When I'm writing a big data to this function - I just cant see then on my logic analyzer.
Only when I cut them down to 0x1000/10 bytes I will be able to see them.
This is super weird. Cause I need to write in 0x1000 chunk size...

My board : Nucleo L4R5ZI
OS: Windows
Arduino IDE version: 1.8.10
STM32 core version: **How can I check that ?**
Tools menu settings if not the default. Newlib Nano (default)
Upload method: Mass storage

Hardware Revision:**How can I check that ?**

I don't use any extra pull-up resistors.. Only connect a micro-controller to my STM with level shifter in the middle.

fpistm, 2019-09-25:
> STM32 core version: **How can I check that ?**

How did you install it? Boards manager? Using git repo? Anyway you can check here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/fb579427481c7d6b39b744f304ef3f7af5fc9f78/cores/arduino/stm32/stm32_def.h#L5-L21

> Hardware Revision:**How can I check that ?**
on the sticker you should have something like `MB1312 X` where `X` is is the revision, anyway this is not very important in our case.

> I don't use any extra pull-up resistors.. Only connect a micro-controller to my STM with level shifter in the middle.

Could you be more precise about this? which microconttoller and level shifter? how are they wired.
I2C lines require PULL up so ensure to have them on your wiring.

amirc2013, 2019-09-25:
I installed it with Boards manager - yes

This is my version:

` */`
`#define STM32_CORE_VERSION_MAJOR    (0x01U) /*!< [31:24] major version */`
`#define STM32_CORE_VERSION_MINOR    (0x06U) /*!< [23:16] minor version */`
`#define STM32_CORE_VERSION_PATCH    (0x01U) /*!< [15:8]  patch version */`
`/*`

About the wiring :

It is an enterprise micro-controller and I just can't say which one is it yet :/
I'm checking if I can communicate with it.

I don't have a  resistors on the I2C but maybe on the board of the other micro-controller  it has one ?
I checked my level shifter (Logic Level Converter) and it doest have any name on it.

I can put some resistor on the line but as I saw on the logic analyzer - all the other part of the i2c communication are working.

uzi18, 2019-09-25:
any photos or schematics?

amirc2013, 2019-09-25:
[image]

:(
I know it looks bad.

I just can't talk about the micro controller.

Just assume I extracted a controller from some enterprise device and I want to communicate with it.
Moreover, I just want to update its firmware.

amirc2013, 2019-09-25:
new update :
It works when i write 0x1000-3 ....
WTF  ?

maybe optimization ?

fpistm, 2019-09-26:
I've found the issue.
Well, based on Arduino API it is not possible to handle more than 255 bytes as API based in uint8_t.
Anyway, I think there is a misalignment between HAL and Arduino layer.

amirc2013, 2019-09-26:
Well, I'm glad that I point the issue in the right place.
Thank you.
Waiting for updates :)

fpistm, 2019-09-26:
I will try to see if I'm able to remove uint8_t date size limitations as HAL can handle uint16_t but I could not promise. In that case you would be able to send your 0x1000 in 16 times.

amirc2013, 2019-09-26:
I wish you luck, it is really important to me.
Thank you for the help :)

urosh-dev, 2019-11-08:
Any updates on that. I have the same issue. I cant send string with Wire. for example Wire.write("hello");
Using blue pill and raspberry pi with command from raspberry pi :
bus.read_i2c_block_data(address,4)
or
bus.read_byte(address)
Blue pill just hangs up.

fpistm, 2019-11-08:
> Any updates on that. I have the same issue. I cant send string with Wire. for example Wire.write("hello");
> Using blue pill and raspberry pi with command from raspberry pi :
> bus.read_i2c_block_data(address,4)
> or
> bus.read_byte(address)
> Blue pill just hangs up.

This is not related. Here the issue is for buffer higher than 32 in receive

fpistm, 2019-12-16:
#### Current status:
* Master:
  * can write up to 255 (limited by Arduino API `uint8_t`)
  * can read up to 32

* Slave:
  * can read/write up to 32

32 is the Arduino standard. The Master Tx has been extend to support the X-Shield NFC nevertheless
the Rx should be able to receive also 255 bytes. And Slave should be able to write also up to 255.

ABOSTM, 2020-01-03:
Hi @amirc2013,
with implementation of PR  #853, the status is 255 bytes transfer in any direction
(254 with the bug identified see warning).
But still limited by Arduino API `uint8_t`

closed 2020-01-06 by fpistm