stm32duino/Arduino_Core_STM32 #662: STM32L010rb support

vasav123, 2019-09-22:
**Is your feature request/improvement related to a problem? Please describe.**
Can the stm32L010rb be supported as one of the boards?

fpistm, 2019-09-23:
Hi @vasav123 
in fact this core support several STM32 series, including STM32L0xx.
Currently there is no variant (board) added with the STM32L010RB.
I do not know if you talk about a specific board or having a generic one based on this MCU.

You can add your board following this Wiki:
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)

Here an example how to add a generic one:
https://github.com/stm32duino/Arduino_Core_STM32/pull/615

uzi18, 2019-09-23:
like NUCLEO-L010RB ?

vasav123, 2019-09-23:
@uzi18 Yes, I meant the Nucleo-L010RB board. I also was hoping that a generic STM32L0 could be supported like it is for the F0 series. 
@fpistm thanks for your help, I'll take a look at that and see If I can try to make board myself.

fpistm, 2019-10-23:
As there is more and more support request for new variant, all are grouped in a single issue to ease tracking: #722.

PR #1384 (2021-05-08): variant: add Nucleo L010RB
Fixes #662
commit 8c83c138 (2021-05-08): variant: add Nucleo L010RB

Fixes #662
closed 2019-10-23 by fpistm