stm32duino/Arduino_Core_STM32 #664: Version 1.7 broke the USB support for F103C8

TByte007, 2019-09-24:
I upgraded to 1.7 from 1.6.1 and the USB on all BluePills that I have stopped working. Switched back to 1.6.1 and everything works. 1.7 is basically unusable on the most used maker's board now. No code is changed in the sketches - just recompiled with 1.7.

fpistm, 2019-09-24:
Hi @TByte007 
could you be more precise on your setup.
On my side, USB works with 1.7.0. Tested on Windows and Linux.

TByte007, 2019-09-24:
I tested it on 2 boards - the moment I compile a simple program which only have a serial port in 1.7 and windows 10 says "There is a problem with your USB device" ... something along those lines. Reverted back to 1.6.1 and re flashed the boards. Everything worked again.

fpistm, 2019-09-24:
Well, this will be hard to help as I could not reproduced.

Do you have the right pull up resistor for USB on those BP?
Driver issue?

TByte007, 2019-09-24:
How could be a driver issue if it works with 1.6.1 ? I just reinstalled 1.7 and this is the result.

[image]

fpistm, 2019-09-24:
@TByte007 
On which USB port type do you connect the BP? 
Could you try to comment this line:
https://github.com/stm32duino/Arduino_Core_STM32/blob/5e43e63e80519c971217285718539ef3b6189a14/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_core.c#L401

TByte007, 2019-09-24:
It works again - Thank you ! What does that function do / change ? 

Btw can you put "__attribute__ ((unused))" before those params coz it's annoying and fills the screen and I really want -Wall parameter - it saved me few times  :) 

In file included from C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino\stm32\HAL\stm32yyxx_hal_tim_ex.c:5:
C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c: In function 'HAL_TIMEx_RemapConfig':
C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:60: warning: unused parameter 'htim' [-Wunused-parameter]
 HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
                                         ~~~~~~~~~~~~~~~~~~~^~~~
C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Drivers/STM32F1xx_HAL_Driver/Src/stm32f1xx_hal_tim_ex.c:1734:75: warning: unused parameter 'Remap' [-Wunused-parameter]
 HAL_StatusTypeDef HAL_TIMEx_RemapConfig(TIM_HandleTypeDef *htim, uint32_t Remap)
                                                                  ~~~~~~~~~^~~~~
In file included from C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\cores\arduino\stm32\usb\usb_device_ctlreq.c:3:
C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c: In function 'USBD_CtlError':
C:\Users\tbyte\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_ctlreq.c:853:42: warning: unused parameter 'req' [-Wunused-parameter]
                    USBD_SetupReqTypedef *req)
                    ~~~~~~~~~~~~~~~~~~~~~~^~~

TByte007, 2019-09-25:
@fpistm btw the port doesn't look like an USB 3.0 port (it's black not blue) and it's mounted on the board and not on an extension cable so it should be USB 2.0. Sorry that I didn't checked that before I was too excited that the patch you gave me worked :)

fpistm, 2019-09-25:
No worry. I just search why on certain USB port a stall is present.

fpistm, 2019-09-26:
@TByte007 
about warning they are all related to HAL/LL. I do not modify them as they are simply copied from the Cube Fw.
Sometimes I submit internally issue but this is very time consuming and in general they correct them.
Anyway do not hesitate to submit an issue on the cube as they are now on GitHub:
https://github.com/STMicroelectronics/STM32CubeF1

PR #670 (2019-09-26): [USB] Fix EPO STALL issue in STM32 USB Device library
Fixes #664
commit 99c9ff18 (2019-09-26): [USB] Fix EPO STALL issue in STM32 USB Device library

USB Specification EP0 should never STALL during setup stage.
Device is not properly setup if STALL present.

Fixes #664
closed 2019-09-26 by fpistm