stm32duino/Arduino_Core_STM32 #665: How to use I2C2?

Pararera, 2019-09-24:
Hi guys,
I'm using STM32F103C8(BluePill?). Until now I used STM32 packacge from Dan Drown's website. Now I'm watching at code and I don't understand how to use I2C on pins PB11 and PB10(also know as I2C2)? I use I2C for SH1106 display, DS3231 RTC and AT24C32 EEPROM chip.

Pararera, 2019-09-24:
Found this -> https://github.com/stm32duino/wiki/wiki/API#i2c
`TwoWire Wire2(PB3, PB10);` PB3 isn't SDA of I2C2. Is that mistake or?

TByte007, 2019-09-24:
> Found this -> https://github.com/stm32duino/wiki/wiki/API#i2c
> `TwoWire Wire2(PB3, PB10);` PB3 isn't SDA of I2C2. Is that mistake or?

It's for different board. You need this `TwoWire Wire2(PB11, PB10);`

Pararera, 2019-09-24:
Btw that is hardware I2C or bit banging?

TByte007, 2019-09-25:
PB6/PB7 and PB11/PB10 should be hardware as far as I know (and all the diagrams show)

fpistm, 2019-09-25:
Hi @Pararera 
we support only hardware I2C.
The link provided by @TByte007 (thanks 😉) is only an example. Up to you to set the correct I2C pins depending to the MCU you used.
So for using I2C2 on BP:
`TwoWire Wire2(PB11, PB10);`

closed 2019-09-25 by Pararera