stm32duino/Arduino_Core_STM32 #666: Toolchain is not ready

Pararera, 2019-09-24:
Hi there,
I use Sublime 3 and Stino plugin. Sublime said installation was done, but when I try to build sketch then I get this message.

> [Error] Toolchain is not ready. Please build the sketch after the toolchain installation done.

Installed packages:
CMSIS 5.5.1.
GCC ARM none eabi 8.2.1.-1.7
STM32-1.7
STM32Tools-1.3.1

I use Windows 10 Pro(latest build). Whole folder has 815MB.

fpistm, 2019-09-25:
Hi @Pararera 
unfortunately we support only Arduino IDE and this working fine with it. I've tested other IDE for debud like VScode + Arduino plugin or Eclipse + Sloeber and it works.
You probably need some customization. I advise you to ask support here if I'm not wrong:
https://github.com/Robot-Will/Stino

Pararera, 2019-09-25:
I have same problem with other STM32 core, but I don't know what I did to fix it.
I hope you will add support for Sublime since it is very great IDE

closed 2019-09-25 by Pararera

fpistm, 2019-09-25:
As said you should ask support directly on plugin.
Unfortunately, support Arduino is already a big stuff. Anyway, if you want contribute you can add a chapter about this IDE in the Wiki.
https://github.com/stm32duino/wiki/wiki/How-to-debug