stm32duino/Arduino_Core_STM32 #667: Using SPI Class crash system

arnold-b, 2019-09-25:
STM32F407VGT

Sample:
#define TFT_CS         PB4
SPIClass HWSPI = SPIClass(PA7,PA6,PA5,TFT_CS);
void setup() {
  Serial.begin(115200);
  Serial.println("Test!");
  pinMode(PB4, OUTPUT);
 HWSPI.begin(); //Initialize the SPI_1 port.
Serial.println("HWSPI");   // No output anymore
   }

void loop(void) {   
Serial.println("Loop IO");

delayMicroseconds(100); 
digitalWrite(PB4, HIGH);
HWSPI.transfer(0xb4);
digitalWrite(PB4, LOW);
}

Output Serial: 
Test!

Thats it

fpistm, 2019-09-25:
Hi @arnold-b 
That is normal as you don't follow SPI API.
https://github.com/stm32duino/wiki/wiki/API#spi

PB4 is not a hardware CS, so you should not give it to constructor:

```C++
#include "SPI.h"
#define TFT_CS         PB4

SPIClass HWSPI = SPIClass(PA7, PA6, PA5);

void setup() {
  Serial.begin(115200);
  Serial.println("Test!");
  pinMode(TFT_CS, OUTPUT);
  HWSPI.begin(); //Initialize the SPI_1 port.
  Serial.println("HWSPI");   // No output anymore
}

void loop(void) {
  Serial.println("Loop IO");
  delayMicroseconds(1000000);
  digitalWrite(TFT_CS, HIGH);
  HWSPI.transfer(0xb4);
  digitalWrite(TFT_CS, LOW);
}
```

arnold-b, 2019-09-25:
OK, thank you. I will test it

closed 2019-09-27 by fpistm