stm32duino/Arduino_Core_STM32 #669: How to use an SPI2? (Incompatibility between cores is very bad)

rtek1000, 2019-09-26:
**Is your feature request/improvement related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I can't use SPI2

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
For the other core you can normally use SPI2 with the code below

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/libraries/SPI/examples/using_SPI_ports/using_SPI_ports.ino

`SPIClass SPI_2(2); //Create an instance of the SPI Class called SPI_2 that uses the 2nd SPI Port
`

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I tried unsuccessfully to follow the library descriptions:

https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/SPI

```
#define MOSI2 PB15
#define MISO2 PB14
#define SCLK2 PB13
#define SS2 PB12

//SPIClass(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel)
SPIClass SPI_2(MOSI2, MISO2, SCLK2, SS2);
```

**Additional context**
Add any other context, example or screenshots about the feature request here.

I don't know which core was started first, but you 2 would have to follow a single pattern for these basic pieces of hardware, every time someone tries to build code using libraries available on the internet, one works on one core, another works on another core. , but you can't put the libraries together in the same sketch, resulting in project abandonment

Any contribution is welcome, so do not hesitate to submit a PR.

rtek1000, 2019-09-26:
I have modified the code, based on the link below, and apparently there is a clock signal on the SPI2 port, but it is still not possible to access W25Q64 memory, while on the other core it works normally.

```
#define MOSI2 PB15
#define MISO2 PB14
#define SCLK2 PB13
#define SS2 PB12

//SPIClass(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel)
SPIClass SPI_2 = SPIClass(MOSI2, MISO2, SCLK2, SS2);

void setup() {
  SPI_2.begin();

  while(1){
    SPI_2.transfer(0xb4);
  }
}
```

https://github.com/stm32duino/Arduino_Core_STM32/issues/667

rtek1000, 2019-09-26:
Unfortunately the code below worked, but the next code, doesn't work, only in the other core. Only direct memory access works. File system access does not work. 

Works:
```
// The MIT License (MIT)
// Copyright (c) 2019 Ha Thach for Adafruit Industries

#include "SdFat.h"
#include "Adafruit_SPIFlash.h"

#define MOSI2 PB15
#define MISO2 PB14
#define SCLK2 PB13
#define SS2 PB12

//SPIClass(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel)
SPIClass SPI_2 = SPIClass(MOSI2, MISO2, SCLK2, SS2);

Adafruit_FlashTransport_SPI flashTransport(SS2, &SPI_2);

Adafruit_SPIFlash flash(&flashTransport);

// the setup function runs once when you press reset or power the board
void setup()
{
  SPI_2.begin();
  
  Serial.begin(115200);
  while ( !Serial ) delay(10);   // wait for native usb

  flash.begin();
  
  Serial.println("Adafruit Serial Flash Info example");
  Serial.print("JEDEC ID: "); Serial.println(flash.getJEDECID(), HEX);
  Serial.print("Flash size: "); Serial.println(flash.size());
}

void loop()
{
  // nothing to do
}
```
Did not work:
```
// Adafruit SPI Flash FatFs Format Example
// Author: Tony DiCola
//
// !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
// !!  NOTE: YOU WILL ERASE ALL DATA BY RUNNING THIS SKETCH!  !!
// !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
//
// Usage:
// - Modify the pins and type of fatfs object in the config
//   section below if necessary (usually not necessary).
// - Upload this sketch to your M0 express board.
// - Open the serial monitor at 115200 baud.  You should see a
//   prompt to confirm formatting.  If you don't see the prompt
//   close the serial monitor, press the board reset buttton,
//   wait a few seconds, then open the serial monitor again.
// - Type OK and enter to confirm the format when prompted.
// - Partitioning and formatting will take about 30-60 seconds.
//   Once formatted a message will be printed to notify you that
//   it is finished.
//
#include <SPI.h>
#include <SdFat.h>
#include <Adafruit_SPIFlash.h>

// Since SdFat doesn't fully support FAT12 such as format a new flash
// We will use Elm Cham's fatfs f_mkfs() to format
[... 133 lines cut ...]
  {
    case CTRL_SYNC:
      flash.syncBlocks();
      return RES_OK;

    case GET_SECTOR_COUNT:
      *((DWORD*) buff) = flash.size()/512;
      return RES_OK;

    case GET_SECTOR_SIZE:
      *((WORD*) buff) = 512;
      return RES_OK;

    case GET_BLOCK_SIZE:
      *((DWORD*) buff) = 8;    // erase block size in units of sector size
      return RES_OK;

    default:
      return RES_PARERR;
  }
}
}
```

https://github.com/adafruit/Adafruit_SPIFlash

fpistm, 2019-09-26:
@rtek1000 
This core follow Arduino API's while other not.
Each core is different and have a different goal.
This one support 13 STM32 series while other support only one (I told only one because they are 2 cores supported in one repo) which are separated (STM32F1/ and STM32F3).
Try to build default official Arduino sketch and you will see that Arduino_STM32 could not build for several of them. That's why it provide lot of example to adapt those examples.
Anyway for advanced user who want use STM32F1 it is probably the best choice :wink:

For ref:
https://github.com/stm32duino/wiki/wiki/API#spi

Anyway if you want add compatibility this probably could be done by adding some API. But as said this core support 13 series and more soon. So, my feeling is this is not a good option.
They are different that's all.

rtek1000, 2019-09-26:
Thank you,

I'm trying to adapt to this core, but it's hard, the wiki example can't even be compiled.

> #include <SPI.h>
> //            MOSI  MISO  SCLK
> SPIClass SPI3(PC12, PC11, PC10);
> 
> void setup() {
>   SPI3.begin(2); //Enables the SPI3 instance with default settings and attaches the CS pin  
>   SPI3.beginTransaction(1, settings); //Attaches another CS pin and configure the SPI3 instance with other settings  
>   SPI3.transfer(2, 0x52); //Transfers data to the first device
>   SPI3.transfer(1, 0xA4); //Transfers data to the second device. The SPI3 instance is configured with the right settings  
>   SPI3.end() //SPI3 instance is disabled
> }

Now I'm trying to remap SPI1 to pins PB5, PB4, PB3 and A15.

I found someone else with the same question, so it may mean that the wiki is not helping users. I know that for anyone writing library code it's easy to find anything, but for outsiders, things need to be less generic, and examples need to work without being edited.

https://github.com/stm32duino/Arduino_Core_STM32/issues/370

I saw that the alternate pins of SPI1 are in the file below, but how can I remap SPI1?
Arduino_Core_STM32/variants/BLUEPILL_F103C8/PeripheralPins.c

fpistm, 2019-09-27:
The wiki example is generic. It is up to user to set the correct pins. 
To remap the default SPI instance simply follow the Wiki:
https://github.com/stm32duino/wiki/wiki/API#change-default-spi-instance-pins

rtek1000, 2019-09-27:
Thanks,

This code did not work below, when I inform the particular SS, the clock signal is not generated in the SPI, but pin PB0 is set at high level:

```
void setup()
{
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);
  //SPI.setSSEL(PA15);
  SPI.begin(PB0);
  while(1){
    SPI.transfer(0x01);
  }
}
```

If I use the hardware SS pin, then the SPI port generates the clock signal:
```
void setup()
{
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);
  SPI.setSSEL(PA15);
  SPI.begin();
  while(1){
    SPI.transfer(0x01);
  }
}
```
How can I make the SPI port work with SS that is not an SS hardware pin?

fpistm, 2019-09-27:
In case you want the SPI class drive the software CS you have to pass it as an argument else it assume you drive manually the CS.
```
void setup()
{
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);
  SPI.begin(PB0);
  while(1){
    SPI.transfer(PB0, 0x01);
  }
}
```

or 
```
void setup()
{
  SPI.setMOSI(PB5);
  SPI.setMISO(PB4);
  SPI.setSCLK(PB3);
  SPI.begin(PB0);
  digitalWrite(PB0, LOW);
  while(1){
    SPI.transfer(0x01);
  }
}
```

rtek1000, 2019-09-27:
The sample (CardInfo*) SD card access code worked with custom SS.

>   created  28 Mar 2011
>   by Limor Fried
>   modified 9 Apr 2012
>   by Tom Igoe

I changed the flash memory wiring to work with SS hardware, but it didn't work when trying to format memory with FatFS, can only access basic information. Therefore, the flash memory access library (Adafruit_SPIFlash) does not accept to operate with this core (for now), even on SPI1.

Anyway thank you.

rtek1000, 2019-10-02:
I found something interesting, with the "Arduino_STM32" core the flash_info (modified*) sketch reads 0xFF after chip erase. But with "Arduino_Core_STM32" core, the first byte is 0x7F for all pages, this may explain why it doesn't work on this core. Note: using SPI1 remaped

* sketch modified:

```
// The MIT License (MIT)
// Copyright (c) 2019 Ha Thach for Adafruit Industries

#include "SdFat.h"
#include "Adafruit_SPIFlash.h"

#define SS1 PA15

Adafruit_FlashTransport_SPI flashTransport(SS1, &SPI);

Adafruit_SPIFlash flash(&flashTransport);

#define MAXPAGESIZE 256
uint8_t buffer[MAXPAGESIZE];

//Serial = &Serial1;
#if defined(__STM32F1__)
#define mySerial Serial1
#else
#define mySerial Serial
#endif

// the setup function runs once when you press reset or power the board
void setup()
{
#if defined(__STM32F1__)
  // Remap USART 1
  //  afio_remap(AFIO_REMAP_USART1);

  // Remap SPI 1
  // Source: https://community.platformio.org/t/pio-arduino-stm32f103cbt6-remap/6786/5
  afio_cfg_debug_ports(AFIO_DEBUG_SW_ONLY); // PB3 free
  afio_remap(AFIO_REMAP_SPI1);

  gpio_set_mode (GPIOB, 3, GPIO_AF_OUTPUT_PP);
  gpio_set_mode (GPIOB, 4, GPIO_INPUT_FLOATING);
  gpio_set_mode (GPIOB, 5, GPIO_AF_OUTPUT_PP);
#else
  SPI.setMOSI(PB5); // using pin number PYn
  SPI.setMISO(PB4); // using pin name PY_n
  SPI.setSCLK(PB3); // using pin number PYn
  SPI.setSSEL(PA15); // using pin number PYn
#endif

  delay(1000);
  
  mySerial.begin(115200);
  //  mySerial.begin(2000000);
  //while ( !Serial ) delay(10);   // wait for native usb

  pinMode(PB0, OUTPUT);
  digitalWrite(PB0, LOW);

  mySerial.println("Adafruit Serial Flash Info example");

  delay(1000);

  flash.begin();

  //  mySerial.println("Adafruit Serial Flash Info example");
  mySerial.print("JEDEC ID: "); mySerial.println(flash.getJEDECID(), HEX);
  mySerial.print("Flash size: "); mySerial.println(flash.size());
  mySerial.print("Flash status register: "); mySerial.println(flash.readStatus());
  mySerial.print("Flash status register 2: "); mySerial.println(flash.readStatus2());

[... 142 lines cut ...]

> Adafruit Serial Flash Info example
> JEDEC ID: EF4017
> Flash size: 8388608
> Flash status register: 0
> Flash status register 2: 0
> Dumping FLASH
> Reading page: 0
> 7F FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
> FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
>

fpistm, 2019-10-02:
@rtek1000 
unfortunately, I will not be able to help on this as I can not reproduced.
If you could, you should use a logic analyzer to spy SPI transfer and also try to debug.

fpistm, 2019-10-02:
Moreover please avoid to reopen an issue for an other subject. This is very confusing.
So please open a new one with appropriate subject.

closed 2019-10-02 by fpistm

rtek1000, 2019-10-02:
Ok

rtek1000, 2019-10-02:
Seems to be just an SPI port speed issue, for some reason after slowing down the 0x7F is no longer found:

Added to main code after flash.begin();

```C++
  flashTransport.setClockSpeed(200*1000000UL); // 200MHz: fail (1st: 0x7F)
```
```C++
  flashTransport.setClockSpeed(1*1000000UL); // 1MHz: ok (all: 0xFF)
```

Main code readBuffer:
```C++
uint16_t r = flash.readBuffer (page * pagesize, buffer, pagesize);```

Library Subroutines:
```C++
uint32_t Adafruit_SPIFlash::readBuffer(uint32_t address, uint8_t *buffer, uint32_t len)
{
  if (!_flash_dev) return 0;

  waitUntilReady();

  SPIFLASH_LOG(address/512, len/512);

  return _trans->readMemory(address, buffer, len) ? len : 0;
}
```

```C++
bool Adafruit_FlashTransport_SPI::readMemory(uint32_t addr, uint8_t *data, uint32_t len)
{
  digitalWrite(_ss, LOW);
  _spi->beginTransaction(_setting);

  _spi->transfer(SFLASH_CMD_READ);

  // 24-bit address MSB first
  _spi->transfer((addr >> 16) & 0xFF);
  _spi->transfer((addr >> 8) & 0xFF);
  _spi->transfer(addr & 0xFF);

  while(len--)
  {
    *data++ = _spi->transfer(0xFF);
  }

  _spi->endTransaction();
  digitalWrite(_ss, HIGH);

  return true;
}

```

I will try to use the library with FAT to test better.

rtek1000, 2019-10-02:
That's it, the FAT part is working now, just modify the speed.

Added to main code after flash.begin();
```C++
flashTransport.setClockSpeed(1*1000000UL); // 1MHz: ok (all: 0xFF)
```

Scketch SdFat_format results on serial monitor:

> Adafruit SPI Flash FatFs Format Example
> Flash chip JEDEC ID: 0xEF4017
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> This sketch will ERASE ALL DATA on the flash chip and format it with a new filesystem!
> Type OK (all caps) and press enter to continue.
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> Creating and formatting FAT filesystem (this takes ~60 seconds)...
> Formatted flash!
> Flash chip successfully formatted with new empty filesystem!
> 

Scketch SdFat_full_usage results on serial monitor:

> Adafruit SPI Flash FatFs Full Usage Example
> Flash chip JEDEC ID: 0xEF4017
> Mounted filesystem!
> Test directory not found, creating...
> Created directory!
> Creating deep folder structure...
> Creating /test/foo/bar
> Created directory!
> Creating /test/foo/baz
> Created directory!
> Opened file /test/test.txt for writing/appending...
> Wrote to file /test/test.txt!
> First line of test.txt: Hello world!
> 
> Total size of test.txt (bytes): 57
> Current position in test.txt: 14
> Available data to read in test.txt: 43
> File name: use getName()
> Is file a directory? No
> Entire contents of test.txt:
> Hello world!
> Hello number: 123
> Hello hex number: 0x7B
> Listing children of directory /test:
> - foo (directory)
> - test.txt
> Deleting /test/foo/test2.txt...
> Deleted file!
> Deleting /test directory and everything inside it...
> Test directory was deleted!
> Finished!
>

rtek1000, 2019-10-02:
@fpistm 
Would it be a problem with the core?

Why does the other core work with SPI set to 133MHz and this core here only works at 35MHz or below?

fpistm, 2019-10-02:
Roger's core use SPI with DMA.
An issue is already opened to make some SPI optimizations, see #257

rtek1000, 2019-10-02:
Nice