stm32duino/Arduino_Core_STM32 #671: [New Variant] Nucleo L452RE-P Support

jgromes, 2019-09-26:
Hi, I'd like to add support for Nucleo L452RE-P. There's already a variant for L452RE, and it looks like code compiled for L452RE does upload to L452RE-P. Since the LED in the Blink example doesn't blink, I guess there's some difference in pin mapping. 

Is it enough to add a few #ifdef statements to the variant files (like with L496ZG and L496ZG-P), and the new variant into boards.txt, or is there something else as well? Thanks!

fpistm, 2019-09-27:
Hi @jgromes 
looking at your point, I saw some minor issue so I've updated it and add the -P variant.
I've tested Blink, I2C, Serial, SPI and Analog.

jgromes, 2019-09-27:
Wow, that was fast - I can confirm it's working, will let you know if any issues pop up.

Thanks!

closed 2019-09-27 by jgromes

fpistm, 2019-09-27:
Welcome do not hesitate to open other issue if you meet one.

PR #673 (2019-09-27): Add NUCLEO-L452RE-P support
Fixes #671
commit 964576b1 (2019-09-27): Add NUCLEO-L452RE-P support

Fixes #671