stm32duino/Arduino_Core_STM32 #672: Help with hardware timers

TadyTheFish, 2019-09-26:
Hello

I'm trying to write some code using hardware timers and I have a lot of problems since there are no examples.
I would like some help with timer interrupts.
`
 HardwareTimer Timer2 = HardwareTimer(TIM1);

  void setup(){
          Timer2.setPrescaleFactor(72);
          Timer2.setMode(1, TIMER_OUTPUT_COMPARE);
          Timer2.setCaptureCompare(1, some_delay_in_microseconds); 
          Timer2.setOverflow(startDelayOneFour+1);
          Timer2.attachInterrupt(1, timerISR);
          Timer2.refresh();
          Timer2.setCount(0);
       }
  void loop(){

  }
  void  timerISR(){ 
    Timer2.detachInterrupt(1); // Stop the timer from firing this ISR
     //do stuff...
   }`

This does not compile....
`error: invalid conversion from 'void (*)()' to 'void (*)(HardwareTimer*)' [-fpermissive]`

So I was digging in the tone library and found that I need to write the ISR diffrently...
`void  timerISR(HardwareTimer *HT){ 
  Timer2.detachInterrupt(1); // Stop the timer from firing this ISR
 //do stuff...
}`
This does compile but it seems it does not work...
I am a begginer in this stuff and I don't really understand what is the meaning of HardwareTimer *HT.

On Arduino boards and Teensy boards yo don't need to write IRSs like that...

Thank you!

fpistm, 2019-09-26:
Hi @TadyTheFish 
Here the wiki:
https://github.com/stm32duino/wiki/wiki/HardwareTimer-library
You can install the library STM32Examples to get some examples.
`HardwareTimer *HT` is the hardware timer pointer, that's all.

TadyTheFish, 2019-09-26:
Wooohoooaa thank you!!!! How did I missed this..

closed 2019-09-26 by TadyTheFish

fpistm, 2019-09-26:
Welcome.

TadyTheFish, 2019-09-26:
Hmmmm library manager doesn't find the library

fpistm, 2019-09-27:
Which library ? STM32Examples?

TadyTheFish, 2019-09-27:
Yes... I tried searching through github for the link you provided but can't find anything. Maybe I missed something.
The only link to examples that I can find is this .. https://github.com/stm32duino/STM32Examples
This should be easier to access...

How can I install the examples? Library manager? If so It cannot find STM32Examples

fpistm, 2019-09-27:
Well, I guess you search `STM32Examples` and it not found it while searching `STM32 Examples` it works:

[image]

TadyTheFish, 2019-09-27:
Thank you ... It was the space :)