stm32duino/Arduino_Core_STM32 #675: package STM32CubeProgrammer CLI

MauroMombelli, 2019-09-29:
Arduino is supposed to be a beginner friendly and hassle free platform for beginner.
Having to manually install 3rd part tool not only is inconvenient, but there is also a problem understanding witch version of the tool should be used with what version of the core, in case future incompatibility/support will be introduced in future.

**Describe the solution you'd like**
Add STM32CubeProgrammer required binary in the core

**Describe alternatives you've considered**
if that is not possible (licence?) add back old tools

fpistm, 2019-09-29:
Hi @MauroMombelli 
Unfortunately I will not add the STM32CubeProgrammer for several reason (too big, license,...) this is up to end user to use it or not. About version, last is the answer.
Back to old tools is not a solution as they do not support all STM32 series.

fthiebolt, 2019-09-30:
Dear,

First of all, thanks for your great work in the Arduino world :)

Then, my position will be a bit different from the previous comment in a way that i want the old serial upload method to get back! In fact, i just want to use my regular USB serial converter (e.g CP2102) to upload code to my stm32 (serial + boot0 setup) ... WITHOUT the stm32CubeProgammer software i don't want to hear about.

Additionally, i absolutely agree with the previous comment about the fact the Arduino env ought to get simple enough to allow you to program your devices without any additional software (with or without liences).

Best regards

fpistm, 2019-09-30:
Hi @fthiebolt 
as said old tools are not compatible with all series/MCU so this is not an options.
Anywa Arduino allow to customize the core to fit your needs:
https://github.com/stm32duino/wiki/wiki/Custom-board-based-on-a-core

With this method each one can use the tool he wants. :wink:

fthiebolt, 2019-09-30:
Bonsoir Frédéric,

You're right, BSFrance is a small company selling great little boards featuring a LoRa module our University buy ... and i teach my students and colleagues how to develop devices with Arduino (hence the variant) :)

Ok, i will have a look on how to add stm32flash (serial uploader) for these boards.

Now, generally speaking, it's not because your car is unable to climb a mountain that you ought to force everyone not to use it ... sometime you just live in a flat land ;)
Byebye
F

fpistm, 2019-09-30:
Bonsoir,
Generally  speaking supporting several different tools for several upload method which do not support all STM32 series is a pain and require a lot of time. 
If you want support officially those tools I will be glad to add them. 😁
The STM32cubeprogrammer is multi OS, support all STM32 series and support several upload method.
This is, I guess, not too hard for end user to install it one time. And I repeat that it is only needed for some upload method.

And my car climbs well in the mountains, the most difficult is not to climb, but to find a beautiful road to satisfy all passengers, which is sometimes very complicated. 

Byebye

MauroMombelli, 2019-09-30:
my user experience:
this weekend I opened a old project to execute a quick fix, just five minute.

Arduino asked me update the plugin, why not?

Well, because now it wont upload anymore, as i need this tool.
No problem i thoug, just and rolling back to 1.5.0.. Nope, i get an error that the compiler is not found (I checked and in the folder there is no arm binary at all).
Must be a fluke, lets start from a clean Arduino install and add plugin 1.5.0. Nope, same issue.
Then I headed myself to download the cube, find where put the path etc..
Once i finally get all working, I had no more time or will to do my "5 minute" fix.

---

And this is me, that I know what a path is, comfortable editing configuration file by hand, and used to the ST website.
But I am is NOT the typical Arduino user; imagine to have to deal with 25-30 17 years old that barely understand the difference between a relative and an absolute path.

With cube, with old tool, i dont really care what is under the hood, as long as it just work

fpistm, 2019-09-30:
@MauroMombelli 
Sorry that rolling back doesn't work, I test this and it works fine for me.
What I can said is some user made some stuff to made some tests and then original Arduino IDE configs are corrupted, (JSON,  staging packages are not properly aligned). I could not handle that.

> (not sure what i was using before)

I remember you already submit issue and your config was not correct:
https://github.com/stm32duino/Arduino_Core_STM32/issues/317#issuecomment-419656878

As you said yourself you are comfortable that's nice. A typical Arduino user will mainly use a ST board which support mass storage upload method and that's all. 
User who need other upload method are generally more comfortable using this tools as they probably need to connect manually the STLink or connect the Serial to flash.
This core is used  by several beginner and they does not claim it was so hard to use. Maybe some of them met some issue but mainly due to lack of documentation, I know that and try to enhance this.
I do my best to have the core as simple and easy to use but we will always find someone who are not satisfied... 
Finally, I guess  we reach the same state like this issue:
https://github.com/stm32duino/Arduino_Core_STM32/issues/217#issuecomment-411898902

At this time it's 22:00 and working at home so time to stop.
For all reasons provided above, STM32CubeProgrammer is the only tools available to handle properly all upload method. If some other tools can fit all requirements then I will be glad to add it but at this time I do not know which one (texane stlink, STM32Flasher,...)

closed 2019-09-30 by fpistm

MauroMombelli, 2019-09-30:
> Sorry that rolling back doesn't work, I test this and it works fine for me.

please note io started from a fresh install, arduino 1.8.9 on linux, if that maybe could be the issue. not gonna open another issue as I guess make no sense to back-patch 1.5.0, but is nice is documented somewhere in case someone hit this bug.

> As you said yourself you are comfortable that's nice. A typical Arduino user will mainly use a ST board which support mass storage upload method and that's all.

oh, didn't know this has been added, just tested and found an strange behavior, see https://github.com/stm32duino/Arduino_Tools/issues/41

fpistm, 2019-09-30:
> please note io started from a fresh install, arduino 1.8.9 on linux, if that maybe could be the issue. not gonna open another issue as I guess make no sense to back-patch 1.5.0, but is nice is documented somewhere in case someone hit this bug.

I've tested and it works. I don't know why you met all this issue but this is related to your environment.