stm32duino/Arduino_Core_STM32 #676: [Question]Can't upload with stlink

fishmingyu, 2019-10-01:
I have some problems dealing with upload 
Firstly, I give my desktop configuration as below:

| settings | info |
| --- | --- |
|OS|windows 10|
|Arduino IDE|Arduino 1.8.9 (windows store 1.8.21.0)|
|STM32core version|1.7.0|
|Board Part Numbers|Black F407VE|
|Upload methods|stm32CubeProgrammer(SWD)|

I have already downloaded the stm32duino package from Boards Manager and installed the stm32CubeProgrammer. Also, I have connected the stlink v2 to my PC.(My stlink v2 doesn't have a virtual COM). Though there is nothing wrong with compiling, the IDE shows these errors: 
> STM32_Programmer_CLI.exe not found.
> Please install it or add <STM32CubeProgrammer path>\bin' to your PATH environment:

But which path does these errors refer to? How can I add the path correctly?
Sincerely thanks for your support!

fpistm, 2019-10-01:
Hi, did you install Cube programmer in the default installation dir? If not then you have to add it to windows PATH environment.
This is what is described in the wiki.
I will add more detailed about what is expected

fishmingyu, 2019-10-01:
Oh, I'm really sorry for my carelessness. Thanks so much!

closed 2019-10-01 by fishmingyu

fpistm, 2019-10-01:
No worry :wink: