stm32duino/Arduino_Core_STM32 #678: SPI external eeprom suport

mayurharge, 2019-10-02:
hello @fpistm ,
I want to integrate SPI based external [EEPROM](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8803-SEEPROM-AT25080B-160B-320B-640B-Auto-Datasheet.pdf) with STM32F103C controller, The issue is the EEPROM is not responding to SPI.h library. 
I have tested the EEPROM on Arduino UNO using This [Code](https://www.arduino.cc/en/Tutorial/SPIEEPROM)

here they have used ATMEL specific instructions   
```
//sample on leading edge of clk,system clock/4 rate (fastest)
  SPCR = (1<<SPE)|(1<<MSTR);
  clr=SPSR;
  clr=SPDR;
```
 which are not supported on stm.
Please suggest the solution for this.
other details: Arduino IDE :- 1.8.0
STMCore: 1.7.0(latest).

regards,
Mayur

fpistm, 2019-10-02:
Hi @mayurharge 
your code is for UNO direct register access. I guess this example was made before API update.
You have to create your own sketch using SPI API:
https://www.arduino.cc/en/reference/SPI

Did you test some third party libraries, ex:
https://github.com/regis44/spieeprom

mayurharge, 2019-10-02:
Okay I will Try this

mayurharge, 2019-10-02:
I have tried the library you suggested its behaving same.
The error I am facing is 
I will get random values like 0,31(32-1),127(128-1),255(256-1) if i read a particular register 
I am attaching the .ino file I am using, please check.
[SPI_EEPROM.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3681524/SPI_EEPROM.zip)

fpistm, 2019-10-02:
@mayurharge 
unfortunately, I'm not able to review all third party or user code and debug it.
Your code seems to be a lot of copy/paste and try and use deprecated API.
Please take time to make your code and try to debug it.

mayurharge, 2019-10-02:
Sure

mayurharge, 2019-10-03:
The issue is solved.
There was problem with spi speed.
After reducing the speed the communication was fine

closed 2019-10-03 by mayurharge