stm32duino/Arduino_Core_STM32 #679: PWM Build Error on Arduino IDE

hyuny0505, 2019-10-02:
Hi!
I'm trying to build my sketch in Arduino IDE, but it can't compile PWM.
[image]
It causes an error like the screenshot.
I'm using Arduino IDE 1.8.9 in Windows and Arduino_Core_STM32 1.7.0 and using BluePill F103C8. The upload method is STM32CubeProgrammer (SWD) with ST-LINK v2.
I searched for many examples on the internet. Almost code uses pinMode(xx, PWM); and pwmWrite(); with no error, but it causes error for me.

fpistm, 2019-10-02:
Hi @hyuny0505 ,
`PWM` is not a valid argument, it is for this core:
https://github.com/rogerclarkmelbourne/Arduino_STM32/

Same for `pwmWrite`.
Those syntaxes do not follow ooficial Arduino API.
Simply use one of the Arduino sketches like the fading.

closed 2019-10-02 by fpistm

hyuny0505, 2019-10-02:
Oh, sorry. Really sorry. Thank you for your help!

fpistm, 2019-10-03:
No worry 😉

deltavra, 2020-08-13:
> Hi @hyuny0505 ,
> `PWM` is not a valid argument, it is for this core:
> https://github.com/rogerclarkmelbourne/Arduino_STM32/
> 
> Same for `pwmWrite`.
> Those syntaxes do not follow ooficial Arduino API.
> Simply use one of the Arduino sketches like the fading.

so which argument is valid?

fpistm, 2020-08-13:
`OUTPUT `even if it is not needed to se pinMode for analogWrite() as it will configure it properly.