stm32duino/Arduino_Core_STM32 #681: Compiler doesn't throw warnings for a function not returning valid type

arihantdaga, 2019-10-04:
For hours i was stuck when my mcu was going in an indefinite loop. Later i found that one of the function was being called continuous in loop, while i called that function only once. 

Later i discovered that it was because the parent function which was calling this function had a syntax of return bool, but in code, i forgot to return anything. 

```
void func1();

// Faulty Method
bool myfunction(){    
    func1();
}

void myMainFunction(){
int a = 0;
myfunction();

}
```

I was wondering if compiler can throw the error for such function. 
I am not sure, i think maybe these compiler warnings could be out of scope from the arduino core. Or is there any option, to enable warning levels etc.

fpistm, 2019-10-04:
Hi @arihantdaga 
the compiler already warn you but I guess you do not configure the Arduino IDE to show **All** the **Compilers warnings**, in the Arduino Preferences:
[image]

Results:
```
/tmp/arduino_modified_sketch_204604/sketch_oct04a.ino: In function 'bool myfunction()':
/tmp/arduino_modified_sketch_204604/sketch_oct04a.ino:9:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
/tmp/arduino_modified_sketch_204604/sketch_oct04a.ino: In function 'void myMainFunction()':
/tmp/arduino_modified_sketch_204604/sketch_oct04a.ino:13:9: warning: unused variable 'a' [-Wunused-variable]
     int a = 0;
         ^
Sketch uses 3696 bytes (5%) of program storage space. Maximum is 65536 bytes.
Global variables use 48 bytes (0%) of dynamic memory, leaving 8144 bytes for local variables. Maximum is 8192 bytes.

```

arihantdaga, 2019-10-05:
Ooo.. Thanks and Sorry.

closed 2019-10-05 by arihantdaga

fpistm, 2019-10-05:
No worry.