stm32duino/Arduino_Core_STM32 #682: DSP functions

superpanza, 2019-10-04:
Hello, how can i use the CMSIS-DSP functions?
I'm trying to perform an FFT over some samples but i'm not able to import arm_math.h from DSP lib. It does not find any math_arm.h lib from filesystem. I'm pretty sure that the header file is there (view img
[image]
)
i'm using a nucleoF401RE board, and i've installed the stm32core via Arduino IDE board manager

fpistm, 2019-10-04:
Hi @superpanza 
Currently, this is not provided.
I've already answer this kind of issue here:
https://github.com/stm32duino/Arduino_Core_STM32/issues/569#issuecomment-514377729

I think I will add a way to use it.

superpanza, 2019-10-04:
Great!
Thank you for considering adding the feature. 
I've just tried to apply the solution you suggest in #569 and now the lib (arm_math.h) is loaded, the code compile successfully, but the linking phase fail with something like 

> /Users/userino/Library/Arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld: FFTTest.ino.cpp:(.text._Z6armFFTPfS_i+0x1a): undefined reference to `arm_cmplx_mag_f32'

The prototype of that function is clearly in arm_math.h
Any idea on how to move on?

fpistm, 2019-10-04:
I guess, you have to add the c source file defining the function in the build:
CMSIS/5.5.1/CMSIS/DSP/Source/ComplexMathFunctions/arm_cmplx_mag_f32.c

like I've made for the filtering:
https://github.com/fpistm/Arduino_Core_STM32/commit/e891921bd42e9f7819ef0d7004feed4f9ec53f96#diff-766ab07e0a5996a0f67900b49895d65f

fpistm, 2019-10-04:
Do not hesitate to provide a small test sketch then I will be able to test when I will implement this request.

fpistm, 2019-10-07:
Hi @superpanza 
I've added the support of CMSIS DSP software library here #685 
Could you test it and give me your feedback, please ?
Thanks in advance.

PR #685 (2019-10-15): Add CMSIS DSP support as built-in library
To use it, add:
` #include <CMSIS_DSP.h>`    

`arm_math.h` is then automatically include.

Fixes #682
commit 92197ec8 (2019-10-15): Add CMSIS DSP support as built-in library

To use it, add:
#include <CMSIS_DSP.h>

Fixes #682
closed 2019-10-15 by fpistm