stm32duino/Arduino_Core_STM32 #683: CAN, I2C, Serial, PWM - should I start project with Arduino?

SuperBoss9, 2019-10-07:
Hello EveryOne!

Due to absence of the official forum according to STM32Duino I have to ask here.

I'm going to start a project at STM32F103RB chip (and use for tests Nucleo STM32F103RB board) that will use broadly the following things:

1. PWM out to 4 pins. It is good to be able to set frequence of the PWM.
2. Voltage measurement at 6 pins.
3. Management of 2 relays.
4. I2C communication with EEPROM chip.
5. CAN communication (using STM32 on board feature) with a vehicle network.
6. Encoders, buttons and LEDs as usually.
7. Waste mathematics in a loop.
8. Serial listener and parser for commands understanding.

It will be good to use RTOS because it can help in development. But ordinary scheduler and interrupts can be also used.

**The question is - should I start my development using STM32Duino or try again for MBED?**

PS. I spent almost two month trying to select the best framework for this development. I have experience in Arduino (AVR, ESP8266). But decided to try MBED. After 2 months with MBED I can say that I'm ready to return back to Arduino (PlatformIO + STM32Duino) because:
a. Absence of normal String class in MBED => Constant hassle with char[], c_str, string etc.
b. Any action requires a lot of efforts + strange behaviour of some components.
c. Lack of memory (103RB has 20 RAM that should be enough). MBED core (OS 5) takes a lot of memory even for empty program. I run of memory not implementing even 1/30 of all functions I need.
d. Long compilation time. MBED compiles everything. Under PIO it takes minutes, under Mbed Studio it takes tens of seconds. Arduino with STM32Duino compiles instantly.

fpistm, 2019-10-07:
Hi @SuperBoss9 

One blocker for you could be 

> * CAN communication (using STM32 on board feature) with a vehicle network.

Currently, this is not supported (See #259 ). anyway it could be implemented on your side using HAL/LL.

fpistm, 2019-10-07:
As you plan to use F103RB, one other possibility for you could be to use this core which I think support CAN:
https://github.com/rogerclarkmelbourne/Arduino_STM32

It is also available thanks PIO:
https://docs.platformio.org/en/latest/platforms/ststm32.html#configuration

SuperBoss9, 2019-10-07:
> 
> Currently, this is not supported (See #259 ). anyway it could be implemented on your side using HAL/LL.

Is there any walkthrough how to implement this to read? For example:
1. Go to Cube, set-up CAN hardware.
2. Generate Keil code
3. Copy-paste code into your sketch

?

fpistm, 2019-10-07:
yes, something like that and also think to define `HAL_CAN_MODULE_ENABLED`

pkourany, 2019-10-08:
@SuperBoss9, the use of Arduino String, though convenient, can cause heap fragmentation due to its use of dynamic memory allocation.  This, in turn, can lead to system instability or even crashing as the heap becomes unusable.   I would strongly recommend becoming proficient in C-string manipulation instead.

SuperBoss9, 2019-10-09:
Thank you everyone for the support for moving back to Arduino at STM32

closed 2019-10-09 by SuperBoss9